esp8266/Arduino #1786: No Hardware RX TX Buffer sizes

PhilColbert, 2016-03-17:
From 2.0.0. to 2.1.0 the TX and RX buffer sizes are gone  :(

My code is dealing with GPS input therefore requires a large buffer size to cope.... please put it back  ! :)

igrr, 2016-03-17:
Removing the software-managed buffers was intentional (#1496), because otherwise we had issues when Serial was used from tasks other than main arduino task. For your application you can probably poll data at a higher rate. There is also a suggestion to support Arduino serialEvent API (#752).

PhilColbert, 2016-03-17:
Thank you for the reply.

Unfortunately I cant poll at a higher rate as the data requires parsing as it comes in and this requires uploading back to a wificlient and to a radio transmitter, this takes some time, in that time the serial buffer fills ...

Sorry I am still new to coding with the esp so might be coding it incorrectly.

I will just have to stick with 2.0.0 for now.

Thank you anyway.

PhilColbert, 2016-03-17:
I am receiving around 500 - 1000 bytes  / sec...  any idea how to deal with this issue in 2.1.0 without the serial buffer issue?

Thanks

tablatronix, 2016-03-20:
Does this mean the docs need to change ?

> Apart from hardware FIFO (128 bytes for TX and RX) HardwareSerial has additional 256-byte TX and RX buffers

far5893, 2016-03-24:
I suppose that hardaware fifo is still used.

devyte, 2017-10-18:
The docs for HardwareSerial need to be updated to describe the setRxBufferSize() method.

PR #3862 (2017-11-22): Added Serial.setRxBufferSize method description in libraries reference document
Hi, this is a proposal to fix #1786
closed 2017-11-22 by devyte