esp8266/Arduino #1787: Error Downloading from Board Manager

fh9g12, 2016-03-17:
### Basic Infos

Cant install board using either stable or Staging link

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Hello there, when I try to install the boards via the board manager in arduino 1.6.8 ( also tried 1.6.5 ) I always get a 'Error Downloading' shown in the botum of the board manager on the first package ( 've tired all version as well ), how if I go to the links I can download them manually... I am ware this is probably a permissions issue or something at my end, but I was wondering if other people had had a similar problem / have a solution / could tell me where t place manually downloaded files so 'it thinks its downloaded them' ??

fh9g12, 2016-03-18:
I 'man'ed up and just learn how do it the git repositry way instead... although the initial issue still stands..

drewfish, 2016-08-01:
I've just run into this too, specifically:
* Arduino IDE 1.6.9
* added `http://arduino.esp8266.com/stable/package_esp8266com_index.json` to the additional board manager URLs in the preferences (along with the json for adafruit and sparkfun boards)
* tried to install 2.3.0 of esp8266 in the board manager
* got `Error downloading https://github.com/igrr/mkspiffs/releases/download/0.1.2/mkspiffs-0.1.2-osx.tar.gz`

drewfish, 2016-08-01:
Hmmm... just tried again and it worked...

vishva8kumara, 2017-05-09:
I tried several times and it failed. So I manually downloaded the json file from http://arduino.esp8266.com/versions/2.3.0/package_esp8266com_index.json and mirrored on http://info.lk/static/package_esp8266com_index.json

Now it work for me. I'm leaving the file on that server for anyone having the same problem to use.

ballew, 2017-05-16:
I can't download it either. Used the link provided by vishva8kumara which worked. Thanks vishva8kumara

sunilvb, 2017-05-17:
Unable to install the package as well... 
get the following error 

*Error downloading https://github.com/esp8266/Arduino/releases/download/2.3.0/esp8266-2.3.0.zip 
javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target*

Although, I can manually point the browser to github and download the esp8266-2.3.0zip.

faisalzed, 2017-06-02:
Just disable the firewall . go and solve it all

devyte, 2017-10-18:
Can't reproduce, closing.

closed 2017-10-18 by devyte