esp8266/Arduino #1788: SSL certificate size

tijnkooijmans, 2016-03-17:
Is there any limit to the SSL certificate size in axTLS?

I'm using WifiClientSecure and when I have fully-chained certificate (6 kb) on the server the connection gets refused (error -1) while if the server only uses the non-chained certificate (1.8 kb) there is no problem.

I have 30 kb heap available when making the SSL connection so it's no memory issue.

Thanks in advance for looking into this!

igrr, 2016-03-17:
Could you please advise how to reproduce this? What server URL should I use for non-chained certificate and fully-chained certificate?

tijnkooijmans, 2016-03-17:
Thanks for your super-fast response! This is the one without chain:

brokerstagingdev.minibrew.io

This is the one with chain:

brokerstaging.minibrew.io

igrr, 2016-03-17:
Okay, this is what seems to be happening... In order to reduce memory pressure during handshake, we allocate smaller TLS fragment buffer (~6kb) than per spec (16kb). Then when connection is established, we allow this buffer to grow as required. I will check if we can allow this buffer to grow during handshake, i.e. whether it will fit into RAM.

tijnkooijmans, 2016-03-17:
Great, thanks! Looking forward to test if you can make this work.

igrr, 2016-03-20:
This issue was moved to igrr/axtls-8266#13

closed 2016-03-20 by igrr