esp8266/Arduino #1791: #include "driver/key.h" causes "No such file or directory"

waynemcl, 2016-03-18:
Hi,

I'm attempting to compile a WPS example (via visualStudio with VisualMicro installed, but I get the same error direct from the arduino IDE, where the additional board was installed).

The code is from CHERTS unofficial SDK, but it is also in the 1.5.2 SDK from espressif. 

Arduino 1.6.8
edit - preferences - settings - Additional Boards Manager URLS: "http://arduino.esp8266.com/stable/package_esp8266com_index.json"

Tools - Board - Boards manager reports version 2.1.0

#### Hardware
Hardware:			ESP-12
Core Version:      	n/a

### Description

Problem description

### Settings in IDE
Module:  			Generic ESP8266 Module?
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

Thanks in advance, great project folks!

igrr, 2016-03-18:
We don't have the whole `driver` directory in this repository: https://github.com/esp8266/Arduino/tree/master/tools/sdk
Did you try adding the necessary files directly to your sketch?

waynemcl, 2016-03-20:
Including the files directly does seem to compile, thank you. I'm working through the rest of the dependencies. For onlookers interested in WPS, there's also an ardiuno library for WPS configuration by button, which then falls back to a softAP/webserver for entering WiFi credentials, here:
https://github.com/tzapu/WiFiManager

closed 2016-03-20 by waynemcl