esp8266/Arduino #1798: Can we add authentication to httpupdateserver ?

CorBer, 2016-03-20:
Hi,
This is not a real issue but more a request to change the sourcecode.

To improve security when using OTA httpupdateserver adding authenticationis helpfull. Ive have tried with succes using only a few minor modifications to the current stable release.

in ESP8266HTTPUpdateServer.h change :
class ESP8266HTTPUpdateServer
{
  private:
    bool _serial_output;
    ESP8266WebServer *_server;
    static const char *_serverIndex;
  public:
   ESP8266HTTPUpdateServer(bool serial_debug=false);
   void setup(ESP8266WebServer *server=NULL);
 /*ADDED*/   char* USR_NAME; 
 /*ADDED*/   char* USR_PWD;
};

in ESP8266HTTPUpdateServer.cpp change 
void ESP8266HTTPUpdateServer::setup(ESP8266WebServer *server)
{
    _server = server;
    // handler for the /update form page
    _server->on("/update", HTTP_GET, [&](){
/*ADDED*/      if(!_server->authenticate(USR_NAME, USR_PWD))
/*ADDED*/   	        return _server->requestAuthentication();
      _server->sendHeader("Connection", "close");
      _server->sendHeader("Access-Control-Allow-Origin", "*");
      _server->send(200, "text/html", _serverIndex);
    });
......
.....
}

the USR_NAME and USR_PWD can be set directly from the sketch as they are part of the structure.
So you can use   (based on the webupdater example)
   httpUpdater.USR_NAME="myName";
   httpUpdater.USR_PWD="myPWD";

Tested and working as planned. I am sure real coders can do better than this.

regards
CorB

mkeyno, 2016-04-27:
dear @igrr  , I know you have busy time but I have question about the bool ESP8266WebServer::authenticate(const char * username, const char * password) and really appreciated for your help
actually I intend to overload authenticate function to return decoded base64 input username and password so I could check it against my user database(all accepted user-pass data) , I've check lib64/cdecode.c header file but I can't figure out how to decode base64 user-pass data
can I have your idea about how to do that?
BR

devyte, 2017-10-18:
ESP8266HTTPUpdateServer has authentication capability in current git. Closing.

closed 2017-10-18 by devyte