esp8266/Arduino #1799: Compiling Error with the esp8266 examples

hiddenotebook, 2016-03-20:
### Basic Infos

#### Hardware
Arduino:                           MEGA
Hardware:			ESP-01
Core Version:      	         2.1.0

### Description

I can't simply compile the examples.

### Settings in IDE

Module:  			MEGA
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		 SERIAL
Reset Method:		?ck / nodemcu?

### Sketch

WiFiWebServer.ino

### Debug Messages

 this is the error from serial.

In file included from /home/zion/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFi.h:33:0,
from /home/zion/Arduino/libraries/ESP8266WiFi/examples/WiFiWebServer/WiFiWebServer.ino:10:
/home/zion/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFiType.h:26:19: fatal error: queue.h: No such file or directory #include <queue.h>

but damn it  the file is there I don't know what I’m doing wrong can anybody help!

hiddenotebook, 2016-03-20:
I'm making progress, I had to choose the programmer genericESPmodule hahahah sorry 

Module: ESP8266-01
Flash Size: 4M(1M SPIFSS)
CPU Frequency: 80Mhz
Flash Mode: qio
Flash Frequency: 40Mhz
Upload Using: SERIAL
Reset Method: ck 

is there any guide or tutorial?

igrr, 2016-03-23:
Sure, you may check out this tutorial by Adafruit: 
https://learn.adafruit.com/adafruit-huzzah-esp8266-breakout/using-arduino-ide

closed 2016-03-23 by igrr

hiddenotebook, 2016-03-25:
Thank you Ivan, I'll check right now this module is driving me crazy!

karanmakharia, 2016-11-11:
What if I want to use ESP8266 with Arduino. Then how do I resolve the queue.h error.

siggiba, 2017-11-27:
Same question as  karanmakharia:  I want to use the simple ESP8266-01 connected to Mega 2560 over Serial2 - any suggestions how to resolve the queue.h error??

Or is there another library for Arduinos to send data to ESPxxxx?

geek-ibrar, 2018-10-01:
Same error with aduino using ESP8266

> In file included from /home/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFi.h:33:0,
>                  from /home/Arduino/wifi_tmp/wifi_tmp.ino:3:
> /home/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFiType.h:26:19: fatal error: queue.h: No such file or directory
>  #include <queue.h>