esp8266/Arduino #1802: UDP will not send until it has received at least one packet

jgrina, 2016-03-21:
### Basic Infos

#### Hardware
Hardware:			Adafruit Huzzah, NodeMCU, Adafruit Feather
Core Version:      	2.1

### Description

Why does udp sendPacket wait for a received packet before sending? It is only the first send command that waits. If I force a transmit from host, a burst of "sent" packets appears and everything works great from there. Sends and receives go. I'm using Arduino IDE version 1.6.8 on a Win 7 machine for this listing, but the same happened with version 1.6.5 on my Win 10 machine. The ESP is an Adafruit Huzzah breakout board, but I also tried the ESP feather 8266 and the NodeMCU boards with the same result. I have updated the libraries to the latest code while installing IDE version 1.6.8.

The attached zipfiles are the program, monitor activity from the host and Arduino com port monitor

The arduino side file shows the ESP messages which have the sends, but no response until the host sends "!Start" message. The host side shows no messages until after the !Start message, then a burst of messages with responses to each message.

I've tried several WiFi.modes, plus udp.flush(), udp.stop(), commands, but nothing changes.

With a looping program, this is not a major problem, I can use the monitor send function to start the ESP8266 noodes, but I want to use battery power and deep sleep function for some nodes. Deep sleep erases any knowledge of previous packet receives, so it must receive a packet every time it wakes up.

Host side is Win 7, MegunoLink Pro
I am using the Timer function in MegunoLink to broadcast a "!Start" message every 1/2 second

Jim

### Settings in IDE

Module:  			Huzzah, generic ESP, NodeMCU 1.0
Flash Size: 		4Mb
CPU Frequency:		80mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Upload Using:		serial 
Reset Method:		ck

[zipfiles.zip](https://github.com/esp8266/Arduino/files/182561/zipfiles.zip)

igrr, 2016-03-21:
It would be really helpful if you could post the sketch which may be used to reproduce this issue!

igrr, 2016-03-21:
For the record, i think UDP should be able to send packets first. You can check NTPClient sample for the demonstration of it.

jgrina, 2016-03-21:
Igrr, I attached the sketch file and the output from both the host and ESP. They are at the end of the post "zipfiles.zip". The NTPclient first uses DNS to get the IP address of the .gov server. I can try that, but not send the UDP request to time server. I'll see if that works. However, I was trying to design a UDP based program which wakes up, sends the data and goes back into deep sleep. Deep sleep is important for battery life. Depending on the amount of time awake vs asleep, It could make a battery last 60 times as long with sleep mode. I'll get back to you about using the Time Server to wake up UDP sends.

pipi61, 2016-03-21:
Hi!
this example not a NTP sample...
try after sendPacket(statusUpdate); long time delay...

jgrina, 2016-03-21:
I tried the UDP timeserver, with a small modification. I ran the time server, got a valid IP for server and used the IP address of the server.- 

  //get a random server from the pool
  // WiFi.hostByName(ntpServerName, timeServerIP); 
  IPAddress timeServerIP(24,56,178,123);
  sendNTPpacket(timeServerIP); // send an NTP packet to a time server

The result was a long list like this -

sending NTP packet...
no packet yet

sending NTP packet...
no packet yet

sending NTP packet...
no packet yet

sending NTP packet...
no packet yet

sending NTP packet...
no packet yet

sending NTP packet...
no packet yet

sending NTP packet...
no packet yet

I cannot receive the NTP packet from server, because I have not actually send the packet. If I just change the program to -

  //get a random server from the pool
  WiFi.hostByName(ntpServerName, timeServerIP); 
  // IPAddress timeServerIP(24,56,178,123);
  sendNTPpacket(timeServerIP); // send an NTP packet to a time server

I get the time with no problem. -

Connecting to Phoenix
......
WiFi connected
IP address: 
192.168.1.102
Starting UDP
Local port: 2390
sending NTP packet...
packet received, length=48
Remote host = 128.138.141.172:123
Seconds since Jan 1 1900 = 3667589078
The UTC time is 22:44:38

sending NTP packet...
packet received, length=48
Remote host = 128.138.141.172:123
Seconds since Jan 1 1900 = 3667589109
The UTC time is 22:45:09

Connecting to Phoenix
......
WiFi connected
IP address: 
192.168.1.102
Starting UDP
Local port: 2390
sending NTP packet...
packet received, length=48
Remote host = 128.138.141.172:123
Seconds since Jan 1 1900 = 3667589078
The UTC time is 22:44:38

sending NTP packet...
packet received, length=48
Remote host = 128.138.141.172:123
Seconds since Jan 1 1900 = 3667589109
The UTC time is 22:45:09

If I trick the remote host that I'm using for temperature monitoring via UDP, into sending packets to port 2390, I get - 

sending NTP packet...
no packet yet

sending NTP packet...
no packet yet

sending NTP packet...
no packet yet

sending NTP packet...
no packet yet

sending NTP packet...
no packet yet

sending NTP packet...
packet received, length=12
Remote host = 192.168.1.105:2881
Seconds since Jan 1 1900 = 0
The UTC time is 6:28:16

sending NTP packet...
packet received, length=12
Remote host = 192.168.1.105:2881
Seconds since Jan 1 1900 = 0
The UTC time is 6:28:16

sending NTP packet...
packet received, length=12
Remote host = 192.168.1.105:2881
Seconds since Jan 1 1900 = 0
The UTC time is 6:28:16

sending NTP packet...
packet received, length=12
Remote host = 192.168.1.105:2881
Seconds since Jan 1 1900 = 0
The UTC time is 6:28:16

Now my temp monitoring host has no idea about time servers, but the IP address is 192.168.1.105.

Jim

I suppose I could include the -
  WiFi.hostByName(ntpServerName, timeServerIP); 

command in my program. but I don't like the idea. I have 4 operational nodes, and ideas for several more. I don't think  having a half dozen nodes asking for ntp server ip's is a good idea.

jgrina, 2016-03-22:
OK, I reopened. I made a mistake

pipi61, 2016-03-22:
Hi!
try network tests program to receive and view sended packet from this page:
http://ruten-proteus.blogspot.hu/2014/12/esp8266-at-command.html

jgrina, 2016-03-22:
I got SocketTest 3.0.0 and the results were the same as my host program. ESP will not speak unless spoken to. So I'm stuck. I see the problem that I reported is listed as troubleshooting, no milestones and no one assigned to it. Dunno, I'll just wait a few days.

igrr, 2016-03-22:
Sorry, I haven't noticed the sketch first time i extracted the archive. I loaded the sketch, made the following changes to fit my network environment:
- change WiFi mode to `WiFi.mode(WIFI_STA);`
- change network SSID and password
- remove WiFi.config call because I have DHCP in my network
- changed remoteIpNo to match the IP address of my laptop
- added `localIpNo = WiFi.localIP();` at the end of `setup` function

Then fired up nc on my laptop, and saw packets coming in:
```
[igrokhotkov@igrokhot-mbp ~]$ nc -l -u -p 2881
{UI|SET|StatusM.Text=192.168.1.35}
{UI|SET|StatusM.Text=192.168.1.34}
{UI|SET|StatusM.Text=192.168.1.35}
{UI|SET|StatusM.Text=192.168.1.34}
```

Output from serial monitor was:
```
Local port: 2880
*********** start loop **************

Send To 192.168.1.34
{UI|SET|StatusM.Text=192.168.1.35}
No response

Send To 192.168.1.34
{UI|SET|StatusM.Text=192.168.1.34}
No response
****** end loop ****** 
0
*********** start loop **************

Send To 192.168.1.34
{UI|SET|StatusM.Text=192.168.1.35}
No response

Send To 192.168.1.34
{UI|SET|StatusM.Text=192.168.1.34}
No response
****** end loop ****** 
1
```
and so on.

So unfortunately I'm not able to reproduce this so far.
I thought that the difference might be due to the fact that i removed WiFi.config initially.
So I have put `WiFi.config` line back, and adjusted localIpNo. Gateway and netmask already matched my network. Loaded the sketch again, and I'm still able to receive packets.

In the end i think the issue you are seeing is due to `WiFi.mode(WIFI_AP)` line. Given that `WiFi.config` sets static config for STA mode, and not for AP, your ESP is setting up it's own network, and the IP is most likely 192.168.4.1, which explains why you don't get the data in SocketTest.

jgrina, 2016-03-23:
Interesting. I just used  - 
  WiFi.hostByName(ntpServerName, timeServerIP); 
In my program to start the udp.beginPacket() ... udp.endPacket();
and it works. Even though the hostByName function has nothing to do with the rest of the program, the udp packet that is sent as a response, does activate packet sending. Fascinating! I just need an activation method that doesn't use the NTP function. WiFi.hostByName obviously can send a udp packet, but udp.beginPacket can't. Hmm. I need to see how the hostByName manages to get around the limitation of receive before send. There are a bunch of issues involving broadcast, packet sending, etc. I wonder if they can all use hostByName as a work around.

igrr, 2016-03-23:
I can confirm that your example also works for me without WiFi.hostByName, once I change WiFi mode to WiFi_STA.

jgrina, 2016-03-23:
I see that you get it to work. Since you do and I don't, maybe it is the ESP libraries. The library included with IDE version 1.6.8 is ESP8266 builtin by Simmon Peter,Markus Sattler,Ivan Grokhotkov version 1.0.0 ESP8266 sketches examples. I tried to click on the "more info" but nothing happens. I downloaded the zipfile from GitHub but the zip installer in the IDE says it is not a library. I can't find where Arduino installs the ESP8266 libraries. I've used Arduino IDE since version 0019, but the new library manager has managed to make it very difficult to upgrade. I have the .json file for ESP8266 in my preferences, but it seems to think this is the newest release.

igrr, 2016-03-23:
You can check the core version in Tools > Boards > Boards manager menu, scroll to the end of the list, it should look like this:

[image]

Also did you change `WiFi.mode` call as I have suggested?

jgrina, 2016-03-23:
I'm looking athe the boards manager and it lloks like this

[image]

jgrina, 2016-03-23:
I went through your changes, made all of them. And yes, I did the Wifi.mode to STA

jgrina, 2016-03-23:
How do I get the version of ESP8266 that you are using? It wasn't included with the install of IDE version 1.6.8, and it doesn't show up on the updatable list of libraries.

igrr, 2016-03-23:
You must be looking at library manager, not the boards manager. Boards manager is in Tools > Boards menu.

jgrina, 2016-03-23:
I used sketch > include library > manage libraries for that snip I sent. Then looked at your instructions again. I went to TOOLS and the upgrade is there. I've upgraded, I think, but I have to go to my daughter's house, three hours away. and I won't be back until Monday.

Arduino: 1.6.8 (Windows 7), Board: "Adafruit HUZZAH ESP8266, 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

Board huzzah (platform esp8266, package esp8266) is unknown

Error compiling for board Adafruit HUZZAH ESP8266.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

jgrina, 2016-03-23:
Oh, yes, I love the new boards manager !@#$%.
OK, you set me on the right path, but the program will no longer compile. I can fix that, I think

jgrina, 2016-03-23:
[image]

igrr, 2016-03-23:
First you need to remove 2.1.0-rc2, then change URL in preferences to the *stable* boards manager package url, and then install 2.1.0 version.

jgrina, 2016-03-23:
However, I might not. Now the wonderful boards manager can't find the board. It shows me the board through the TOOLS menu but the compiler can't compile it.
Arduino: 1.6.8 (Windows 7), Board: "Adafruit HUZZAH ESP8266, 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

Board huzzah (platform esp8266, package esp8266) is unknown

Error compiling for board Adafruit HUZZAH ESP8266.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

jgrina, 2016-03-23:
[image]
So, I have the stable version installed, and I closed and reopened ArduinoIDE, but I still get -
Arduino: 1.6.8 (Windows 7), Board: "Adafruit HUZZAH ESP8266, 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

Board huzzah (platform esp8266, package esp8266) is unknown

Error compiling for board Adafruit HUZZAH ESP8266.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Juppit, 2016-03-23:
> new capture

Mostly it helps me to rename the folder 
C:\Users\[user_name]\AppData\Local\Arduino15 in somthing like 
Arduino.old and relod it. The preferences.txt must be reloaded.

jgrina, 2016-03-23:
I have two computers, The test computer is an old Win7, and the main computer is Win10. I went to the Win10 machine and loaded the ESP community version 2.1.0 I think this works. It has IDE version 1.6.5 on it. The program compiles and runs. It does not appear to need a awake up packet. I'd like to close this issue and straighten out the IDE version 1.6.8.

Thanks,
Jim

closed 2016-03-27 by jgrina