esp8266/Arduino #1805: ESP-WROOM-02 support?  Supported module list?

Humancell, 2016-03-22:
I'm curious if there is a list of supported modules, and specifically is the ESP-WROOM-02 is supported?

We're using the ESP-12F extensively, and have been testing on the ESP07 also ... with the upcoming ESP-07s coming soon.

Are their "officially" supported modules?

krzychb, 2016-03-22:
Hi Humancell,

The list of supported modules is [here](https://github.com/esp8266/Arduino/blob/master/doc/boards.md). Among them I see ESPino (WROOM-02 Module) by ThaiEasyElec that has ESP-WROOM-02 on board.

If you have a bare ESP-WROOM-02 module, then I would select "Generic ESP8266 Module" and adjust settings to match ESP-WROOM-02.

Krzysztof

closed 2016-03-23 by igrr

deepikavira, 2016-07-22:
@krzychb When you say change the settings to match ESP-WROOM-02 does that mean going to tools and then change the flash size, flash frequency etc..

krzychb, 2016-07-23:
@deepikavira, yes, this is what I meant.  "Generic ESP8266 Module" has all possible options available / exposed to talk to any supported module.

[image]

deepikavira, 2016-07-24:
Ok thank you I will try this and let you know how it works