esp8266/Arduino #1809: ets Jan  8 2013,rst cause:4, boot mode:(3,6)

EagleSmith, 2016-03-23:
### Basic Infos

#### Hardware
Hardware:		ESP-12E
Core Version:      	2.1.0

### Description

Problem description

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	80Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

```cpp

#include <Arduino.h>

void setup() {
Serial.begin(115200);
  Serial.setDebugOutput(true);

WiFi.mode(WIFI_STA);
    WiFi.disconnect(); 
    DEBUG_WM(F("WiFi mode STA"));

    if (connectWifi("", "") == WL_CONNECTED) {DEBUG_WM(F("connect wifi with no saved "));}
    else
    {
      DEBUG_WM(F("connect wifi with config value "));
      if (connectWifi(ssid.c_str(), password.c_str()) == WL_CONNECTED) {}
    }
    Serial.print(F("IP Address:"));
    Serial.println(WiFi.localIP());
    
    Serial.println("");
    Serial.print("Connected to ");
    Serial.println(ssid);
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  }
 HX711_Init();
}

void loop() {

}
```

### Debug Messages

ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

igrr, 2016-03-23:
Is this the same as https://github.com/esp8266/Arduino/issues/1335?

igrr, 2016-03-23:
Also would be cool if you provide the link to the source of HX711_Init function, since it is obviously causing issues.

EagleSmith, 2016-03-23:
SPIFFSImpl: allocating 512+180+1400=2092 bytes
SPIFFSImpl: mounting fs @100000, size=2fb000, block=2000, page=100
SPIFFSImpl: mount rc=0

EagleSmith, 2016-03-23:
thanks ,the code is :
#define SCK  12
#define DOUT 13

SCK = sck;
	DOUT = dout;
	set_gain(gain);
	COEFFICIENT = scale;
	pinMode(SCK, OUTPUT);
	pinMode(DOUT, INPUT);
	digitalWrite(SCK, LOW);
	if (is_ready())
	{
		read();
	}

EagleSmith, 2016-03-23:
i removed the hx711_init(),but the exception still happend

igrr, 2016-03-23:
I tried compiling your example, but I am getting an error that connectWifi OS not defined. Also in your hx117_init function, read function is not defined. Could you please post the complete source code?
It is also not clear why you are getting messages related to SPIFFS if you are not using it (at least not in the code you have posted).

igrr, 2016-03-23:
If you are getting an exception, please post the exception message here and decode it using EspExceptionDecoder tool.

EagleSmith, 2016-03-24:
https://github.com/bogde/HX711

sorry,My chip is 13

EagleSmith, 2016-03-24:
I use 1.6.5 （SDK2.0.0）to recompile， it is working fine

Exception occurs when WIFI connecting

jumpjack, 2016-10-05:
Solutions?

badf007, 2016-10-24:
Any solution ? tutorial ?

Adrianotiger, 2016-10-26:
> ets Jan 8 2013,rst cause:4, boot mode:(3,6)

Same with my ESP-12 using the AT commands firmware (boot1.2).

sayak-brm, 2016-11-11:
Got this same problem:

```c++
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

MDNSResponder mdns;

char ssid[] = "Darji_MBlaze";    // your network SSID (name) 
char pass[] = "santiniketan"; // your network password (use for WPA, or use as key for WEP)

ESP8266WebServer server(80);
String webPage = "<!DOCTYPE html><html><head><title>Remote Controlled Vehicle</title><script src=\"https://ajax.googleapis.com/ajax/libs/jquery/3.1.1/jquery.min.js\"></script></head><body><center><img src=\"http://192.168.1.100/?action=stream\" /><div id=\"panel\" style=\"color:#f2f2f2;height:100px;width:100%;\"></div></center><script>global.$ = require(\"jquery\");require(\"jquery-touch-events\");$(\"#panel\").bind(\"swipeup\", function(e){$.post(\"/\",{task:'f'});});$(\"#panel\").bind(\"swipedown\", function(e){$.post(\"/\",{task:'b'});});$(\"#panel\").bind(\"swipeleft\", function(e){$.post(\"/\",{task:'l'});});$(\"#panel\").bind(\"swiperight\", function(e){$.post(\"/\",{task:'r'});});$(\"#panel\").bind(\"swipeend\", function(e){$.post(\"/\",{task:'h'});});</script></body></html>";

PROGMEM const int l=7,r=6,f=5,b=4,h=3;

void setup() {
  Serial.begin(115200);
  Serial.println("yesss");
  pinMode(f, OUTPUT);
  pinMode(b, OUTPUT);
  pinMode(l, OUTPUT);
  pinMode(r, OUTPUT);
  pinMode(h, OUTPUT);
  digitalWrite(f, LOW);
  digitalWrite(b, LOW);
  digitalWrite(l, LOW);
  digitalWrite(r, LOW);
  digitalWrite(h, LOW);
  Serial.println(WiFi.begin(ssid, pass));
  delay(2000);
  Serial.println(WiFi.status());
  Serial.println(mdns.begin("sayak", WiFi.localIP()));
  server.on("/", [](){
    if(server.hasArg("task"))
    {
      String task=server.arg("task");
      digitalWrite(f, LOW);
      digitalWrite(b, LOW);
      digitalWrite(l, LOW);
      digitalWrite(r, LOW);
      digitalWrite(h, LOW);
      if(task=="h") digitalWrite(h,HIGH);
      else if(task=="f") digitalWrite(f,HIGH);
      else if(task=="b") digitalWrite(b,HIGH);
      else if(task=="l") digitalWrite(l,HIGH);
      else if(task=="r") digitalWrite(r,HIGH);
      return;
    }
    server.send(200, "text/html", webPage);
  });
  server.begin();
  Serial.println("yesss");
}

void loop() {
  server.handleClient();
}
```

sayak-brm, 2016-11-11:
Works  after changing `PROGMEM const int l=7,r=6,f=5,b=4,h=3;` to `PROGMEM const int l=D4,r=D3,f=D2,b=D1,h=D0;`

john3exonets, 2016-12-18:
And for those of us who are less psychic, where would I go to change this PROGMEM const int??

JZ-SmartThings, 2016-12-20:
Ditto, would be nice to know exactly where to change PROGMEM. Few things I already searched, my Arduino IDE folder and this location where the IDE places upgraded libs: C:\Users\myuser\Documents\Arduino 

I also found PROGMEM referenced in ArduinoJson lib which seems to say that PROGMEM is only available in AVR architecture which contradicts what we have with the ESP8266 (non-AVR):
**// PROGMEM is only available on AVR architecture**

[My project (SmartThings generic HTTP driver/Device-Handler controlling 2 relays & more)](https://community.smartthings.com/t/raspberry-pi-with-php-or-arduino-esp8266-nodemcu-to-gpio-to-relay-to-gate-garage-trigger/43335) is much larger than the few concise samples posted above but **this error is killing me and made me put in reboot logic, occurring every 8 hours;** pathetic solution... posting that reboot sample below for others to reference but it's a crappy workaround and **we'd love to get this figured out, please elaborate on the proposed fix.** 

> //SELF-STANDING FUNCTION AT THE TOP OF THE SKETCH OUTSIDE OF SETUP/LOOP
> void(* resetFunction) (void) = 0;

 
> //AT TOP OF THE LOOP --- RESET EVERY 8 HOURS
>   if (millis () >= 28800000) {
>     resetFunction();
>   }

hemangjoshi37a, 2017-05-31:
What is the meaning of this "boot mode:(3,6)"??
There are also other types of boot modes, what do they mean??
Please provide any table that has all possible values and meanings of boot modes.
Please help. Thank you....

lrmoreno007, 2017-05-31:
Read the documentation before asking something already explained or do it at www.esp8266.com

http://arduino-esp8266.readthedocs.io/en/latest/boards.html#boot-messages-and-modes

hemangjoshi37a, 2017-05-31:
@lrmoreno007  thax yo... That is what I was looking for my whole day... You saved my day...

akashash, 2017-06-27:
Hi, I am also getting same isssue but only while using "Wifi.config()" to set static IP. If I am not using this...my code is working fine. But with this...I am getting this issue. Can anyone please help??

Kondenado, 2017-08-21:
Hi

I am getting the same problem in a D1 mini (ESP8266 based board).

"ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
"

When i go the webpage to know which is the problem here, i find: That reset cause 4 is watchdog reset, 

and the modes are: 

Number | GPIO15 | GPIO0 | GPIO2 | Mode
-- | -- | -- | -- | --
0 | 0V | 0V | 0V | Not valid
1 | 0V | 0V | 3.3V | Uart
2 | 0V | 3.3V | 0V | Not valid
3 | 0V | 3.3V | 3.3V | Flash
4 | 3.3V | 0V | 0V | SDIO
5 | 3.3V | 0V | 3.3V | SDIO
6 | 3.3V | 3.3V | 0V | SDIO
7 | 3.3V | 3.3V | 3.3V | SDIO

So my personal guess is that my D1 mini is in " flashing mode" right? So the issue is to put it in the "running mode" right? Any suggestion on where to look to do that? (I am not very proefficient in this things ..)

devyte, 2017-10-18:
This looks like user error due to use of wrong pins, or use of 3rd party lib that wasn't supported for the ESP.
Closing per #3655 .

closed 2017-10-18 by devyte

junwoo091400, 2017-10-22:
Absolutely, after the same error, my code worked perfectly, without any need for better POWER-SUPPLY....
it was just a **'Wrong, Unsupported 3rd party library**' !!!!!!

StarsoftAnalysis, 2017-12-10:
Ah, yes.  It was wrong pin numbers in my case.  Thanks.

ABHI072, 2018-04-05:
Issue solved:
ets Jan 8 2013,rst cause:4, boot mode:(3,6)
wdt reset.

Hardware used: ESP-12F

Debug Messages:

ets Jan 8 2013,rst cause:4, boot mode:(3,6)
wdt reset
load 0x4010f000, len 1264, room 16
tail 0
chksum 0x42
csum 0x42
~ld

Reason(in my case):
pins RST(reset), EN(enable) and VCC were facing fluctuations in power supplied irrespective of the constant power supply. This was causing a rst of no. 4, i.e. Hardware WDT reset.

Solution:
Add a 47uF capacitor at each of the pin mentioned wrt GND.

koltevinod1, 2018-06-29:
i used 47uf capacitor to RST,EN and VCC wrt GND but still having program,
IC esp12e is flashing properly but still not working.
after some time serial monitor shows following error.

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld

so what is exact problem and how to solve it.

junwoo091400, 2018-06-29:
@koltevinod1 
I actually had a code that did THIS:
--------------------------------------------------
typedef struct
{
int a;
int b;
}AB_struct

AB_struct* ab = NULL;//AB_struct pointer. Init to NULL(0)

ab -> a = 23;//Tries to put data into address 0!! => wdt_reset();
ab -> b = 45;//Same as above.
--------------------------------------------------
In ESP8266, if you access the area around pointer NULL, the program goes into watchdog reset. This was hard to find but when I fixed that Wrong address access, the problem was solved.
--------------------------------------------------
Conclusion : Watch out for your mistake in the software where you access NULL pointer as a valid pointer on ESP8266.

koltevinod1, 2018-06-30:
Thank you for your reply..
fist thing is i am using arduino IDE to build and upload a program for esp12e.
but i am trying to upload any of a code then what happen is,
1. its uploaded successfully up to 100% 
2. nothing is happen then for some seconds i.e. program is also not running.
3. after some time  

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld

this error is occur on serial monitor.

and between some it repeatedly come,.

A am not able to find out what exactly happening, What wrong with it???
Please give me a solution.

pashinen, 2018-08-05:
I am facing the same problem like below--
ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld

if I upload normal blink program it works fine but when I upload minimum bare program to access the AT commands I get above error. And also this error comes only in baud rate of 74880 which I didn't configure anywhere before.

Kindly let me know the reasons and solution for above error.

Kondenado, 2018-08-05:
Once I had a similar problems, not sure if the same.

it turned out that I was not declaring properly the pins. For instance, PIN 1 in the arduino IDE code, is not PIN 1 in the D1 mini. I just looked the pinmap change the pins and it worked OK.

frustrat01, 2018-12-09:
I am facing the similar issue with the ESP8266-01 as the following below:

chksum 0x42
csum 0x42
~ld
 ets Jan  8 2013,rst cause:4, boot mode:(1,0)
wdt reset

By the way, I have tried using the WiFiClient example to upload the dummy data to the PHP localhost and it works. But it cannot work with uploading the ultrasonic sensor data to the PHP localhost.
Anyone has the solution please help me to get rid of this issue ASAP. Thanks

dcolley, 2019-05-31:
@Kondenado thanks, I also moved from Uno to D1. Your suggestion solved my issue!

msalik42, 2019-06-10:
> Any solution ? tutorial ?

Maybe you are not using the right version of the Library.

dcolley, 2019-06-10:
@msalik42 you should read the comments from @junwoo091400 

"**In ESP8266, if you access [...] a null pointer [...], the program goes into watchdog reset**."

I would recommend you double-check all variables and pin numbers, make sure they have been initialised and have correct values.

ffbboy30, 2020-01-15:
I'm facing the same problem of many people with the reboot of the esp8266 board.
I'm using the arduino lib as it is , why it is not working directly ? on the 8266 arduino file there is more the 500 structure which one is wrong .
I'm using ESP-01
Thanks

chimeranzl, 2020-05-25:
rst cause:4, boot mode:(3,6) also happens if you define the pins incorrectly. For example, I was using a Wemos D1 Mini and specified constants to define pins 6 and 7, whereas I should have put in D6 and D7

beshur, 2020-06-11:
I have the same issue, but with no externals attached, on a BasicHttpClient example, that is below.

Board is Wemos D1 mini lite, nothing is connected. Board is powered via USB from computer.

Blink example is fine though. Also it worked before on the same board.

Serial log:
```
[SETUP] WAIT 4...
[SETUP] WAIT 3...
[SETUP] WAIT 2...
[SETUP] WAIT 1...

ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
va5432625
~ld

SDK:2.2.2-dev(38a443e)/Core:2.7.1=20701000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-30-g92add50/BearSSL:5c771be

[SETUP] WAIT 4...
```

Code:

```
/**
   BasicHTTPClient.ino

    Created on: 24.05.2015

*/

#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <ESP8266HTTPClient.h>

#include <WiFiClient.h>

ESP8266WiFiMulti WiFiMulti;

void setup() {

  Serial.begin(115200);
  // Serial.setDebugOutput(true);

  Serial.println();
  Serial.println();
  Serial.println();

  for (uint8_t t = 4; t > 0; t--) {
    Serial.printf("[SETUP] WAIT %d...\n", t);
    Serial.flush();
    delay(1000);
  }

  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP("SSID", "PASSWORD");

}

void loop() {
  // wait for WiFi connection
  if ((WiFiMulti.run() == WL_CONNECTED)) {

    WiFiClient client;

    HTTPClient http;

    Serial.print("[HTTP] begin...\n");
    if (http.begin(client, "http://jigsaw.w3.org/HTTP/connection.html")) {  // HTTP

      Serial.print("[HTTP] GET...\n");
      // start connection and send HTTP header
      int httpCode = http.GET();

      // httpCode will be negative on error
      if (httpCode > 0) {
        // HTTP header has been send and Server response header has been handled
        Serial.printf("[HTTP] GET... code: %d\n", httpCode);

        // file found at server
        if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY) {
          String payload = http.getString();
          Serial.println(payload);
        }
      } else {
        Serial.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
      }

      http.end();
    } else {
      Serial.printf("[HTTP} Unable to connect\n");
    }
  }

  delay(10000);
}
```

chimeranzl, 2020-06-11:
> I have the same issue, but with no externals attached, on a BasicHttpClient example, that is below.
>
> Board is Wemos D1 mini lite, nothing is connected. Board is powered via USB from computer.
>
> Blink [...]

Seems to be caused by your wireless multi definition - not sure why there are numerous wifi headers included.  Try getting rid of **#include <ESP8266WiFiMulti.h>** and just use **#include <ESP8266Wifi.h>**

Also get rid of the wifi declare **ESP8266WiFiMulti WiFiMulti;** and change to **WiFiClient wifiClient;** instead and use that in your code. 

I usually create a function to start Wifi and call it from the setup or loop.  Eg like this...

```
//
// Starts WIFI connection
//
void startWIFI()
{

  // If we are not connected
  if (WiFi.status() != WL_CONNECTED)
  {
    int iTries;
    iTries = 0;
    digitalWrite(LED_BUILTIN, LOW);
    Serial.println("Starting WIFI connection");
    WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    WiFi.begin(ssid, password);

    // If not WiFi connected, retry every 3 seconds for 1 minute
    while (WiFi.status() != WL_CONNECTED)
    {
      iTries++;
      Serial.print(".");
      delay(3000);

      // If can't get to Wifi after 1 minute, reboot ESP
      if (iTries > 20)
      {
        Serial.println("TOO MANY WIFI ATTEMPTS, REBOOTING!");
        ESP.reset();
      }
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println(WiFi.localIP());

    // Turn off LED
    digitalWrite(LED_BUILTIN, HIGH);

    // Let network have a chance to start up
    delay(500);

  }

}
```

beshur, 2020-06-11:
> Seems to be caused by your wireless multi definition - not sure why there are numerous wifi headers included. Try getting rid of **#include <ESP8266WiFiMulti.h>** and just use **#include [...]

This is from the ESP8266 library example.

> I usually create a function to start Wifi and call it from the setup or loop. Eg like this...

Thanks.

Anyway, I tried your suggestions and nothing changed.

chimeranzl, 2020-06-11:
> This is from the ESP8266 library example.
>
>
> Thanks.
>
> Anyway, I tried your suggestions and nothing changed.

Do you have the latest libraries installed and updated?

Have you tried another Wemos D1?  I've been delivered one in the past that just failed like this all the time, regardless of the code I had on it.

beshur, 2020-06-12:
Thanks for reply, @chimeranzl 

> Do you have the latest libraries installed and updated?

Yes

> Have you tried another Wemos D1? I've been delivered one in the past that just failed like this all the time, regardless of the code I had on it. Although, if Blink works on it and this code doesn't, then that kind of points to the code or library.

Yes, my other Wemos D1 works fine, but this one was also okay just a couple weeks ago, and I did only sketch changes.

beshur, 2020-06-12:
Fixed my issue with:

Reduced the main delay to 3000 in `loop`, dropped wifi multi, and I was getting a new error:

> Fatal exception 3(LoadStoreErrorCause)

I did some more searching, and found [this thread](https://www.esp8266.com/viewtopic.php?p=72760) that suggests erasing flash.

I changed the Erase Flash in Aduino to "All Flash Contents" and it helped.

blaskovicz, 2020-06-25:
My exception was happening because I was hitting an infinite recursion case...

jayanta-banik, 2020-10-26:
this is probably a fluke, but my error was resolved when i used D6, instead of 6 as reference to pin number,
I am using node-mcu. Hope this helps someone.

rTiagoS, 2020-11-02:
> Absolutely, after the same error, my code worked perfectly, without any need for better POWER-SUPPLY....
> it was just a **'Wrong, Unsupported 3rd party library**' !!!!!!

That's correct!!!! I had the same issue. I was using #include <WiFi.h> library instead <ESP8266WiFi.h>. So after chaging that, i was able to connect to the Wireless Network and broker without the watchdog reset and all those stuffs. Thanks!

Nardjis, 2020-11-19:
I changed the values ​​of the variables and everything worked fine

firtinahg, 2021-01-08:
This problem  happened to me , cause is 'PROGMEM' usage. If you are change the PROGMEM values to another type e.g. String you can manage the wdt reset.

eliazgheib, 2021-04-04:
I am having the same problem. I tried to upload a blank sketch and I still receiving the following on the serial monitor:
ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

jayanta-banik, 2021-04-04:
Can you try uploading blink example. And try experimenting with board
select ...nodemcu12E and the other node mcu12 also try generic one.

renanlido, 2021-07-01:
Hey guys, i found this code that explains everything. It's in Portuguese but it's worth taking a look. https://portal.vidadesilicio.com.br/watchdog-esp8266/

Eronrhcp, 2021-12-27:
> Works after changing `PROGMEM const int l=7,r=6,f=5,b=4,h=3;` to `PROGMEM const int l=D4,r=D3,f=D2,b=D1,h=D0;`

Obrigado amigo, para mim funcionou!

feraldruid22, 2021-12-29:
It's cause you use pin6 as input lol. Don't do that and problem solved.

hneel, 2022-02-14:
I got the error when I wanted to send a `F("string")` instead of a `"string"`.
It didn't give a problem with an Arduino Uno or an ESP32.

pedator92, 2022-02-19:
simple fix, use the GPIO Numbers, not the D7 or so.
work fine

wasimghafar-15530, 2022-03-31:
I was facing this problem with JSON. 
In my case size of the payload and JSON document was not suitable, I changed the size and fix it.

vyk2rr, 2022-04-09:
> That's correct!!!! I had the same issue. I was using #include <WiFi.h> library instead <ESP8266WiFi.h>. So after chaging that, i was able to connect to the Wireless Network and broker without the [...]

4 years to get the right answer 😄

kcchang61, 2022-06-30:
If you are using ESP8266, pls include <ESP8266WiFi.h> instead of <WiFi.h>.