esp8266/Arduino #1810: Watchdog reset when a client is connecting to ESP8266

Codebreaker101, 2016-03-23:
When i load unchanged WiFiAccessPoint example on to my ESP-12E and try to connect to it I get this printed to the console:
```
Configuring access point...[AP] softap config unchanged
AP IP address: 192.168.4.1
HTTP server started
wifi evt: 7
wifi evt: 7
wifi evt: 7
add 1
aid 1
station: f4:9f:f3:71:22:27 join, AID = 1
wifi evt: 5
wifi evt: 7
wifi evt: 7
wifi evt: 7

 ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Configuring access point...[AP] softap config unchanged
AP IP address: 192.168.4.1
HTTP server started
wifi evt: 7
wifi evt: 7

add 1
aid 1
station: f4:9f:f3:71:22:27 join, AID = 1
wifi evt: 5
Fatal exception 0(IllegalInstructionCause):
epc1=0x40225d7c, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

Exception (0):
epc1=0x40225d7c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffdf0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff90:  00000000 3fffdad0 3fff0278 40205b03  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Configuring access point...[AP] softap config unchanged
AP IP address: 192.168.4.1
HTTP server started

 ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Configuring access point...[AP] softap config unchanged
AP IP address: 192.168.4.1
HTTP server started
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
```
After a couple of attempts it finally connects and works for the time I've tested it. The stack trace pointed to the _loop_task_ at core_esp8266_main.cpp line 43. So there's something that is taking too long to execute. Is there a solution for this?

igrr, 2016-03-23:
Could you please attach the stack trace? Loop_task is the bottom of the
stack, you need to look at the top.

Codebreaker101, 2016-03-23:
Got a larger stack on the repeated tries. 
Also, here is the full console log and the .elf file https://dl.dropboxusercontent.com/u/11218705/github/report.zip

```
0x4010576b: wdt_feed at ?? line ?
0x4010574c: wdt_feed at ?? line ?
0x40105762: wdt_feed at ?? line ?
0x40105762: wdt_feed at ?? line ?
0x40224d20: user_uart_wait_tx_fifo_empty at ?? line ?
0x40224d20: user_uart_wait_tx_fifo_empty at ?? line ?
0x40234650: inet_chksum_pbuf at ?? line ?
0x4010574c: wdt_feed at ?? line ?
0x4010574c: wdt_feed at ?? line ?
0x4010576b: wdt_feed at ?? line ?
0x4010574c: wdt_feed at ?? line ?
0x40105762: wdt_feed at ?? line ?
0x40224d20: user_uart_wait_tx_fifo_empty at ?? line ?
0x40224d20: user_uart_wait_tx_fifo_empty at ?? line ?
0x40234650: inet_chksum_pbuf at ?? line ?
0x4010574c: wdt_feed at ?? line ?
0x40105765: wdt_feed at ?? line ?
0x40105762: wdt_feed at ?? line ?
0x40224d08: user_uart_wait_tx_fifo_empty at ?? line ?
0x4010576b: wdt_feed at ?? line ?
0x4010576b: wdt_feed at ?? line ?
0x4010574c: wdt_feed at ?? line ?
0x40105762: wdt_feed at ?? line ?
0x4010574c: wdt_feed at ?? line ?
0x40224d20: user_uart_wait_tx_fifo_empty at ?? line ?
0x40224d20: user_uart_wait_tx_fifo_empty at ?? line ?
0x40234650: inet_chksum_pbuf at ?? line ?
0x4010574c: wdt_feed at ?? line ?
0x40105765: wdt_feed at ?? line ?
0x40105762: wdt_feed at ?? line ?
0x40224d08: user_uart_wait_tx_fifo_empty at ?? line ?
0x4010576b: wdt_feed at ?? line ?
0x4010574c: wdt_feed at ?? line ?
0x4010576b: wdt_feed at ?? line ?
0x4010574c: wdt_feed at ?? line ?
0x40105762: wdt_feed at ?? line ?
0x40105762: wdt_feed at ?? line ?
0x40224d20: user_uart_wait_tx_fifo_empty at ?? line ?
0x40224d20: user_uart_wait_tx_fifo_empty at ?? line ?
0x40234650: inet_chksum_pbuf at ?? line ?
0x4010574c: wdt_feed at ?? line ?
0x40105765: wdt_feed at ?? line ?
0x40105762: wdt_feed at ?? line ?
0x40224d08: user_uart_wait_tx_fifo_empty at ?? line ?
0x4010576b: wdt_feed at ?? line ?
0x4010574c: wdt_feed at ?? line ?
0x4010576b: wdt_feed at ?? line ?
0x4010574c: wdt_feed at ?? line ?
0x40105762: wdt_feed at ?? line ?
0x40105762: wdt_feed at ?? line ?
0x40224d20: user_uart_wait_tx_fifo_empty at ?? line ?
0x40224d20: user_uart_wait_tx_fifo_empty at ?? line ?
0x40234650: inet_chksum_pbuf at ?? line ?
0x4010574c: wdt_feed at ?? line ?
0x4010576b: wdt_feed at ?? line ?
0x4010574c: wdt_feed at ?? line ?
0x4010576b: wdt_feed at ?? line ?
0x4010574c: wdt_feed at ?? line ?
0x40105762: wdt_feed at ?? line ?
0x40105762: wdt_feed at ?? line ?
0x40224d20: user_uart_wait_tx_fifo_empty at ?? line ?
0x40224d20: user_uart_wait_tx_fifo_empty at ?? line ?
0x40234650: inet_chksum_pbuf at ?? line ?
0x4010574c: wdt_feed at ?? line ?
0x40225712: ets_vsnprintf at ?? line ?
0x40216de3: ieee80211_add_ie_vendor_esp_head at ?? line ?
[... 51 lines cut ...]
0x402198da: etharp_tmr at ?? line ?
0x40102c14: trc_NeedRTS at ?? line ?
0x40103fbc: lmacProcessTXStartData at ?? line ?
0x40103fb9: lmacProcessTXStartData at ?? line ?
0x4010576b: wdt_feed at ?? line ?
0x4010574c: wdt_feed at ?? line ?
0x40105762: wdt_feed at ?? line ?
0x40102dec: wDev_ProcessFiq at ?? line ?
0x4021b37f: pbuf_header at ?? line ?
0x40219fe8: ethernet_input at ?? line ?
0x401006d8: malloc at C:\Users\Antonio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266\umm_malloc/umm_malloc.c line 1658
0x40107208: pvPortMalloc at C:\Users\Antonio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/heap.c line 13
0x402100f0: ieee80211_decap at ?? line ?
0x4021b211: pbuf_alloc at ?? line ?
0x4020e200: ieee80211_hostap_attach at ?? line ?
0x4020e216: ieee80211_hostap_attach at ?? line ?
0x4020e70a: hostap_input at ?? line ?
0x4010020c: _umm_free at C:\Users\Antonio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266\umm_malloc/umm_malloc.c line 1281
0x40219988: etharp_tmr at ?? line ?
0x40219938: etharp_tmr at ?? line ?
0x40219a47: etharp_find_addr at ?? line ?
0x40219fe1: ethernet_input at ?? line ?
0x402259e3: ets_snprintf at ?? line ?
0x40205b03: loop_task at C:\Users\Antonio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/core_esp8266_main.cpp line 43
```

rallicknom, 2016-03-31:
I am getting a similar error for the latest version 2.1.0

This is what I get in serial monitor:

---------------------------------------------------------------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

---------------------------------------------------------------------

devyte, 2017-10-17:
I can't reproduce this issue with the WiFiAccessPoint with latest git. Closing due to can't reproduce.

closed 2017-10-17 by devyte