esp8266/Arduino #1815: With v2.2.0 core, can no longer run SPI and I2C together

bHogan2, 2016-03-25:
### WDT error under circumstances described below.
#### Hardware
Hardware:		ESP-12
Core Version:      	2.2.0 stable

### Description      

I'm using an I2C OLED and an SPI SD card. Under 2.2.1 I could run both together - as long as I started SPI before I2C. (A workaround) If I didn't, I could still read the card, but if there was NO CARD, the ESP would hang with a WDT error.

With 2.2.0 it doesn't matter which I do first, I'll always get the WDT hang.
Not sure why - but wanted to call the difference to your attention in case it rings a bell..
### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512k
CPU Frequency:	80Mhz
Flash Mode:		DIO
Flash Frequency:	80Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch
pseudocode of a big sketch . . .

void setup() {
EEPROM.begin(4096);  // FWIW
SPI.begin()

if (!SD.begin(SD_SELECT)) {           // triggers the WDT error if no card
    Serial.println("No Card");
    return NO_CARD;             
}
SPI.end();                            //done with SPI so close it so pins can be reused
///////////////////////////////////
Wire.pins(SDA_PIN, SCL_PIN);          // define pins used for I2C
  Wire.begin();                                     // HAD TO DO THIS AFTER SPI else  "no card WDT reset"
}

void loop() {
}
```
#define SDA_PIN         0               // I2C SDA pin (pulled up) + PGM button and LED
#define SCL_PIN         2               // I2C SCL pin (pulled up)
#define MISO_PIN     12               // (when used) MISO (DO on SD)   
#define MOSI_PIN      13              // MOSI (DI on SD)
#define SCLK_PIN      14               // SCLK (SCK on SD)
#define CS_PIN           15              // SS (CS on SD)- always LOW that's OK - not used
// WAKE                16               // wired to RESET through cap to wake from sleep

### Debug Messages

WDT timeout

pipi61, 2016-03-25:
Hi!
Try in c:\Arduino167\hardware\esp8266com\esp8266\libraries\SD\src\utility\
in Sd2Card::init function:
.....
 while ((status_ = cardCommand(CMD0, 0)) != R1_IDLE_STATE) {
    if (((uint16_t)(millis() - t0)) > SD_INIT_TIMEOUT) {
      error(SD_CARD_ERROR_CMD0);
      goto fail;
    }
  }
.....
change to:
 while ((status_ = cardCommand(CMD0, 0)) != R1_IDLE_STATE)  
  {
    if (((uint16_t)millis() - t0) > SD_INIT_TIMEOUT) {
      error(SD_CARD_ERROR_CMD0);
      goto fail;
    }
    yield();
  }

pipi61, 2016-03-25:
Hi!
Try in c:\Arduino167\hardware\esp8266com\esp8266\libraries\SD\src\utility\Sd2Card.cpp
in Sd2Card::init function:
.....
while ((status_ = cardCommand(CMD0, 0)) != R1_IDLE_STATE) {
if (((uint16_t)(millis() - t0)) > SD_INIT_TIMEOUT) {
error(SD_CARD_ERROR_CMD0);
goto fail;
}
}
.....
change to:
while ((status_ = cardCommand(CMD0, 0)) != R1_IDLE_STATE)

{
if (((uint16_t)millis() - t0) > SD_INIT_TIMEOUT) {
error(SD_CARD_ERROR_CMD0);
goto fail;
}
yield();
}

bHogan2, 2016-03-25:
Thank you. Appreciated. Makes sense. Made change but 2.0.0 still needs SPI init before I2C init (for some reason).
Haven't tried going back to 2.1.0.
In any case, you mod should probable be included in the next lib - IMHO.

igrr, 2016-06-13:
Fixed in 9172033

closed 2016-06-23 by igrr