esp8266/Arduino #1822: ESP-01 restarting when WiFi connection is lost

jorgegarciadev, 2016-03-27:
### Basic Infos

#### Hardware
Hardware:		ESP-01
Core Version:      	2.1.0

### Description

The module restarts when the WiFi connection is lost (tested disabling WiFi on router and restarting the router?. I don't know if this is a desired behaviour, and if it is. I would like to know if there is a way to disable it.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1MB?
CPU Frequency:	80Mhz
Flash Mode:		DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
// #include <ESP8266mDNS.h>
 
const char* ssid = "...";
const char* password = "...";

IPAddress ip(192,168,1,30);
IPAddress gateway(192,168,1,1);
IPAddress subnet(255,255,255,0);

const char* response = "{\"status\": %d, \"code\": %d}";

char buffer[26];
// MDNSResponder mdns;

ESP8266WebServer server(80);

const int R1 = 0;
const int R2 = 2;

void handleNotFound(){
  String message = "Recurso no encontrado\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMetodo: ";
  message += (server.method() == HTTP_GET)?"GET":"POST";
  message += "\nArgumentos: ";
  message += server.args();
  message += "\n";
  for (uint8_t i=0; i<server.args(); i++){
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
}

void handleRelay(int relay, int level) {
  digitalWrite(relay, level);
  sprintf(buffer, response, level , 200);
  server.send(200, "application/json", buffer);
}

void handleErr() {
  sprintf(buffer, response, -1, 401);
  server.send(401, "application/json", buffer);
}

void statusRelay(int relay){
  int status;
  status = digitalRead(relay);
  sprintf(buffer, response, status, 200);
  server.send(200, "application/json", buffer);
}
 
void setup(void){
  digitalWrite(R1, HIGH);
  digitalWrite(R2, HIGH);
  pinMode(R1, OUTPUT);
  pinMode(R2, OUTPUT);
  
  Serial.begin(115200);
  
  server.on("/encender", [](){
    int relay = server.arg("rele").toInt();
    (relay == 1) ? handleRelay(R1, LOW): (relay == 2) ? handleRelay(R2, LOW) : handleErr();
    Serial.println(buffer);
  });

  server.on("/apagar", [](){
    int relay = server.arg("rele").toInt();
    (relay == 1) ? handleRelay(R1, HIGH): (relay == 2) ? handleRelay(R2, HIGH) : handleErr();
    Serial.println(buffer);
  });

  server.on("/estado", [](){
    int relay = server.arg("rele").toInt();
    (relay == 1) ? statusRelay(R1): (relay == 2) ? statusRelay(R2) : handleErr();
    Serial.println(buffer);
  });

  server.onNotFound(handleNotFound);
  
  server.begin();
  Serial.println("Servidor HTTP iniciado.");
}
 
void loop(void){
  if (WiFi.status() != WL_CONNECTED){
    WiFi.begin(ssid, password);
    WiFi.config(ip, gateway, subnet);
    if (WiFi.waitForConnectResult() != WL_CONNECTED){
      return;
    }
    Serial.println("");
    Serial.print("Conectado a ");
    Serial.println(ssid);
    Serial.print("Direccion IP: ");
    Serial.println(WiFi.localIP());
    
    // if (mdns.begin("verja", WiFi.localIP())) {
    //   Serial.println("MDNS iniciado.");
    // }
  }
  
  server.handleClient();
}

hakeemta, 2016-03-27:
I am encountering similar issue.

Pls does it give the following exceptions:
Exception (2):
epc1=0x3ffe9614 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffe9614 depc=0x00000000

ctx: sys
sp: 3ffffc90 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe30:  401046a6 00040000 7fffffff 00000000
3ffffe40:  00000022 40103fb9 00040000 00000000
3ffffe50:  4010418a 0000007f 7fffffff 00000000
3ffffe60:  0000007f 40103fb9 00040000 4000050c
3ffffe70:  401030ef 00080000 3fffc200 00000022
3ffffe80:  00000000 00000004 3ffffee0 3ffffef1
3ffffe90:  40107020 00000000 3fff3860 000e7e43
3ffffea0:  4010549c 00000000 000000a5 00000000
3ffffeb0:  40000f68 00000000 00000000 3ffffe40
3ffffec0:  40000f58 00000000 00000020 00000000
3ffffed0:  3ffffef0 3ffffee0 00000004 00000000 
3ffffee0:  ffffffff 3ffe972c 3fff3860 000000e5 
3ffffef0:  000000a5 00003a98 3ffef648 00000030 
3fffff00:  00000000 00000000 00000020 4010031d 
3fffff10:  3ffe9d90 00001d40 3fff0a20 40100426 
3fffff20:  3ffe9bc4 40105c27 3ffee7c8 40100537 
3fffff30:  00000000 3ffef344 3fff1b94 401008c1 
3fffff40:  3ffef620 3ffe9bc0 00000000 40107218 
3fffff50:  4020d20c 3ffe9bc4 3ffef620 402231b5 
3fffff60:  00000000 4022a2fd 3ffef5f8 3ffef620 
3fffff70:  0168d492 60000600 00000000 4022a2fd 
3fffff80:  4022a342 3fffdab0 00000000 3fffdcb0 
3fffff90:  3ffef638 3fffdad0 3fff0a2c 40209c87 
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49 
<<<stack<<<

ets Jan  8 2013,rst cause:2, boot mode:(1,6)

ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

gmag11, 2016-03-27:
You don't need to control WiFi reconnection. It is done automatically by ESP8266. You only have to add `if(WiFi.status()!=WL_CONNECTED)` not to execute connection dependant code. You shouldn't do `WiFi.begin()` inside `loop()` function.

jorgegarciadev, 2016-03-27:
@gmag11 Thanks, changed it but the problem persists. @hakeemta Just changed my setup to register the exception, will post results soon here.

jorgegarciadev, 2016-03-27:
Got these exceptions:

[20:57:27][recv] - Exception (0):
[20:57:27][recv] - epc1=0x4023b3de epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
[20:57:27][recv] -
[20:57:27][recv] - ctx: sys
[20:57:27][recv] - sp: 3ffffda0 end: 3fffffb0 offset: 01a0
[20:57:27][recv] -
[20:57:27][recv] - >>>stack>>>
[20:57:27][recv] - 3fffff40:  00000000 40220ce0 3fffff80 3fff1de4
[20:57:28][recv] - 3fffff50:  3ffe0000 00000056 00000000 00000018
[20:57:28][recv] - 3fffff60:  40221e2c 00000002 3ffebd5e 40220d9f
[20:57:28][recv] - 3fffff70:  3fff2e04 00000018 4022134a 3fff2e04
[20:57:28][recv] - 3fffff80:  3fff1de4 3fffdc80 3fff1e64 3fff0740
[20:57:28][recv] - 3fffff90:  4022ccff 3fff2e04 00000000 4020ba0f
[20:57:28][recv] - 3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49
[20:57:28][recv] - <<<stack<<<
[20:57:28][recv] -
[20:57:28][recv] - ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[20:57:28][recv] -
[20:57:29][recv] - load 0x4010f000, len 1264, room 16
[20:57:29][recv] - tail 0
[20:57:29][recv] - chksum 0x42
[20:57:29][recv] - csum 0x42

[20:57:37][recv] - Exception (0):
[20:57:37][recv] - epc1=0x4022c05f epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
[20:57:38][recv] -
[20:57:38][recv] - ctx: sys
[20:57:38][recv] - sp: 3ffffcc0 end: 3fffffb0 offset: 01a0
[20:57:38][recv] -
[20:57:38][recv] - >>>stack>>>
[20:57:38][recv] - 3ffffe60:  401057dd 401057da 00040000 00000001
[20:57:38][recv] - 3ffffe70:  400005e1 4010300e 53000000 4000050c
[20:57:38][recv] - 3ffffe80:  4022d081 00000033 0000001e 00000022
[20:57:38][recv] - 3ffffe90:  40105b10 0032957c 3ffef3b8 fffffffa
[20:57:39][recv] - 3ffffea0:  3ffede68 40225ec8 00003a98 00000000
[20:57:39][recv] - 3ffffeb0:  ffffffff 3fffc6fc 00000001 3ffef3b8
[20:57:39][recv] - 3ffffec0:  3ffef3e0 00a1f990 60000600 00000000
[20:57:39][recv] - 3ffffed0:  ffffffff 3fffc6fc 00000001 3fffdab0
[20:57:39][recv] - 3ffffee0:  00000000 3fffdad0 3fff0734 00000030
[20:57:39][recv] - 3ffffef0:  00000000 3fffdad0 3fff0734 00000030
[20:57:39][recv] - 3fffff00:  ffffffff 00000000 3ffe9ca1 00000008
[20:57:39][recv] - 3fffff10:  402110b2 3ffed428 00000000 3fffdc00
[20:57:39][recv] - 3fffff20:  3ffedee0 0000003c 40101f33 3ffed428
[20:57:40][recv] - 3fffff30:  000005e0 00000000 40103ee9 3ffedcd0
[20:57:40][recv] - 3fffff40:  0000003c 00000000 3ffed428 00000100
[20:57:40][recv] - 3fffff50:  0000001c 00000001 40105012 3ffedcd0
[20:57:40][recv] - 3fffff60:  3ffed428 4010197e 00000000 3ffed940
[20:57:40][recv] - 3fffff70:  3ffed428 4010174b 00000000 00000000
[20:57:40][recv] - 3fffff80:  4022d0d6 3fffdab0 00000000 3fffdcb0
[20:57:40][recv] - 3fffff90:  3ffef408 3fffdad0 3fff0734 4020ba0f
[20:57:40][recv] - 3fffffa0:  40000f49 40000f49 3fffdab0 40000f49
[20:57:40][recv] - <<<stack<<<
[20:57:40][recv] -
[20:57:41][recv] - ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[20:57:41][recv] -
[20:57:41][recv] - load 0x4010f000, len 1264, room 16
[20:57:41][recv] - tail 0
[20:57:41][recv] - chksum 0x42
[20:57:41][recv] - csum 0x42
[20:57:41][recv] - ~ld

jorgegarciadev, 2016-03-29:
I have been doing some changes on the circuit, replaced one of the caps on the power supply with one bigger and the ESP doesn't restart when losing connection anymore. Now, sometimes, resets parsing a mDNS package. Don't know if the problem is in the library or in the device sending the package. When I find the culprit or capture the package I will open a new issue. For now I'm disabling mDNS on the ESP and closing this issue.

Thanks you.

closed 2016-03-29 by jorgegarciadev