esp8266/Arduino #1824: Exception when using wifi_promiscuous_enable

mic159, 2016-03-28:
### Basic Infos

#### Hardware
Hardware:			ESP-07, But also can reproduce the same on Adafruit Huzzah (ESP-12)
Core Version:      	2.1.0

### Description

When using promiscuous mode, It crashes after a while (anywhere between 1-15 mins).
It usually has the same stack trace, which looks pretty messed up, is something messing up the stack?

Or am I doing something stupid?

### Settings in IDE

Module:  			ESP-07 (using 'Generic ESP8266 Module' board type)
Flash Size: 		4M/1M SPIFFS
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40MHZ
Upload Using:		serial
Reset Method:		ck

I can also repdocuce this on an Adafruit Huzzah

### Sketch

Reproducible with this minimal sketch:

```cpp
extern "C" {
#include <user_interface.h>
}

static void ICACHE_FLASH_ATTR handle_pkt(uint8_t* buf, uint16_t len) {
  
}

void setup() {
  Serial.begin(115200);
  Serial.println(ESP.getFlashChipId());
  delay(10);
  wifi_set_opmode(STATION_MODE);
  wifi_set_channel(1);
  wifi_set_promiscuous_rx_cb(handle_pkt);
  delay(10);
  wifi_promiscuous_enable(1);
}

void loop() {
  delay(5000);
  Serial.println("tick");
}
```

### Decoded stack trace

```
Decoding 44 results
0x40107020: printf at /home/michaelc/.arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/libc_replacements.c line 206
0x40105b39: ets_timer_disarm at ?? line ?
0x40105b39: ets_timer_disarm at ?? line ?
0x4010076f: malloc at /home/michaelc/.arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/umm_malloc/umm_malloc.c line 1655
0x401003b9: check_poison at /home/michaelc/.arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/umm_malloc/umm_malloc.c line 817
0x401004c2: check_poison_block at /home/michaelc/.arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/umm_malloc/umm_malloc.c line 851
0x401005d3: check_poison_all_blocks at /home/michaelc/.arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/umm_malloc/umm_malloc.c line 880
0x40103677: wDevDisableRx at ?? line ?
0x4010075e: malloc at /home/michaelc/.arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/umm_malloc/umm_malloc.c line 1642
0x40107208: pvPortMalloc at /home/michaelc/.arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/heap.c line 13
0x4010383e: wDevDisableRx at ?? line ?
0x40102ea5: wDev_ProcessFiq at ?? line ?
0x40102e3c: wDev_ProcessFiq at ?? line ?
0x40102dd0: wDev_ProcessFiq at ?? line ?
0x40102dd0: wDev_ProcessFiq at ?? line ?
0x4010073a: _umm_malloc at /home/michaelc/.arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/umm_malloc/umm_malloc.c line 1456
0x40101dae: pp_post2 at ?? line ?
0x40102ea5: wDev_ProcessFiq at ?? line ?
0x40102e22: wDev_ProcessFiq at ?? line ?
0x40105b39: ets_timer_disarm at ?? line ?
0x40105480: chm_get_current_channel at ?? line ?
0x4010076f: malloc at /home/michaelc/.arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/umm_malloc/umm_malloc.c line 1655
0x40107208: pvPortMalloc at /home/michaelc/.arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/heap.c line 13
0x4021a25a: sys_timeout at ?? line ?
0x40105480: chm_get_current_channel at ?? line ?
0x40105499: chm_get_current_channel at ?? line ?
0x4021a394: sys_check_timeouts at ?? line ?
0x4021a38d: sys_check_timeouts at ?? line ?
0x40222795: ets_timer_handler_isr at ?? line ?
0x40107218: vPortFree at /home/michaelc/.arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/heap.c line 18
0x402227da: ets_timer_handler_isr at ?? line ?
0x4020249f: loop_task at /home/michaelc/.arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/core_esp8266_main.cpp line 43
```

### Raw stack trace

```

Exception (2):
epc1=0x3ffe8682 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffe8680 depc=0x00000000

ctx: sys 
sp: 3ffffad0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffc70:  ffffffff ffffffff ffffffff ffffffff  
3ffffc80:  ffffffff ffffffff ffffffff ffffffff  
3ffffc90:  ffffffff ffffffff ffffffff ffffffff  
3ffffca0:  ffffffff ffffffff ffffffff ffffffff  
3ffffcb0:  ffffffff ffffffff ffffffff ffffffff  
3ffffcc0:  ffffffff 00000004 3ffffd20 3ffffd31  
3ffffcd0:  40107020 00000000 3ffefe8c 00000000  
3ffffce0:  3ffe9390 ffffffff 000000a5 ffffffff  
3ffffcf0:  00000000 00000000 00000000 3ffffc80  
3ffffd00:  4000050c 00000000 0000001f 40105b39  
3ffffd10:  3ffffd30 3ffffd20 00000004 40105b39  
3ffffd20:  4000050c 3ffe877a 3ffefe8c 00000000  
3ffffd30:  000000a5 00000001 0000001f fffffffe  
3ffffd40:  4010076f 00000000 0000000c 401003b9  
3ffffd50:  00000000 000000c0 3ffefcf4 401004c2  
3ffffd60:  4000050c 00000000 0000001f 401005d3  
3ffffd70:  40103677 00000001 00000080 4010075e  
3ffffd80:  0df9525b 00000004 3ffeacc0 40107208  
3ffffd90:  00000006 3fff0328 00000000 4010383e  
3ffffda0:  ffffffff 3fffc6fc 60000600 3ff20a00  
3ffffdb0:  00000001 40102ea5 00000104 7fffffff  
3ffffdc0:  3ffe9390 3ffe9390 7fffffff 7fffffff  
3ffffdd0:  40102e3c 00000004 00000000 4000050c  
3ffffde0:  3fffc278 40102dd0 3fffc200 00000022  
3ffffdf0:  3ffffe00 40102dd0 3fffc200 00000022  
3ffffe00:  40000f83 00000030 0000001a ffffffff  
3ffffe10:  4010073a 00000023 00000000 000000c4  
3ffffe20:  00000000 0000001c 3ffefd08 00000020  
3ffffe30:  00000020 3fffc6fc 60000600 3ffefde8  
3ffffe40:  00000004 000000c0 3ffefcf4 00000030  
3ffffe50:  00000009 3ffefdf4 00000020 40101dae  
3ffffe60:  00000001 40102ea5 00000104 7fffffff  
3ffffe70:  3ffe9384 40102e22 00000008 7fffffff  
3ffffe80:  000000a5 000000a5 00000000 4000050c  
3ffffe90:  00000000 00000000 0000001f 40105b39  
3ffffea0:  4000050c 3fffdcc0 3ffe8b50 00000030  
3ffffeb0:  40000f68 00000030 00000011 ffffffff  
3ffffec0:  40000f58 00000000 00000020 00000000  
3ffffed0:  000000c4 3ffefde8 00000000 7fffffff  
3ffffee0:  ffffffff 3fffc6fc 60000600 3fffdab0  
3ffffef0:  00000000 3fffdcc0 3ffe8c10 00000030  
3fffff00:  0000001c ffffffff ffffffff 40105480  
3fffff10:  0100ad72 3ffee094 0000001c 4010076f  
3fffff20:  3ffefcf4 00000018 00000064 40107208  
3fffff30:  0100ad72 3ffee094 3ffefdcc 4021a25a  
3fffff40:  40105480 00000000 40105499 00000000  
3fffff50:  4021a394 000000c4 000000c4 4021a38d  
3fffff60:  00000000 40222795 3ffee378 3ffee3a0  
3fffff70:  0dfcc140 60000600 3fffdab0 40107218  
3fffff80:  402227da 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffee3c0 3fffdab0 3ffeebb0 4020249f  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

```

waltertakens, 2016-04-03:
same problem here on a WEMOS D1

Exception (2):
epc1=0x3ffe83f4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffe83f4 depc=0x00000000

ctx: sys 
sp: 3ffffad0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffc70:  4010020c 3fffda00 00000001 3fff042c  
3ffffc80:  401008c8 ffffffff 00000011 00000001  
3ffffc90:  00000000 00000000 0000001f 40105b55  
3ffffca0:  4000050c 00000000 0000001f 40105b55  
3ffffcb0:  00000000 00000000 0000001f 40105b55  
3ffffcc0:  4000050c 00000004 3ffffd20 3ffffd31  
3ffffcd0:  40107020 00000000 3ffefddc 00000000  
3ffffce0:  3ffe934c 00000000 000000a5 401003f9  
3ffffcf0:  00000000 00000000 00000000 3ffffc80  
3ffffd00:  4000050c 000000a5 00000001 000000a5  
3ffffd10:  3ffffd30 3ffffd20 00000004 fffffffe  
3ffffd20:  40100537 3ffe8500 3ffefddc 000000ac  
3ffffd30:  000000a5 00000001 40105b55 4000050c  
3ffffd40:  00000000 00000000 0000001f 4010031d  
3ffffd50:  4000050c 00000080 3ffeebd8 40100426  
3ffffd60:  4010369f 00000030 0000001f 40100537  
3ffffd70:  2e1acf03 00000001 00000080 401006c2  
3ffffd80:  00000011 00000000 3ffe99d4 40107208  
3ffffd90:  ffffffff 3fffc6fc 00000001 4010385a  
3ffffda0:  7fffffff 00000000 00000022 00000030  
3ffffdb0:  00000001 40102ec1 00000104 7fffffff  
3ffffdc0:  3ffe934c 3ffe934c 00000048 401006d8  
3ffffdd0:  00000009 3fff02dc 00000000 4000050c  
3ffffde0:  3fffc278 40102dec 3fffc200 00000022  
3ffffdf0:  3ffffe00 401038ca 3ffe937c 00000000  
3ffffe00:  40000f83 00000030 0000001a ffffffff  
3ffffe10:  4010069e 00000023 00000000 00000014  
3ffffe20:  000000bc 000000bc 00000000 00000000  
3ffffe30:  ffffffff 3fffc6fc 00000001 00000010  
3ffffe40:  00000004 00000080 3ffeebd8 00000030  
3ffffe50:  00000004 7fffffff 7fffffff ffffffff  
3ffffe60:  40102e58 00000004 7fffffff 7fffffff  
3ffffe70:  40102e58 40102e3e 00000000 4000050c  
3ffffe80:  3fffc278 40102dec 3fffc200 00000022  
3ffffe90:  00000000 00000000 0000001f 40105b55  
3ffffea0:  4000050c 00000030 0000001a ffffffff  
3ffffeb0:  40000f49 00000030 00000011 ffffffff  
3ffffec0:  40000f49 00000000 00000000 00000000  
3ffffed0:  feefeffe 00000000 3fffd9d0 3ffeec24  
3ffffee0:  00000000 00000000 00000001 3fffdab0  
3ffffef0:  00000000 3fffdad0 3ffeec24 00000030  
3fffff00:  0000001c 00000000 00000000 40105474  
3fffff10:  0e72be98 3ffee074 0000001c 401006d3  
3fffff20:  3ffeebd8 00000010 000001f4 40107208  
3fffff30:  0e72be98 3ffee074 3ffefd24 4021b862  
3fffff40:  40105474 00000000 4010548d 00000000  
3fffff50:  4021b99c 3ffe88f4 3ffee350 4021b995  
3fffff60:  00000000 40222add 3ffee328 3ffee350  
3fffff70:  2e3c604d 60000600 3fffdab0 40107218  
3fffff80:  40222b22 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffee360 3fffdad0 3ffeec24 40202a3f  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

boddeke, 2016-04-05:
same here

leaving minimal sketch running for a night gives crashes every now and then. grep on the log:

2016/04/05 00:31:24: Exception (2):
2016/04/05 00:43:29: Exception (2):
2016/04/05 02:08:36: Exception (2):
2016/04/05 02:14:00: Exception (2):
2016/04/05 02:44:36: Exception (2):
2016/04/05 02:49:06: Exception (2):
2016/04/05 02:53:27: Exception (2):
2016/04/05 02:55:04: Exception (2):
2016/04/05 03:18:50: Exception (2):
2016/04/05 03:22:09: Exception (2):
2016/04/05 03:40:20: Exception (2):
2016/04/05 04:42:13: Exception (2):
2016/04/05 04:48:57: Exception (2):
2016/04/05 04:54:42: Exception (2):
2016/04/05 04:58:13: Exception (2):
2016/04/05 05:17:18: Exception (2):
2016/04/05 05:27:10: Exception (2):
2016/04/05 05:28:01: Exception (2):
2016/04/05 05:39:55: Exception (2):
2016/04/05 05:54:43: Exception (2):
2016/04/05 06:01:06: Exception (2):
2016/04/05 07:19:49: Exception (2):
2016/04/05 07:35:31: Exception (2):

the stack:

>>>stack>>>
3ffffc70:  4010020c 3fffda00 00000001 3fff07f4
3ffffc80:  401008c8 0000001c 000000ef 00000001
3ffffc90:  40107218 40224345 00000001 00000001
3ffffca0:  4022435c 3fff07fc 40224392 00000001
3ffffcb0:  3ffeea90 00000000 00000000 00000000
3ffffcc0:  00000018 00000004 3ffffd20 3ffffd31
3ffffcd0:  40107020 00000000 3ffefdbc 00000000
3ffffce0:  3ffe9318 feefeffe 000000a5 feefeffe
3ffffcf0:  00000000 00000000 00000000 3ffffc80
3ffffd00:  00000000 00000000 0000001f 40105b55
3ffffd10:  3ffffd30 3ffffd20 00000004 fffffffe
3ffffd20:  40103690 3ffe84ed 3ffefdbc 00000000
3ffffd30:  000000a5 00000001 0000001d 02001cc3
3ffffd40:  0000000f 3ffefc60 00000000 4010031d
3ffffd50:  00000000 000000a0 3ffeeb98 40100426
3ffffd60:  4000050c 00000000 0000001f 40100537
3ffffd70:  4010368d 00000001 0000003c 401006c2
3ffffd80:  127176b8 00000004 3ffe9fd8 40107208
3ffffd90:  00000013 3ffefc60 00000000 40103867
3ffffda0:  ffffffff 3fffc6fc 60000600 3ff20a00
3ffffdb0:  00000000 40102ec1 00000104 7fffffff
3ffffdc0:  3ffe9318 3ffe9318 00000008 7fffffff
3ffffdd0:  3ffe9318 40102e3e 00000000 4000050c
3ffffde0:  3fffc278 40102dec 3fffc200 00000022
3ffffdf0:  3ffffe00 40102dec 3fffc200 00000022
3ffffe00:  40000f83 00000030 0000001a ffffffff
3ffffe10:  4010069e 00000023 00000000 00000018
3ffffe20:  000000bc 000000bc 00000000 00000fa0
3ffffe30:  00000fa0 3fffc6fc 60000600 00000014
3ffffe40:  00000004 000000a0 3ffeeb98 00000030
3ffffe50:  00000000 05c55c73 4010549c 00000030
3ffffe60:  00000001 40102ec1 00000104 7fffffff
3ffffe70:  3ffe930c 40102e3e 00000008 7fffffff
3ffffe80:  40000f83 00000030 00000000 4000050c
3ffffe90:  00000000 00000000 0000001f 40105b55
3ffffea0:  4000050c 00000001 00000000 00000000
3ffffeb0:  40000f68 00000030 00000011 ffffffff
3ffffec0:  40000f58 00000000 00000020 00000000
3ffffed0:  000000bc 3ffefc60 00000000 7fffffff
[... 30 lines cut ...]
0x40107208: pvPortMalloc at /Users/frank/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/heap.c line 13
0x40103867: wDevDisableRx at ?? line ?
0x40102ec1: wDev_ProcessFiq at ?? line ?
0x40102e3e: wDev_ProcessFiq at ?? line ?
0x40102dec: wDev_ProcessFiq at ?? line ?
0x4010069e: _umm_malloc at /Users/frank/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/umm_malloc/umm_malloc.c line 1456
0x401003f4: check_poison at /Users/frank/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/umm_malloc/umm_malloc.c line 809
:  (inlined by) check_poison_block at /Users/frank/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/umm_malloc/umm_malloc.c line 843
0x40102ec1: wDev_ProcessFiq at ?? line ?
0x40102e3e: wDev_ProcessFiq at ?? line ?
0x40105b55: ets_timer_disarm at ?? line ?
0x401054c4: chm_get_current_channel at ?? line ?
0x401006d3: malloc at /Users/frank/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/umm_malloc/umm_malloc.c line 1655
0x40216212: dns_tmr at ?? line ?
0x40107208: pvPortMalloc at /Users/frank/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/heap.c line 13
0x40215fdf: dns_tmr at ?? line ?
0x4021b8a6: sys_timeout at ?? line ?
0x401054c4: chm_get_current_channel at ?? line ?
0x401054dd: chm_get_current_channel at ?? line ?
0x4021b9e0: sys_check_timeouts at ?? line ?
0x4021b9d9: sys_check_timeouts at ?? line ?
0x40222b21: ets_timer_handler_isr at ?? line ?
0x40107218: vPortFree at /Users/frank/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/heap.c line 18
0x40222b66: ets_timer_handler_isr at ?? line ?
0x402027c3: loop_task at /Users/frank/Library/Arduino15/packages/esp8266/hardware/esp8266/2.1.0/cores/esp8266/core_esp8266_main.cpp line 43

mkeyno, 2016-04-10:
hi @mic159  , I have problem to programming ESP-07 ,sometimes is working sometimes is not and not even programmed, can you tell me which IDE setting you were used for and if you have any schematic I really appreciated if you could share it

mic159, 2016-04-10:
@mkeyno the settings I used are listed there under "Settings in IDE". As for schematics, im using [this board from DX.com](http://www.dx.com/p/esp-07-esp8266-serial-wi-f-wireless-module-w-built-in-antenna-compatible-with-3-3v-5v-for-arduino-400559) with a USB to UART adaptor.

I also have some bare modules I program with a [breakout board](http://www.aliexpress.com/item/Free-Shipping-10Set-ESP8266-WiFi-Module-Breakout-Board-Adapter-Plate-for-ESP-07-ESP-08-ESP/32419541860.html), the same UART adaptor and 2 buttons that pull GPIO0 and RST to ground.

mic159, 2016-05-09:
@igrr I have updated to version 2.2.0 and this is no longer happening!

My module has been running for over 6 hours now and it has not crashed once!!
Thankyou!

closed 2016-05-09 by mic159

paysan, 2016-07-01:
Hi Mic, can you please tell me a bit more about this promiscuous project?
What do yoy normaly do in the main loop?

All I need to know in my own project is to catch a certain packet without further traffic handling.

Regards, Wim

mic159, 2016-07-04:
@paysan my project listens for two types of packets, wifi probe requests, and packets from other esp8266 modules to form a mesh network.

You don't need much in your main loop, you put your code in the callback that you registered in `wifi_set_promiscuous_rx_cb`.
Be sure to early-out so that you don't miss any packets.

The structure of the buffer can change with the length, indicated by the second parameter.

Check the NONOS SDK documentation for the structure.

An example:
```c
static void ICACHE_FLASH_ATTR handle_pkt(uint8_t* buf, uint16_t len) {
  // Bounds check
  if (len <= 12) return;
  
  // Espressif header
  RxControl* phy_hdr = (RxControl*) buf;
  // 802.11 header
  WiFiHeader* wifi_hdr = (WiFiHeader*) (buf + sizeof(RxControl));

  // Check for probe requests
  if (wifi_hdr->frame_control.version != 0) return;
  if (wifi_hdr->frame_control.type != WIFI_TYPE_CONTROL) return;

  // Captures
  if (wifi_hdr->frame_control.subtype == WIFI_SUBTYPE_PROBE) {
    ... handle it ...
  }
}

void setup() {
  Serial.begin(115200);
  wifi_set_opmode(STATION_MODE);
  wifi_set_channel(1);
  wifi_set_promiscuous_rx_cb(handle_pkt);
  wifi_register_send_pkt_freedom_cb(send_cb);

  // Start!
  wifi_promiscuous_enable(1);
}

void loop() {
}
```

You need to note that you can only see the wifi headers from here, not the actual payload contents of the packet. It seems to have a maximum of 128 bytes.

codemasterPL, 2016-07-11:
So let me ask a question, what's the point of a limited promiscous mode? This makes US (Makers) to step backward. Not an IOT approach @Espressif﻿.

mic159, 2016-07-11:
@codemasrerPL I think here just so they can do smartconfig setup.
Similar to this: http://electronics.stackexchange.com/questions/61704/how-does-ti-cc3000-wifi-smart-config-work

I agree, it totally sucks and I am also disappointed about the limitations on wifi_send_pkt_freedom.

I am really hoping for some reverse engineering efforts to find a better way by digging into internal functions.