esp8266/Arduino #1830: ESP become freeze when i transmit data to fast

Lexander24, 2016-03-29:
### Basic Infos

#### Hardware
Hardware:		ESP-01
Core Version:      	2.1.0

### Description
ESP freeze and no response

Problem description
I use ESP8266-01 as communication module for arduino UNO. ESP connect to server, i create my own server using VB6.
Sometimes when i transmit the data to fast, the esp become freeze and no responce

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512K (64K SPIFFS)
CPU Frequency:	80Mhz
Flash Mode:		DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch
#include <ESP8266WiFi.h>

char host[16],ssid[16],password[16];
int port;

int ipd1,ipd2,ipd3,ipd4; //ip device
int sub1,sub2,sub3,sub4; //subnet
int gat1,gat2,gat3,gat4; //gateway

IPAddress ip,subnet,gateway;

// Use WiFiClient class to create TCP connections
WiFiClient client;

String Serial_data,temp_ip;
int flag,stat;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  WiFi.disconnect();
  WiFi.mode(WIFI_STA);
  delay(10);
  Serial_data.reserve(150);
  
  String data;
  flag = 0;
  stat = 0;
  //s_ip=10.1.39.110~n_ap=ronaldo~p_ap=12345678~s_pt=6011~
  //d_ip=192.168.59.18~g_ip=192.168.59.1~n_ip=255.255.255.0~

  while (stat == 0){
    if (Serial.available()>8){
      Serial_data = Serial.readStringUntil('\n');
      //show setting recv
      Serial.print(Serial_data);
      process_Config();
      Serial_data = "";
      Serial.flush();
      stat = 2;     
    }    
  }
  
  while (stat == 2){
    if (Serial.available()>8){
      Serial_data = Serial.readStringUntil('\n');
      //show setting recv
      Serial.print(Serial_data);
      process_Config();
      Serial_data = "";
      Serial.flush();
      stat = 1;     
    }    
  }

  con2WiFi();
}

void loop() {
  // put your main code here, to run repeatedly:
  //Check WiFi Connection
  if(WiFi.status() == WL_CONNECTED){
    // check connection with server 
    if(!client.connected()){
      //stop connection
      client.stop();
      //reconnect to server
      con2server();  
    }
  }
  else{
    WiFi.disconnect();
    con2WiFi();
  }
      
  
  // Read all the lines of the reply from server and print them to Serial
  if(client.available()>0){
    String line = client.readStringUntil('\n');
    delay(50);
    Serial.println(line);
  }

  // Read command from arduino
  if (Serial.available()>0){
    Serial_data = Serial.readStringUntil('\n');
    //This will send the request to the server
    client.print(Serial_data);
    //Serial.print(Serial_data);
    Serial_data = "";     
  }
}

//Recv init value for setup network
void process_Config(){
  String task;
  String temp_HOST,temp_SSID,temp_PASS,temp_PORT;
  int m0,m11,m12,m13,m14,m15,m21,m22,m23,m24,m25,m31,m32,m33,m34,m35,m41,m42;
  m0 = Serial_data.indexOf('=');
  task = Serial_data.substring(0,m0);
   
  if(task == "s_ip"){
    Serial.println("");
    Serial.println("1 masuk");
      m12 = Serial_data.indexOf('~',m0+1);
      m21 = Serial_data.indexOf('=',m12+1);
      m22 = Serial_data.indexOf('~',m21+1);
      m31 = Serial_data.indexOf('=',m22+1);
      m32 = Serial_data.indexOf('~',m31+1);
      m41 = Serial_data.indexOf('=',m32+1);
      m42 = Serial_data.indexOf('~',m41+1);
    
      temp_HOST = Serial_data.substring(m0+1,m12);
      temp_SSID = Serial_data.substring(m21+1,m22);
      temp_PASS = Serial_data.substring(m31+1,m32);
      temp_PORT = Serial_data.substring(m41+1,m42);

      temp_HOST.toCharArray(host,16);
      temp_SSID.toCharArray(ssid,16);
      temp_PASS.toCharArray(password,16);
      port = temp_PORT.toInt();
  }
  else if(task == "d_ip"){
      String temp_oct;
      Serial.println("");
      Serial.println("2 masuk");
      m12 = Serial_data.indexOf('.',m0+1);  //d_ip_oct1
      m13 = Serial_data.indexOf('.',m12+1); //d_ip_oct2
      m14 = Serial_data.indexOf('.',m13+1); //d_ip_oct3
      m15 = Serial_data.indexOf('~',m14+1); //d_ip_oct4
      
      m21 = Serial_data.indexOf('=',m15+1); 
      m22 = Serial_data.indexOf('.',m21+1); //g_ip_oct1
      m23 = Serial_data.indexOf('.',m22+1); //g_ip_oct2
      m24 = Serial_data.indexOf('.',m23+1); //g_ip_oct3
      m25 = Serial_data.indexOf('~',m24+1); //g_ip_oct4
       
      m31 = Serial_data.indexOf('=',m25+1); 
      m32 = Serial_data.indexOf('.',m31+1); //s_ip_oct1
      m33 = Serial_data.indexOf('.',m32+1); //s_ip_oct2
      m34 = Serial_data.indexOf('.',m33+1); //s_ip_oct3
      m35 = Serial_data.indexOf('~',m34+1); //s_ip_oct4

      temp_oct = Serial_data.substring(m0+1,m12);
      ipd1 = temp_oct.toInt();
      temp_oct = Serial_data.substring(m12+1,m13);
      ipd2 = temp_oct.toInt();
      temp_oct = Serial_data.substring(m13+1,m14);
      ipd3 = temp_oct.toInt();
      temp_oct = Serial_data.substring(m14+1,m15);
      ipd4 = temp_oct.toInt();
      Serial.print("IP set : ");
      Serial.print(ipd1);
      Serial.print(".");
      Serial.print(ipd2);
      Serial.print(".");
      Serial.print(ipd3);
      Serial.print(".");
      Serial.println(ipd4);
      
      temp_oct = Serial_data.substring(m21+1,m22);
      gat1 = temp_oct.toInt();
      temp_oct = Serial_data.substring(m22+1,m23);
      gat2 = temp_oct.toInt();
      temp_oct = Serial_data.substring(m23+1,m24);
      gat3 = temp_oct.toInt();
      temp_oct = Serial_data.substring(m24+1,m25);
      gat4 = temp_oct.toInt();
      Serial.print("Gat set : ");
      Serial.print(gat1);
      Serial.print(".");
      Serial.print(gat2);
      Serial.print(".");
      Serial.print(gat3);
      Serial.print(".");
      Serial.println(gat4);
        
      temp_oct = Serial_data.substring(m31+1,m32);
      sub1 = temp_oct.toInt();
      temp_oct = Serial_data.substring(m32+1,m33);
      sub2 = temp_oct.toInt();
      temp_oct = Serial_data.substring(m33+1,m34);
      sub3 = temp_oct.toInt();
      temp_oct = Serial_data.substring(m34+1,m35);
      sub4 = temp_oct.toInt();
      Serial.print("Sub set : ");
      Serial.print(sub1);
      Serial.print(".");
      Serial.print(sub2);
      Serial.print(".");
      Serial.print(sub3);
      Serial.print(".");
      Serial.println(sub4);
  }    
}

// Connect to WiFi
void con2WiFi(){
  IPAddress ip(ipd1,ipd2,ipd3,ipd4);
  IPAddress subnet(sub1,sub2,sub3,sub4);
  IPAddress gateway(gat1,gat2,gat3,gat4);

  // Connect to WiFi network
  Serial.println("Connecting to AP ");
  //delay(5000);
  Serial.print("SSID Name : ");
  Serial.println(ssid);
  Serial.print("SSID Pass : ");
  Serial.println(password);
  //delay(250);
  delay(50);
  
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    //delay(500);
    delay(100);
  }
  Serial.println("");
  Serial.println("WiFi connected");
  
  //set static ip
  WiFi.config(ip,gateway,subnet);
  
  // device identification
  Serial.println("Device information");
  ip = WiFi.localIP();
  Serial.print("Device IP : ");
  Serial.println(ip);
  
  subnet = WiFi.subnetMask();
  Serial.print("Subnet    : ");
  Serial.println(subnet);
  
  gateway = WiFi.gatewayIP();
  Serial.print("Gateway   : ");
  Serial.println(gateway);
  
  Serial.println();
  Serial.print("Connecting to : ");
  Serial.println(host);

  //delay(500);
  con2server();
}

// Set-up connetion to server
void con2server(){
  if (!client.connect(host, port)) {
    flag++;
    //test new update
    if(flag % 50 == 0)
      flag = 1;
    if (flag == 1)
      Serial.println("S=Searching server#$");
    return;
  }
  else{
    flag = 0;
    Serial.println("S=Connect#$");
    delay(500);
    temp_ip = ipd1;
    temp_ip += ".";
    temp_ip += ipd2;
    temp_ip += ".";
    temp_ip += ipd3;
    temp_ip += ".";
    temp_ip += ipd4;
    temp_ip += "#RONALDO#$";
    client.print(temp_ip);  
  }
}

### Debug Messages
here is view from serial monitor
:rn 20
:c0 1, 20
:rn 20
192.168.59.20:M1:$

:c0 1, 20
192.168.59.20:M2:$

:wr
:sent 19
:rn 20
:ww
:c0 1, 20
192.168.59.20:M3:$

:wr
:sent 19
:ww
:wr
:sent 19
:rn 20
:ww
:c0 1, 20
192.168.59.20:M4:$

:wr
:sent 19
:rn 20
:ww
:c0 1, 20
192.168.59.20:M5:$

:wr
:sent 19
:ww
:wr
:ww

my normal data transaction when i see
:rn 20
:c0 1, 20
192.168.59.20:M1:$
:wr
:sent 19
:ww

:rn 20
:c0 1, 20
192.168.59.20:M2:$
:wr
:sent 19
:ww

:rn 20
:c0 1, 20
192.168.59.20:M3:$
:wr
:sent 19
:ww

:rn 20
:c0 1, 20
192.168.59.20:M4:$
:wr
:sent 19
:ww

:rn 20
:c0 1, 20
192.168.59.20:M5:$
:wr
:sent 19
:ww

:wr
:sent 31
:ww

i try to close my server, but nothing see happen on serial monitor. When the esp normal, if i close my server i got this message
:rcl
:abort
:ur 1
:del
[hostByName] Host: 10.1.39.110 is a IP!
:err -8
S=Searching server#$
[hostByName] Host: 10.1.39.110 is a IP!
:err -8
[hostByName] Host: 10.1.39.110 is a IP!
:err -8
[hostByName] Host: 10.1.39.110 is a IP!
:err -8
[hostByName] Host: 10.1.39.110 is a IP!
:err -8
[hostByName] Host: 10.1.39.110 is a IP!
:err -8
[hostByName] Host: 10.1.39.110 is a IP!        <-message when i reconnect to server
:ref 1
S=Connect#$
:wr
:sent 23
:ww

devyte, 2017-10-17:
Closing per #3655 .

d-a-v, 2017-10-17:
This sketch will be worth trying with lwip2 #3362.

d-a-v, 2017-10-18:
@Lexander24 are you willing to give a try, or is it too late since it's been 1½ year ?

Lexander24, 2017-10-18:
I believe there is no too late. Because this problem can appear at different time, people, and conditions.

devyte, 2017-10-25:
@Lexander24 please try #3362 over latest git, and report your findings here. In the meantime, I'll add this to the lwip2 project.

devyte, 2018-02-28:
@Lexander24 is this issue still valid with latest git and lwip2?

devyte, 2018-07-12:
In the meantime, lwip glue has undergone a lot of rework and seems solid now, and the Serial code has had several major bug fixes and has undergone stress testing.
Closing.

closed 2018-07-12 by devyte