esp8266/Arduino #1831: pin macro digitalPinHasPWM returns invalid results

jnsbyr, 2016-03-29:
### Basic Infos

#### Hardware
Hardware:		ESP8266 - all boards
Core Version:      	2.1.0

### Description

The macro ```digitalPinHasPWM(p)``` from _variants/.../pins_arduino.h_ has a similar problem as A0 from issue #1766.

```digitalPinHasPWM``` returns true for pins 1 to 17 with an Adafruit Huzzah board. But it should return true  for 0 to 16 (or even better 0 to 5 and 12 to 16, because you should not do PWM on your flash IOs). The reason is that the macro is intended to return a boolean (0/1) result but current macro returns ```NOT_A_PIN``` for pin 17 but that resolves to -1 and that is _true_ but should be _false_ and it otherwise returns the pin number and for pin 0 that is _false_ but should be _true_.

With the macro ```digitalPinToInterrupt``` the case is a little bit different. It should return the number of the external interrupt for a specific pin to be used with ```attachInterrupt()```. With AVR architecture this results in 0 for pin 2 and 1 for pin 3 and otherwise ```NOT_AN_INTERRUPT``` (-1) for all other pins being an offset for the interrupt vector address of INT0. For the ESP8266 the current solution seems to return the right values but the macro should be defined using the constant ```NOT_AN_INTERRUPT``` instead of ```NOT_A_PIN```.

### Possible Solution

* minimum change
```C
#define digitalPinToInterrupt(p)  	(((p) < EXTERNAL_NUM_INTERRUPTS)? (p) : NOT_AN_INTERRUPT)
#define digitalPinHasPWM(p)         (((p) < NUM_DIGITAL_PINS)? 1 : 0)
```

* saveguard approach
```C
#define digitalPinHasPWM(p)         (((p) < NUM_DIGITAL_PINS && ((p) < 6 || (p) > 11))? 1 : 0)
```

in all variants of pins_arduino.h

This issue blocks https://github.com/firmata/arduino/pull/278.

igrr, 2016-03-29:
@jnsbyr
I'm sorry, i haven't noticed your comment on the old issue (#1766).
Please let me know if #1833 looks good to you.

jnsbyr, 2016-03-29:
@igrr 
No problem - it was my fault to add to an old issue. The diffs look good. Will test it.

jnsbyr, 2016-04-01:
@igrr 
Tested by replacing the *variants* folder of the 2.1.0 release with the version of commit *0f719e8* using an Adafruit Huzzah board. The result for ```digitalPinHasPWM``` is now as expected:
*pins 0,1,2,3,4,5,12,13,14,15,16*

Thank you very much!

PR #1833 (2016-04-01): Clean up variants, fix digitalPinHasPWM definition
Should fix #1831.
closed 2016-04-01 by igrr