esp8266/Arduino #1834: Wire.read() cause the exception

robertgregor, 2016-03-30:
Hello, I am using this method to read the data:

void MMA8452::readMultiple(uint8_t reg, uint8_t *buffer, uint8_t numBytes)
{
        Serial.println("1");
	Wire.beginTransmission(MMA8452_ADDRESS);
        Serial.println("2");
	Wire.write(reg);
        Serial.println("3");
	Wire.requestFrom((uint8_t)MMA8452_ADDRESS, numBytes);
        Serial.println("5");
	while (Wire.available() < numBytes) {};
        Serial.println("6");
	while (numBytes--)
	{
                Serial.println("7");
		*buffer++ = Wire.read();
                Serial.println("8");
	}
        Wire.endTransmission();
}

When tried to read, I have an exception:

ESP chip ID: 1193309
Connected to AP. IP:
192.168.1.5
1
2
3
5
6
7

Exception (29):
epc1=0x40208bb5 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff1440 end: 3fff16b0 offset: 01a0

>>>stack>>>
3fff15e0:  3fff05c0 3fff047c 3fff05c0 40208bad  
3fff15f0:  0000000d 00000003 3fff1d94 401008c8  
3fff1600:  3fff05c0 00000003 3fff047c 3fff067c  
3fff1610:  3fff05a0 3fff03f8 3ffef87c 40208bee  
3fff1620:  4022f692 3fff05c0 3fff03f8 40208c18  
3fff1630:  402304af 3ffe92a0 00ffffff 3ffe92a0  
3fff1640:  3fff183c 4020936c 0501a8c0 402019ab  
3fff1650:  3fff05a0 3fff03f8 3fff05c0 40206096  
3fff1660:  feefeffe feefeffe feefeffe feefeffe  
3fff1670:  feefeffe feefeffe feefeffe feefeffe  
3fff1680:  feefeffe feefeffe feefeffe 3fff067c  
3fff1690:  3fffdad0 00000000 3fff0675 4020cda8  
3fff16a0:  feefeffe feefeffe 3fff0690 40100958  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

Please note, the same code works on Arduino UNO without any problem...
Any idea?

igrr, 2016-03-30:
Could you please decode the stack dump and paste the result here? Thanks.
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

robertgregor, 2016-03-30:
Decoding 11 results
0x40208bad: ESP8266WebServer::_uploadReadByte(WiFiClient&) at C:\Users\gregorro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\libraries\ESP8266WebServer\src/Parsing.cpp line 311
0x401008c8: free at C:\Users\gregorro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266\umm_malloc/umm_malloc.c line 1727
0x40208bee: ESP8266WebServer::urlDecode(String const&) at C:\Users\gregorro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\libraries\ESP8266WebServer\src/Parsing.cpp line 530
0x4022f692: wifi_get_opmode at ?? line ?
0x40208c18: ESP8266WebServer::urlDecode(String const&) at C:\Users\gregorro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\libraries\ESP8266WebServer\src/Parsing.cpp line 534
0x402304af: wifi_station_get_connect_status at ?? line ?
0x4020936c: ESP8266WebServer::_parseForm(WiFiClient&, String, unsigned int) at C:\Users\gregorro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\libraries\ESP8266WebServer\src/Parsing.cpp line 451 (discriminator 1)
0x402019ab: delay at C:\Users\gregorro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/core_esp8266_wiring.c line 50
0x40206096: setup at C:\Users\gregorro\Disk Google\RH\RemoteHomeArduinoSketches\WiFi_ESP\AccelerometerSensor/AccelerometerSensor.ino line 1083
0x4020cda8: loop_wrapper at C:\Users\gregorro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/core_esp8266_main.cpp line 43
0x40100958: cont_norm at C:\Users\gregorro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/cont.S line 109

igrr, 2016-03-30:
Interestingly, the stack trace doesn't show anything related to Wire library.
Perhaps you have changed the sketch slightly after capturing the stack trace?

Also please see a comment from @me-no-dev on the related issue: https://github.com/esp8266/Arduino/issues/470#issuecomment-199238171

me-no-dev, 2016-03-30:
It will be interesting to see that code working on AVR Arduino also.
The flow above is as follows:
-beginTransmission
-write
-readBytes
-endTransmission
when it should have been:
-beginTransmission
-write
-endTransmission
-readBytes

robertgregor, 2016-03-31:
OK, I have modified the code as you proposed:
void MMA8452::readMultiple(uint8_t reg, uint8_t *buffer, uint8_t numBytes)
{
        Serial.println("1");
	Wire.beginTransmission(MMA8452_ADDRESS);
        Serial.println("2");
	Wire.write(reg);
        Wire.endTransmission();
        Serial.println("3");
	Wire.requestFrom((uint8_t)MMA8452_ADDRESS, numBytes);
        Serial.println("5");
	while (Wire.available() < numBytes) {};
        Serial.println("6");
	while (numBytes--)
	{
                Serial.println("7");
		*buffer++ = Wire.read();
                Serial.println("8");
	}
}

The output is:
ESP chip ID: 1193309
Connected to AP. IP:
192.168.1.5
1
2
3
5
6
7

Exception (29):
epc1=0x40207baa epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff1320 end: 3fff1590 offset: 01a0

>>>stack>>>
3fff14c0:  3fff04ac 3fff04ac 3fff0368 40207ba2  
3fff14d0:  0000000d 00000003 3fff1500 4010068c  
3fff14e0:  3fff04ac 00000003 3fff0368 3fff0568  
3fff14f0:  3fff048c 3fff0330 3ffef7dc 40207bde  
3fff1500:  4022e81e 3fff04ac 3fff0330 40207c08  
3fff1510:  4022f63b 3ffe9270 00ffffff 3ffe9270  
3fff1520:  3fff16cc 402082f0 0501a8c0 402018ef  
3fff1530:  3fff048c 3fff0330 3fff04ac 40205e3a  
3fff1540:  feefeffe feefeffe feefeffe feefeffe  
3fff1550:  feefeffe feefeffe feefeffe feefeffe  
3fff1560:  feefeffe feefeffe feefeffe 3fff0568  
3fff1570:  3fffdad0 00000000 3fff0560 4020d7a0  
3fff1580:  feefeffe feefeffe 3fff0570 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

Decoded stacktrace:
Decoding 11 results
0x40207ba2: MMA8452::readMultiple(unsigned char, unsigned char*, unsigned char) at C:\Users\gregorro\Disk Google\RH\RemoteHomeArduinoSketches\WiFi_ESP\libraries\MMA8452/MMA8452.cpp line 52
0x4010068c: free at C:\RH\arduino-1.6.6\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c line 1731
0x40207bde: MMA8452::read(unsigned char) at C:\Users\gregorro\Disk Google\RH\RemoteHomeArduinoSketches\WiFi_ESP\libraries\MMA8452/MMA8452.cpp line 437
0x4022e81e: wifi_get_opmode at ?? line ?
0x40207c08: StringSumHelper at C:\RH\arduino-1.6.6\hardware\esp8266com\esp8266\cores\esp8266/WString.h line 266
:  (inlined by) MMA8452::init() at C:\Users\gregorro\Disk Google\RH\RemoteHomeArduinoSketches\WiFi_ESP\libraries\MMA8452/MMA8452.cpp line 19
0x4022f63b: wifi_station_get_connect_status at ?? line ?
0x402082f0: ESP8266WiFiSTAClass::status() at C:\RH\arduino-1.6.6\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp line 482
0x402018ef: delay at C:\RH\arduino-1.6.6\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_wiring.c line 50
0x40205e3a: setup at C:\Users\gregorro\Disk Google\RH\RemoteHomeArduinoSketches\WiFi_ESP\AccelerometerSensor/AccelerometerSensor.ino line 1083
0x4020d7a0: loop_wrapper at C:\RH\arduino-1.6.6\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 43
0x40100718: cont_norm at C:\RH\arduino-1.6.6\hardware\esp8266com\esp8266\cores\esp8266/cont.S line 109

igrr, 2016-03-31:
Exception 29 is a store by an illegal address, see
https://github.com/esp8266/Arduino/blob/master/doc/exception_causes.md
`excvaddr` contains the address which caused an exception.
It seems that your `buffer` variable is zero, i.e. you must have passed a null pointer into this function.

robertgregor, 2016-03-31:
Hmmm, OK, this is from MMA8452 library, which I am successfully using and used several times with Arduino. In fact, it is going from init() function of the accelerometer:

bool MMA8452::init()
{
	// Check who-am-i register
	uint8_t reg = read(WHO_AM_I);
	return (reg == 0x2A);
} 

uint8_t MMA8452::read(uint8_t reg)
{
	uint8_t *buf = 0;
	readMultiple(reg, buf, 1);
	return *buf;
}
void MMA8452::readMultiple(uint8_t reg, uint8_t *buffer, uint8_t numBytes)
 {
 Wire.beginTransmission(MMA8452_ADDRESS);
 Wire.write(reg);
 Wire.endTransmission();
 Wire.requestFrom((uint8_t)MMA8452_ADDRESS, numBytes);
 while (Wire.available() < numBytes) {};
 while (numBytes--)
    {
      *buffer++ = Wire.read();
    }
 } 

So that's true, that uint8_t *buf = 0; but why it is working with Arduino? (This is originally Arduino library...)

me-no-dev, 2016-03-31:
if RAM on arduino starts from 0x00, then it will not be a problem to address that space. That is not the case on the ESP though and the fact that it works on AVR Arduino does not make it correct :)

igrr, 2016-03-31:
IIRC, AVR memory map starts with a bunch of general-purpose registers. Writing them doesn't trigger an exception, but they may be allocated by the compiler to store some variables which will be overwritten. So this code gets you an UB (undefined behavior) — it may work or you may get eaten by a raptor, no guarantees.

robertgregor, 2016-03-31:
I see the point. Thanks you pointed it. Dont understand, why someone wrote it like that :) 
I have fixed it like that:
uint8_t MMA8452::read(uint8_t reg)
{
	uint8_t buf[1];
	readMultiple(reg, buf, 1);
	return buf[0];
}

And I am very sorry to bother you with the stupid issues :))))))))

closed 2016-03-31 by robertgregor