esp8266/Arduino #1835: Handling and writing 2 json files in function is unstable

koffienl, 2016-03-30:
I'm working on a project where I store all the settings in a json file on SPIFFS. For reading/writing the json object I use the ArduinoJson library.
The problem is : when reading 2 josn files, changing values and writing back to SPIFFS is unstable. 2 out of 10 times I get wdt resets or panics.
To rule out any other work in the program, I have created a (nearly) empty loop function:
```
void loop (void)
{
  server.handleClient();
  if (WMode == "AP" && ssidStored != "" && passStored != "" && millis() > (60000 * 5) )
  {
    Serial.println("SoftAP Timer reboot");
    ESP.restart();
  }
}
```

I open a webpage from the webserver, POSTing some settings back to the ESP. The values are stored in 2 seperate json files on SPIFFS.
When I comment out the writing of the second file, I don't get any crashes. Somehow the ArduinoJson library is to slow(?) of the SPIFFS is giving problems?

In most of the times I can see on the serial monitor the line stating everything is OK and sending answer to the webbrowser. But the browser doesn't get the answer and the Serial Monitor shows a wdt reset:

```
  {
    String dht_boolArg = server.arg("dht_bool");
    String dht_pinArg = server.arg("dht_pin");
    String dht_tempvarArg = server.arg("dhttemp_var");
    String dht_humvarArg = server.arg("dhthum_var");
    String dht_typeArg = server.arg("dht_type");
    String dht_intervalArg = server.arg("dht_interval");

    if (dht_boolArg == "on")
    {
      dht_boolArg = "1";
    }
    else
    {
      dht_boolArg = "0";
    }

    String OldGpio;

    Serial.println("lezen plugins.json");
    Serial.println(ESP.getFreeHeap());
    String pluginsjsn = ReadConfig("/plugins.json");
    Serial.println("gelezen. aanmaken jsonbuffer");
    Serial.println(ESP.getFreeHeap());
    DynamicJsonBuffer jsonBufferdht;
    JsonObject& plugins = jsonBufferdht.parseObject(const_cast<char*>(pluginsjsn.c_str()));
    Serial.println("buffer gemaakt");
    Serial.println(ESP.getFreeHeap());
    if (!plugins.containsKey("dht"))
    {
      plugins.createNestedObject("dht");
    }
    JsonObject& dht = plugins["dht"];

    if (dht.containsKey("gpio"))
    {
      OldGpio = dht["gpio"];
    }

    dht["temp-variable"] = const_cast<char*>(dht_tempvarArg.c_str());
    dht["hum-variable"] = const_cast<char*>(dht_humvarArg.c_str());
    dht["enabled"] = const_cast<char*>(dht_boolArg.c_str());
    dht["gpio"] = const_cast<char*>(dht_pinArg.c_str());
    dht["dhttype"] = const_cast<char*>(dht_typeArg.c_str());
    dht["interval"] = const_cast<char*>(dht_intervalArg.c_str());

    Serial.println("wegschrijven json naar file");
    Serial.println(ESP.getFreeHeap());
    int len = plugins.measurePrettyLength() + 1;
    char ch[len];
    size_t n = plugins.prettyPrintTo(ch, sizeof(ch));
    String tt(ch);
    File configFile = SPIFFS.open("/plugins.json", "w");
    if (!configFile)
    {
      Serial.println("Failed to open config file for writing");
      return;
    }
    else
    {
      configFile.print(tt);
      //configFile.flush();
      configFile.close();
    }
    delay(500);

    Serial.println("old gpio: " + String(OldGpio) + " nieuw gpio " + String(dht_pinArg) );

    if (OldGpio != dht_pinArg)    // If GPIO number is not the same as current, update gpio.json
    {
      Serial.println("lezen gpio.json");
      Serial.println(ESP.getFreeHeap());

      String gpiojsn = ReadConfig("/gpio.json");

      Serial.println("gelezen. nieuwe buffer maken");
      DynamicJsonBuffer jsonBufferGpio;
      JsonObject& gpios = jsonBufferGpio.parseObject(const_cast<char*>(gpiojsn.c_str()));
      Serial.println("buffer gemaakt");
      String CurrentGpio = "gpio" + String(dht_pinArg);
      Serial.println("CurrentGpio is : " + String(CurrentGpio) ) ;
      if (!gpios.containsKey(const_cast<char*>(CurrentGpio.c_str())))
      {
        Serial.println("aanmaken " + String(CurrentGpio) );
        gpios.createNestedObject(const_cast<char*>(CurrentGpio.c_str()));
      }
      JsonObject& currgpio = gpios[const_cast<char*>(CurrentGpio.c_str())];
      currgpio["devicetype"] = "dht";

      String OldGpioName = "gpio" + String(OldGpio);
      if (gpios.containsKey(const_cast<char*>(OldGpioName.c_str())))
      {
        JsonObject& tempje = gpios[const_cast<char*>(OldGpioName.c_str())];
        tempje["devicetype"] = "";
      }

      Serial.println("wegschrijven json");
      Serial.println(ESP.getFreeHeap());
      int len2 = gpios.measurePrettyLength() + 1;
      char ch2[len2];
      size_t n2 = gpios.prettyPrintTo(ch2, sizeof(ch2));
      String tt2(ch2);

      File configFile2 = SPIFFS.open("/gpio.json", "w");

      configFile2.print(tt2);
      //configFile2.flush();
      configFile2.close();

    }

    delay(500);
    Serial.println("klaar. stuur OK naar browser");
    server.send ( 200, "text/html", "OK");
  }
}
``` 

I have this on Core Version 2.0.0 and 2.1.0

Any idea on where to find out what I'm doing wrong ?

igrr, 2016-03-30:
Could you please post the complete sketch (not just an excerpt) which demonstrates the issue?
If you are getting an exception, please decide it using EspExceptionDecoder and post the output as well. Thanks.

koffienl, 2016-03-30:
The entire sketch is a bit to big to post. This snippet of code is the actual code that is called when the url is accessed.
To be more specific :
part in main:
```
server.on("/dht_ajax", PluginWebServerDHT_ajax);
```

entire function:
```
void PluginWebServerDHT_ajax()
{
  String form = server.arg("form");
  if (form != "dht")
  {
       not related
  }
  if (form == "dht")
  {
    String dht_boolArg = server.arg("dht_bool");
    String dht_pinArg = server.arg("dht_pin");
    String dht_tempvarArg = server.arg("dhttemp_var");
    String dht_humvarArg = server.arg("dhthum_var");
    String dht_typeArg = server.arg("dht_type");
    String dht_intervalArg = server.arg("dht_interval");

    if (dht_boolArg == "on")
    {
      dht_boolArg = "1";
    }
    else
    {
      dht_boolArg = "0";
    }

    String OldGpio;

    Serial.println("lezen plugins.json");
    Serial.println(ESP.getFreeHeap());
    String pluginsjsn = ReadConfig("/plugins.json");
    Serial.println("gelezen. aanmaken jsonbuffer");
    Serial.println(ESP.getFreeHeap());
    DynamicJsonBuffer jsonBufferdht;
    JsonObject& plugins = jsonBufferdht.parseObject(const_cast<char*>(pluginsjsn.c_str()));
    Serial.println("buffer gemaakt");
    Serial.println(ESP.getFreeHeap());
    if (!plugins.containsKey("dht"))
    {
      plugins.createNestedObject("dht");
    }
    JsonObject& dht = plugins["dht"];

    if (dht.containsKey("gpio"))
    {
      OldGpio = dht["gpio"];
    }

    dht["temp-variable"] = const_cast<char*>(dht_tempvarArg.c_str());
    dht["hum-variable"] = const_cast<char*>(dht_humvarArg.c_str());
    dht["enabled"] = const_cast<char*>(dht_boolArg.c_str());
    dht["gpio"] = const_cast<char*>(dht_pinArg.c_str());
    dht["dhttype"] = const_cast<char*>(dht_typeArg.c_str());
    dht["interval"] = const_cast<char*>(dht_intervalArg.c_str());

    Serial.println("wegschrijven json naar file");
    Serial.println(ESP.getFreeHeap());
    int len = plugins.measurePrettyLength() + 1;
    char ch[len];
    size_t n = plugins.prettyPrintTo(ch, sizeof(ch));
    String tt(ch);
    File configFile = SPIFFS.open("/plugins.json", "w");
    if (!configFile)
    {
      Serial.println("Failed to open config file for writing");
      return;
    }
    else
    {
      configFile.print(tt);
      //configFile.flush();
      configFile.close();
    }
    delay(500);

    Serial.println("old gpio: " + String(OldGpio) + " nieuw gpio " + String(dht_pinArg) );

    if (OldGpio != dht_pinArg)    // If GPIO number is not the same as current, update gpio.json
    {
      Serial.println("lezen gpio.json");
      Serial.println(ESP.getFreeHeap());

      String gpiojsn = ReadConfig("/gpio.json");

      Serial.println("gelezen. nieuwe buffer maken");
      DynamicJsonBuffer jsonBufferGpio;
      JsonObject& gpios = jsonBufferGpio.parseObject(const_cast<char*>(gpiojsn.c_str()));
      Serial.println("buffer gemaakt");
      String CurrentGpio = "gpio" + String(dht_pinArg);
      Serial.println("CurrentGpio is : " + String(CurrentGpio) ) ;
      if (!gpios.containsKey(const_cast<char*>(CurrentGpio.c_str())))
      {
        Serial.println("aanmaken " + String(CurrentGpio) );
        gpios.createNestedObject(const_cast<char*>(CurrentGpio.c_str()));
      }
      JsonObject& currgpio = gpios[const_cast<char*>(CurrentGpio.c_str())];
      currgpio["devicetype"] = "dht";

      String OldGpioName = "gpio" + String(OldGpio);
      if (gpios.containsKey(const_cast<char*>(OldGpioName.c_str())))
      {
        JsonObject& tempje = gpios[const_cast<char*>(OldGpioName.c_str())];
        tempje["devicetype"] = "";
      }

      Serial.println("wegschrijven json");
      Serial.println(ESP.getFreeHeap());
      int len2 = gpios.measurePrettyLength() + 1;
      char ch2[len2];
      size_t n2 = gpios.prettyPrintTo(ch2, sizeof(ch2));
      String tt2(ch2);

      File configFile2 = SPIFFS.open("/gpio.json", "w");

      configFile2.print(tt2);
      //configFile2.flush();
      configFile2.close();

    }

    delay(500);
    Serial.println("klaar. stuur OK naar browser");
    server.send ( 200, "text/html", "OK");
  }
}
```

Values are filled in on a page and submitted to the /dht_ajax URL. The function detects if it should answer with current settings (server.arg("form") is empty) or that it should process the posted values (server.arg("form") = "dht" ).

Sometimes it give an exception, but most of the time a wdt reset. As far as I can see I'm not doing any strange time-consuming stuff ?

The exception doesn't give much information:
```
Panic C:\Users\richa.DESKTOP-Q4DFU98\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266\core_esp8266_main.cpp:114 loop_task

ctx: sys 
sp: 3ffffdd0 end: 3fffffb0 offset: 01b0

>>>stack>>>
3fffff80:  40223f86 3fffdab0 00000000 3fff21c0  
3fffff90:  00000000 3fffdc20 3fff21b4 40217ac1  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<

```
```

Decoding 5 results
0x40223f86: ets_timer_handler_isr at ?? line ?
0x40217ac1: loop_task at C:\Users\richa.DESKTOP-Q4DFU98\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266/core_esp8266_main.cpp line 43
```

koffienl, 2016-04-03:
OK, I have stripped almost everything from my project, leaving a basic sketch that connects to wifi, setup a webserver and respond to one page request.

the problems is as follows: a HTML file on the SPIFFS is requested presenting a form with some input fields. When the user change the values in the page and hit submit, the data is send to the ESP. The ESP detects that `server.arg("form")` has the value `dht` and takes all the arguments and writes it to 2 seperate JSON files. On the serial monitor I can see the last message `Done. Send OK to browser` but the OK is not send and a few seconds later the WDT crash kicks in:
```
Done. Send OK to browser

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
```

I can push the values from the HTML form with the page, or just do a GET url (http://esp/dht_ajax?dht_bool=1&dht_pin=12&dhttemp_var=vartemp&dhthum_var=varhum&dht_type=1&dht_interval=1&form=dht) i both cases the problem is the same : most of the time everything is OK but every now and then the WDT crash occurs.

If I comment out the actual writing to SPIFFS of one of the 2 files, the crash doesn't occur.
I have test with Arduino IDE 1.6.5 with core version 2.0.0 and 2.1.0 with a NodeMCU and a Wemos D1 mini.

The stripped down sketch : 
```
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <FS.h>
#include <ArduinoJson.h>

const char* APssid = "ESPimatic";
const char* APpassword = "espimatic";
String WMode;
String sep = "____";

ESP8266WebServer  server(80);
WiFiClient client;

//holds the current upload
File UploadFile;
String fileName;

//-------------- FSBrowser application -----------
//format bytes
String formatBytes(size_t bytes) {
  if (bytes < 1024) {
    return String(bytes) + "B";
  } else if (bytes < (1024 * 1024)) {
    return String(bytes / 1024.0) + "KB";
  } else if (bytes < (1024 * 1024 * 1024)) {
    return String(bytes / 1024.0 / 1024.0) + "MB";
  } else {
    return String(bytes / 1024.0 / 1024.0 / 1024.0) + "GB";
  }
}

void setup()
{
  Serial.begin(115200);
  Serial.println("");
  delay(500);
  // Check if SPIFFS is OK
  if (!SPIFFS.begin())
  {
    Serial.println("SPIFFS failed, needs formatting");
    delay(500);
    ESP.restart();
  }
  else
  {
    FSInfo fs_info;
    if (!SPIFFS.info(fs_info))
[... 586 lines cut ...]
      gpios[const_cast<char*>(OldGpioName.c_str())]["devicetype"] = "";
    }

    Serial.println("WRITE json to file");
    Serial.println(ESP.getFreeHeap());
    int len2 = gpios.measurePrettyLength() + 1;
    char ch2[len2];
    size_t n2 = gpios.prettyPrintTo(ch2, sizeof(ch2));
    String tt2(ch2);

    File configFile2 = SPIFFS.open("/gpio.json", "w");
    configFile2.print(tt2);
    configFile2.close();

    Serial.println("Feed the dog");
    ESP.wdtFeed();
 
    Serial.println("Done. Send OK to browser");
    server.send ( 200, "text/html", "OK");
  }
}
```

I have tried yield, delay and wdtFeed on places where I think there might be some CPU delay for the task but didn't helped.
It's driving me nuts, why does the WDT kick in for these quite simple tasks ?

jakalada, 2016-04-04:
@koffienl
Hi!
I copy the stripped down sketch and  push verify button in Arduino IDE.
I am getting the following error.

```
ambiguous overload for 'operator=' (operand types are 'String' and 'const ArduinoJson::JsonObjectSubscript<const char*>')
```

Are you getting same error?
I have used the following version of the library .

* ArduinoJson: 5.1.1
* Arduino IDE: 1.6.8

koffienl, 2016-04-04:
Hi @jakalada I don't get that error. Perhaps it has something todo with Arduino IDE 1.6.8? I'm using 1.6.5
I'm using the ArduinoJson v4.5 that was (as far as I remember) part of the installed libraries from core ?

jakalada, 2016-04-04:
@koffienl 
Thank you for the information.
I was successful upload sketch!

I can continue to use Arduino IDE 1.6.8.
but, I change ArduinoJson v5.1.1 to v4.5.

I try the sketch and reports if I find something :)

koffienl, 2016-04-04:
Thanks!
For troubleshooting : just continuously 'save' the settings. Sometimes it works 10 times without issues, sometimes it crash first time after restart.

jakalada, 2016-04-05:
I had to add `ESP.wdtFeed()` for the sketch running.
Why dont'you add `ESP.wdtFeed()` in the back of the SPIFFS process?

I add a `ESP.wdtFeed()` in `setup()`
```
  // Check if SPIFFS is OK
  if (!SPIFFS.begin())
  {
    Serial.println("SPIFFS failed, needs formatting");
    delay(500);
    ESP.restart();
  }
  else
  {
    FSInfo fs_info;
    if (!SPIFFS.info(fs_info))
    {
      Serial.println("fs_info failed\n");
    }
  }
  //  I add line
  ESP.wdtFeed();
```

> just continuously 'save' the settings. 

Save the settings using the above sketch?
I can't found a web page with any input forms...

koffienl, 2016-04-05:
I'm not sure if I understand your reply?
I could add a `ESP.wdtFeed()` during `setup()` but I don't have any WDT errors during the setup. Nevertheless I will add a wdtfeed in the setup to see any improvement.

You can push the settings with a GET URL : `http://esp/dht_ajax?dht_bool=1&dht_pin=12&dhttemp_var=vartemp&dhthum_var=varhum&dht_type=1&dht_interval=1&form=dht` or you can use this HTML file : https://github.com/koffienl/ESPimatic/blob/master/Upload%20this%20to%20SPIFFS/dht.html

jakalada, 2016-04-05:
I'm sorry!
I am using ESP-WROOM-02 board.
Perhaps, I'm using a board different from you. So, I got a WDT error that do not you know.

> You can push the settings with a GET URL : http://esp/dht_ajax?dht_bool=1&dht_pin=12&dhttemp_var=vartemp&dhthum_var=varhum&dht_type=1&dht_interval=1&form=dht or you can use this > HTML file : https://github.com/koffienl/ESPimatic/blob/master/Upload%20this%20to%20SPIFFS/dht.html

Does [master brach of your project](https://github.com/koffienl/ESPimatic) will have the issue?
If so, I will try it.

koffienl, 2016-04-06:
No the master branch doens't have this issue, that version is pre-json. Haven't pushed any code with json yet, but the HTML file is still the same.
Thanks for trying!

jakalada, 2016-04-11:
I repeated request the GET URL. 50 times.

* `http://esp/dht_ajax?dht_bool=1&dht_pin=12&dhttemp_var=vartemp&dhthum_var=varhum&dht_type=1&dht_interval=1&form=dht`

And I got repeated the following serial output.

```
READ plugins.json
34592
Done. Create JsonBuffer
34416
Done
33808
WRITE json to file
33808
Feed the dog
READ gpio.json
33616
Done. Create second JsonBuffer
Done
CurrentGpio is : gpio12
WRITE json to file
33168
Feed the dog
Done. Send OK to browser
```
I didn't get the error.

My hadware info.

* Hardware: ESP-WROOM-02
* Core Version: 2.1.0

My IDE settings info.

* Module: Generic ESP8266 Module
* Flash Mode: QIO
* Flash Frequency: 80MHz
* Upload Using: Serial
* CPU Frequency: 160MHz
* Flash Size: 2M(1M SPIFFS)

koffienl, 2016-04-18:
Thanks for your feedback. Strange you couldn't reproduce.
The WDT is still bugging me, but I had other priority's in the past few weeks. Will look into into when I have a bit more time.

renearts, 2016-04-26:
A difference I spotted is that koffienl probably uses the ESP in 80MHz mode, while jakalada seems to be running it in 160MHz mode... naturally that does give it an edge. I was wondering whether koffienl tried removing some of the Serial.println out of the critical loops. Don't know the size of the ESPs serial buffer, but generally these aren't too big and Serial.println are generally quite slow. Especialy as there are quite a few large lines. Maybe replacing them with short, 2-4 character debug symbols might already help?

leader21, 2016-04-26:
i can confirm the findings of @jakalada
with your stripped down sketch there are no issues here too. have tested the same way.
have tested on nodemcu 1.0 running on 80mhz with 4mb flash, ide 1.6.5/BM2.0.0 and ArduinoJson 4.5

koffienl, 2016-05-10:
Thanks for feedback. Was away for holiday and hope to continue with this in the coming days

devyte, 2017-10-17:
Closing per #3655 .

closed 2017-10-17 by devyte