esp8266/Arduino #1840: sporadic wdt reset on custom hw

putyn, 2016-03-31:
### Basic Infos
Hi all and thanks for all the great work,
I am currently trying to debug a project but without much luck, maybe somone here has an idea  
#### Hardware
Hardware:			ESP-07 on custom board (i can provide schematic if needed)
Core Version:      	currently 2.1.0 but problem existed on 2.0 also 

### Description
the project is a clock with daisy chained displays each display is controlled by a shit register and the esp talks to the shift registers via a spi like interface (clock and data). It gets the time via ntp (once at boot) and keeps the time using the ticker  
 
Problem description
From time to time the display goes blank or freezes followed by a reset caused by the wdt  but the restarts are very sporadic sometimes i get 1-2 restarts per day sometimes more than a couple in a few minutes 
### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1MB/64k 
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch
https://gist.github.com/putyn/6654eb00fb0ecda4a5596e45c7f68ba2

### Debug Messages
most of the time at reboot only this message is printed 
```
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
```

but i got also a exception 
```
Decoding 8 results
0x40208d65: rc_get_trc at ?? line ?
0x4021263d: sta_input at ?? line ?
0x4021274f: sta_input at ?? line ?
0x40207bcb: pp_tx_idle_timeout at ?? line ?
0x402074f3: ppPeocessRxPktHdr at ?? line ?
0x402040e3: loop_task at C:\Users\Vlad\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/core_esp8266_main.cpp line 43
```

any suggestion would be much appreciated since in the last few days restarts keep happening quit frequent

putyn, 2016-04-04:
looks like it was more of a hw issue than a software one. 
My project required 12V, 5V, 3V3, not taking into account the power dissipation i used linear regulators to step down the voltages(12V-5V, 5v-3V3) and the regulator used to provide 5V dissipated a lot more power than i anticipated making it run very hot and because it was mounted on the bottom of the PCB with the ESP-07 on top it raised the temperature quite a lot making the ESP behave erratic.

Changing the LDO with a DC/DC regulator greatly reduced the power dissipation making the project run more stable 

So for anyone making projects with ESP devices take into account if you need to use voltage regulators and based on the voltages you need to step down from chose the appropriate one because of the power dissipation issues and of course for efficiency.

I hope this will save someone some debugging time and frustration.

Keep up the good work,
Vlad

closed 2016-04-04 by putyn

omaro2009, 2019-02-10:
I have the same problem I am using Nodemcu development kit all that i want to do is to send dweet a message could you help me