esp8266/Arduino #1841: Wifi upload speed

raduprv, 2016-03-31:
I am having trouble uploading data from an SD card. The upload speed is about 6.6KBps, about 1 meter away from the router.
This is the code:
`
  unsigned long start_time = millis();
  unsigned long wifi_start_time = millis();
  unsigned long wifi_time=0;
  
    while (fh.available()) {
      clientBuf[clientCount] = fh.read();
      clientCount++;
      if (clientCount > (bufSizeFTP - 1)) {
        //Serial.print(".");
        wifi_start_time=millis();
        dclient.write((const uint8_t *) &clientBuf[0], bufSizeFTP);
        wifi_time += millis()-wifi_start_time;
        clientCount = 0;
        delay(1);
      }
    }
    if (clientCount > 0) dclient.write((const uint8_t *) &clientBuf[0], clientCount);
`
The buffer size is 1460

Is this the max upload speed? Is there something I can do to improve it? Almost all of the time is spent in dclient.write()

raduprv, 2016-04-02:
Ok, I tried it on Linux, which has a different ACK behavior than Windows, and the upload speed was much better, at 30KBps. Still not really great, but that's about 0.24mbps.
Is there anything that can be done to improve this further?

igrr, 2016-04-04:
Can this be closed since you have opened #1853 on the same topic?

raduprv, 2016-04-04:
Yes, it can be closed now. If someone googles it, please refer to #1853 for the solution.

closed 2016-04-04 by igrr