esp8266/Arduino #1843: WiFi.scanNetworks() doesn't return the correct number of networks in range.

HW-Siew, 2016-04-01:
I tried the ESP8266WiFi -> WiFiScan example sketch. In the loop to display all available networks in range returned by WiFi.scanNetworks() shows that there are multiple duplicated SSID. How can this be avoided? 

Version used esp8266 arduino: 2.1.0
Arduino IDE :1.6.7

raduprv, 2016-04-02:
You can have multiple networks with the same name (and password, etc.). Are you sure this is not the case?
And if it is, then you should process the data returned yourself, to remove the duplicates.

HW-Siew, 2016-04-02:
I am certain that in my environment there are only 7 networks available with different SSID and password(some open network). However, when i scanned available networks with NodeMCU using WiFi,scanNetworks, the number varies every time. By listing down each SSIDs, i found a number of duplicated SSID in list.

tablatronix, 2016-04-08:
They are probably not duplicated they are probably real interfaces, print the BSSID() out and see their hardware addresses. Some might be channel switching, some might be 5ghz, some might expose multiple interfaces for guest access, or different channels, might explain the varying counts.

There is some code in the `wifimanager` library that does rssi sorting and duplicate removal you can steal.

tablatronix, 2016-04-08:
Here is my tester

```C/*
 *  This sketch demonstrates how to scan WiFi networks. 
 *  The API is almost the same as with the WiFi Shield library, 
 *  the most obvious difference being the different file you need to include:
 */
#include "ESP8266WiFi.h"

void setup() {
  Serial.begin(115200);
//   Serial.setDebugOutput(true);
  
  // Set WiFi to station mode and disconnect from an AP if it was previously connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();

  delay(100);

  Serial.println("");
  Serial.println("Setup done");
  Serial.println("");
  Serial.println("MAC: " + WiFi.macAddress());
}

String encryptionTypeStr(uint8_t authmode) {
    switch(authmode) {
        case ENC_TYPE_NONE:
            return "NONE";
        case ENC_TYPE_WEP:
            return "WEP";
        case ENC_TYPE_TKIP:
            return "TKIP";
        case ENC_TYPE_CCMP:
            return "CCMP";
        case ENC_TYPE_AUTO:
            return "AUTO";
        default:
            return "?";
    }
}

void loop() {
  //  Serial.println("scan start");

  // WiFi.scanNetworks will return the number of networks found
  int n = WiFi.scanNetworks(false,false);
  //  Serial.println("scan done");
  int o = n;
  int loops = 0;
  
  if (n == 0)
    Serial.println("no networks found");
  else
  {
    // sort by RSSI
    int indices[n];
    int skip[n];
    
    String ssid;
    
    for (int i = 0; i < n; i++) {
      indices[i] = i;
    }

    // CONFIG
    bool sortRSSI   = true; // sort aps by RSSI
    bool removeDups = false; // remove dup aps ( forces sort )
    bool printAPs   = true; // print found aps
    
    bool printAPFound = false; // do home ap check   
    const char* homeAP = "MYAP"; // check for this ap on each scan
    // --------

    bool homeAPFound   = false;
    
    if(removeDups || sortRSSI){
      for (int i = 0; i < n; i++) {
        ssid = WiFi.SSID(i);
        for (int j = i + 1; j < n; j++) {
          if (WiFi.RSSI(indices[j]) > WiFi.RSSI(indices[i])) {
            loops++;
            //int temp = indices[j];
            //indices[j] = indices[i];
            //indices[i] = temp;
            std::swap(indices[i], indices[j]);  
            std::swap(skip[i], skip[j]);  
          }
        }
      }
    }

    if(removeDups){
      for (int i = 0; i < n; i++) {
        if(indices[i] == -1){
          --o;
          continue;
        }
        ssid = WiFi.SSID(indices[i]);
        for (int j = i + 1; j < n; j++) {
          loops++;
          if (ssid == WiFi.SSID(indices[j])) {
            indices[j] = -1;
          }
        }
      }
    }

//    Serial.println((String)loops);
    Serial.print(o);
    Serial.println(" networks found of " + (String)n);

    Serial.println("00: (RSSI)[BSSID][hidden] SSID [channel] [encryption]");
    for (int i = 0; i < n; ++i)
    {
      if(printAPFound && (WiFi.SSID(indices[i]) == homeAP)) homeAPFound = true;
    
      if(printAPs && indices[i] != -1){
      // Print SSID and RSSI for each network found
      Serial.printf("%02d", i + 1);
      Serial.print(":");

      Serial.print(" (");
      Serial.print(WiFi.RSSI(indices[i]));
      Serial.print(")");
      
      Serial.print(" [");
      Serial.print(WiFi.BSSIDstr(indices[i]));
      Serial.print("]");
      
      Serial.print(" [");
      Serial.print((String) WiFi.isHidden(indices[i]));
      Serial.print("]");

      Serial.print(" " + WiFi.SSID(indices[i]));
      // Serial.print((WiFi.encryptionType(indices[i]) == ENC_TYPE_NONE)?" ":"*");
      
      Serial.print(" [");
      Serial.printf("%02d",(int)WiFi.channel(indices[i]));
      Serial.print("]");

      Serial.print(" [");
      Serial.print((String) encryptionTypeStr(WiFi.encryptionType(indices[i])));
      Serial.print("]");

//      Serial.print(" WiFi index: " + (String)indices[i]);

      Serial.println();
      }
      delay(10);
    }
     if(printAPFound && !homeAPFound) Serial.println("HOME AP NOT FOUND");
     Serial.println("");
  }

  // Wait a bit before scanning again
  delay(500);
}
```

closed 2016-04-18 by igrr

beicnet, 2017-07-31:
I got error message, saying:
`no matching function for call to 'ESP8266WiFiClass::encryptionType()'`

#include <ESP8266WiFi.h> included, running v2.3.0