esp8266/Arduino #1845: Scan networks don't work after WPS config failed

QuentinC, 2016-04-01:
### Basic Infos

#### Hardware
Hardware:			Sparkfun Thing (with 4Mbyte flash)
Core Version:      	Arduino 2.1.0

### Description

I'm trying to make an automatic Wifi management.
When I want to scan for networks, I have a FAILED error (-2) if it's done after a failed WPS config.

Below if the sample sketch for Wifi scan, with a WPS attempt at the end.
Scan only works once here, where it should always return some networks.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:		OTA / SERIAL (same problem for both)
Reset Method:		ck

### Sketch

```cpp

#include "ESP8266WiFi.h"

void setup() {
  Serial.begin(115200);

  // Set WiFi to station mode and disconnect from an AP if it was previously connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);

  Serial.println("Setup done");
}

void loop() {
  Serial.println("scan start");

  // WiFi.scanNetworks will return the number of networks found
  int n = WiFi.scanNetworks();
  Serial.println("scan done");
  if (n == 0)
    Serial.println("no networks found");
  else
  {
    Serial.print(n);
    Serial.println(" networks found");
    for (int i = 0; i < n; ++i)
    {
      // Print SSID and RSSI for each network found
      Serial.print(i + 1);
      Serial.print(": ");
      Serial.print(WiFi.SSID(i));
      Serial.print(" (");
      Serial.print(WiFi.RSSI(i));
      Serial.print(")");
      Serial.println((WiFi.encryptionType(i) == ENC_TYPE_NONE)?" ":"*");
      delay(10);
    }
  }
  Serial.println("");

  // Wait a bit before scanning again
  delay(5000);
  WiFi.beginWPSConfig();
  delay(5000);
}

```
### Serial output

```
Setup done                              
scan start                                                           
scan done                                           
5 networks found                          
1: ***** (-92)                   
2: ***** (-90)*                     
3: ***** (-80)*                                
4: ***** (-78)*                               
5: ***** (-79)*
                                                                          
scan start                                           
scan done                                           
-2 networks found
```

jbheren, 2016-07-21:
Same issue here, plus doing a `WiFi.begin(ssid, password);` after WPS failure will not work either.

jbheren, 2016-07-21:
OK, there must be a missing `wifi_wps_disable();` somewhere, **it works** for me when I Add it manually to the sketch. Issue is that beginWPSConfig thinks it succeded but it did not actually.

```
#include "ESP8266WiFi.h"

extern "C" {
  #include "user_interface.h"
}
...
void setup() {
  Serial.begin(74880);
  Serial.setDebugOutput(1);
  Serial.setDebugOutput(true);
  delay(10);

  // If not connected, run the WPS procedure.
  if(!connectWPS()) {
    Serial.println("WPS Failed");
    **wifi_wps_disable();**
  }
  // if all failed, connect to failback
  if (WiFi.status() != WL_CONNECTED) {
    Serial.println("Attempting connection to failback... ");
    WiFi.begin(ssid, password);
    while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
    }
  }
}

// https://gist.github.com/copa2/fcc718c6549721c210d614a325271389
bool connectWPS() {
  if (WiFi.status() != WL_CONNECTED) {
    Serial.println("\nAttempting WPS connection ...");
    bool wpsSuccess = WiFi.beginWPSConfig();
    if(wpsSuccess) {
      // Ckeck SSID
      String newSSID = WiFi.SSID();
      if(newSSID.length() > 0) {
        // WPSConfig has already connected in STA mode successfully to the new station.
        Serial.printf("WPS finished. Connected successfull to SSID '%s'\n", newSSID.c_str());
      } else {
        Serial.println("WiFi.beginWPSConfig() returns true BUT WiFi.SSID() is Empty");
        wpsSuccess = false;
      }
    }
    return wpsSuccess; 
  } 
}
```

see the log

> Attempting WPS connection ...
> wifi_wps_enable
> wps scan
> build public key start
> build public key finish
> f r0, scandone
> f r0, scandone
> f r0, scandone
> f r0, scandone
> f r0, scandone
> wps cb status: 4
> WiFi.beginWPSConfig() returns true BUT WiFi.SSID() is Empty
> WPS Failed
> **wifi_wps_disable**
> Attempting connection to failback... 
> f r0, ....scandone
> state: 0 -> 2 (b0)
> .state: 2 -> 3 (0)
> state: 3 -> 5 (10)
> add 0
> aid 3
> cnt 
> 
> connected with foobar, channel 1
> dhcp client start...
> chg_B1:-40
> ..chg_B1:-80
> .ip:192.168.1.157,mask:255.255.255.0,gw:192.168.1.1
> .Setup done
> 18:FE:34:A6:01:38
> force slp enable,type: 1
> Connected to foobar

QuentinC, 2016-07-24:
Hello !

I can confirm this ! Scan is working after wifi_wps_disable(); !

Thanks !

ricardoquesada, 2016-09-18:
would be great to apply the proposed patch. I'm having this issue as well.

closed 2018-01-02 by igrr