esp8266/Arduino #1846: Esp8266webserver connection reset when making a post request.

HW-Siew, 2016-04-02:
I am trying to connect a nodeMcu to wifi network by sending a post request with SSID and password from browser (AJAX) to Esp8266webserver. I confirmed that the server received the request with correct arguments(SSID and Password). However, occasionally the browser(chrome) debug console showed an error "net::ERR_CONNECTION_RESET". What could be the reason for that?

Javascript post request from browser:

```
var xhttp = new XMLHttpRequest(); 
xhttp.onreadystatechange = function(){
   if(xhttp.readyState == 4){
	if(xhttp.status==200){
		alert('Successfully connected');
	}else{
		alert('Unable to connect.'); // the server should return a 404
         };
   };
};
xhttp.open('POST', '/connect', true);
xhttp.setRequestHeader('Content-type', 'application/x-www-form-urlencoded');
xhttp.send('ssid='+ssid+'&'+'pswd='+pswd);

```
Server Post request handler :

```
WiFi.begin(ssid,pswd);
while(WiFi.status() != WL_CONNECTED){
  
// occasionally browser got the connection reset error while serer is still running this loop.
    Serial.print('.');  
      delay(500);
      if(WiFi.status() == WL_CONNECT_FAILED) break;
}

Server.sendHeader("Connection", "close");
Server.sendHeader("Access-Control-Allow-Origin",` "*");

if(WiFi.status ==  WL_CONNECTED){
// connected to wifi network
Server.send(200, "text/plain", "OK");
return;
}

// unable to connect to wifi network
Server.send(404, "text/plain", "fail");
```

Thank you.

HW-Siew, 2016-04-03:
I made a series of post requests in sequence with random interval inbetween. The following are the response i captured from browser (chrome) debug console. I got a number of status 0 - "failed"  responses and some even status pending, didn't receive data from server. Even though I confirmed the post handler on the nodeMcu had reached the end of function (I println something at the end of post handler). 

[image]

HW-Siew, 2016-04-03:
I narrowed down the potential cause. It has nothing to do with the Esp8266WebServer, instead, it is about the WiFiClient used in the library. WiFiClient seems to be reset after sometime while looping for the wifi network to be connected in this case. Any workaround for this?

igrr, 2016-04-03:
Which WiFi.mode setting do you use? Which interface (sta or AP) does your server run on?

HW-Siew, 2016-04-03:
My WiFi.mode is set to WIFI_AP_STA. The server is running on AP. 

Thanks for advice.

igrr, 2016-04-04:
What is happening to the STA end in your test? Is the STA
- trying to connect to an AP, but failing and retrying constantly 
- connected to AP, but disconnects sometimes
- connected steadily to an AP

HW-Siew, 2016-04-05:
Hi, @igrr. 

STA seems working fine. I can connect to a wifi network with correct SSID and password. However,  I encountered a peculiar case where the wifi.status didn't give the correct status as explained following.

In short, post request handler gets ssid and pswd from a request and connect to the wifi network. if successfully connected return status 200, else return status 400.

Code for post request handler :
```
WiFi.begin(ssid,pswd);
while(WiFi.status() != WL_CONNECTED && WiFi.status() != WL_CONNECT_FAILED){
      Serial.print('.');
      delay(500);
}
Serial.println();

// loop for connection
if(WiFi.status() == WL_CONNECTED){

      webServer.send(200, "text/plain", "OK");
      Serial.println("connected");
      return;
}

webServer.send(400, "text/plain", "fail");
Serial.println(WiFi.status());
```

case 1: ssid and pswd are correct.
result : "connected" --> expected result

case 2: first attempt, wrong ssid and pswd. second attemp correct ssid and pswd.
result : 
attempt 1 :   WL_CONNECT_FAILED --> expected result, however browser got a response status 0.
attempt 0 :   WL_CONNECT_FAILED --> unexpected result, since correct ssid and pswd are provided, and I confirmed that the STA mode is connected to the wifi network yet wifi status give incorrect result.

case 3: wrong ssid and pswd in several attempts, browser response status varies between 0 and 400.
result : I noticed that whenever the "loop for connection" takes longer time, the response status will be 0. I am wondering would it be because of the timeout of WifiClient. As mentioned in #1570, the current default timeout is 1 second.

Thank you for your advice.

devyte, 2017-10-17:
@HW-Siew I don't see the SoftAP setup in your code. The SoftAP needs to be set to the same channel as the AP to which the STA interface is connected to. The ESP has only one radio. If the STA connect to an AP on one channel, and the SoftAP is set up on a different channel, then the radio may not behave correctly (may jump back and forth).
Closing as user error. If there is still a problem, please open a new issue and add the requested details, including a MCVE sketch.

closed 2017-10-17 by devyte