esp8266/Arduino #1847: HelloMesh example makes the mysterious reset

jakalada, 2016-04-02:
### Basic Infos

#### Hardware
Hardware: ESP-WROOM-02
Core Version: 2.1.0

### Description

[HelloMesh](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFiMesh/examples/HelloMesh/HelloMesh.ino) sketch makes the mysterious reset after few minutes.

### Settings in IDE

Module: Generic ESP8266 Module
Flash Mode: QIO
Flash Frequency: 80MHz
Upload Using: Serial
CPU Frequency: 80MHz
Flash Size: 1MB (512K SPIFFS)
Debug port: Serial
Debug level: Core + WiFi
Reset Method: nodemcu

### Debug Messages

Serial outputs.
Other node is none.

```
Setting up mesh node...
[AP] softap config unchanged
f r0scandone
wifi evt: 7
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
wifi evt: 7
wifi evt: 7
f r0, scandone
f r0, scandone
f r0, scandone
f r0, wifi evt: 7
wifi evt: 7
wifi evt: 7
scandone
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, wifi evt: 7
scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, scandone
f r0, 
 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

jakalada, 2016-04-02:
I found the line of the reset cause by the print debug.

1. call `mesh_node.attemptScan(request);`([line](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFiMesh/examples/HelloMesh/HelloMesh.ino#L51)) in sketch.
2. call `int n = WiFi.scanNetworks();`([line](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFiMesh/src/ESP8266WiFiMesh.cpp#L128)) in attemptScan method.
3. call `esp_yield();`([line](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiScan.cpp#L97)) in scanNetworks method.

igrr, 2016-04-04:
/cc @jtfell

gguuss, 2016-07-26:
I added a few delay(0) calls in scan and in mesh and it has fixed the crash for me. If [this commit](https://github.com/gguuss/Arduino/commit/0cbeadfc66cc826fe349a23d0df4c34798103cfa), fixes the reset / crash LMK and I'll submit a PR.

devyte, 2017-10-17:
@jakalada is this issue still valid with latest git? Is the PR still relevant?

jakalada, 2017-11-04:
@devyte I don't have an environment of checking the Wi-Fi Mesh now. Should I close this issue?