esp8266/Arduino #1848: Licensing a project using ESP8266 Arduino libraries, examples

Nurgak, 2016-04-02:
I'd like to publish a project where I used the libraries and examples developed within this project. I made extensive use of the examples which I modified for my needs and have to include with my code.

I'm not clear on how I should licence my project now, for example: am I allowed to choose the MIT licence, or because of the dependancies I need to use the LGPL? I'd like to use the most permissive licence possible. The [licensing section](https://github.com/esp8266/Arduino#license-and-credits) doesn't help much.

It'd be nice to have a licensing section here describing how the derivative works should be licensed. I'm an engineer, not a lawyer, no amount of Googleing can help me.

Nurgak, 2016-04-03:
I found the answer myself: as long as my code is a derivative, and using the example code counts as derivative, [I need to adopt the same licence as the original work (LGPL)](https://en.wikibooks.org/wiki/FOSS_Licensing/How_to_Make_the_Source_Free_or_Open).

Using tools to create the binary does not count as derivative work, so my work does not need to comply with the GPL licence of Arduino IDE, Xtensa gcc toolchain and Esptool.

closed 2016-04-03 by Nurgak

igrr, 2016-04-03:
Not a lawyer here as well.
You can license your code under MIT, if you like, but when you are distributing it, you will have to comply with all licenses involved.
For instance, if you provide binaries for download, you will have to make sure the end user can exercise the rights given by LGPL with respect to library code.
In practice this means that you need to publish object files of your application to allow re-linking against LGPL-licensed library. (If you do plan to publish the source code, then this isn't an issue)