esp8266/Arduino #1852: fatal error: Hash.h: No such file or directory

vitaminwater, 2016-04-03:
Hi,

I can't seem to get the esp8266 to work with WebSockets library.

here is my platform.ini:

```
#
# Project Configuration File
#
# A detailed documentation with the EXAMPLES is located here:
# http://docs.platformio.org/en/latest/projectconf.html
#

# A sign `#` at the beginning of the line indicates a comment
# Comment lines are ignored.

# Simple and base environment
# [env:mybaseenv]
# platform = %INSTALLED_PLATFORM_NAME_HERE%
# framework =
# board =
#
# Automatic targets - enable auto-uploading
# targets = upload

[env:esp8266]
platform = espressif
framework = arduino
board = nodemcu

```

I've installed the WebSockets library with:

```
platformio lib install 549
```

and on a:

```
platformio run
```

It ends up with a:

```
fatal error: Hash.h: No such file or directory
```

vitaminwater, 2016-04-03:
And if I try to put `#include <Hash.h>` directly in my main.cpp, I get:

```
In file included from src/main.cpp:1:0:                                                                                                                                                                                  [18/1070]
.pioenvs/esp8266/Hash/Hash.h:30:11: error: variable or field 'sha1' declared void
void sha1(uint8_t * data, uint32_t size, uint8_t hash[20]);
^
.pioenvs/esp8266/Hash/Hash.h:30:11: error: 'uint8_t' was not declared in this scope
.pioenvs/esp8266/Hash/Hash.h:30:21: error: 'data' was not declared in this scope
void sha1(uint8_t * data, uint32_t size, uint8_t hash[20]);
^
.pioenvs/esp8266/Hash/Hash.h:30:27: error: 'uint32_t' was not declared in this scope
void sha1(uint8_t * data, uint32_t size, uint8_t hash[20]);
^
.pioenvs/esp8266/Hash/Hash.h:30:42: error: 'uint8_t' was not declared in this scope
void sha1(uint8_t * data, uint32_t size, uint8_t hash[20]);
^
.pioenvs/esp8266/Hash/Hash.h:31:24: error: 'uint32_t' has not been declared
void sha1(char * data, uint32_t size, uint8_t hash[20]);
^
.pioenvs/esp8266/Hash/Hash.h:31:39: error: 'uint8_t' has not been declared
void sha1(char * data, uint32_t size, uint8_t hash[20]);
^
.pioenvs/esp8266/Hash/Hash.h:32:17: error: 'uint8_t' does not name a type
void sha1(const uint8_t * data, uint32_t size, uint8_t hash[20]);
^
.pioenvs/esp8266/Hash/Hash.h:32:27: error: ISO C++ forbids declaration of 'data' with no type [-fpermissive]
void sha1(const uint8_t * data, uint32_t size, uint8_t hash[20]);
^
.pioenvs/esp8266/Hash/Hash.h:32:33: error: 'uint32_t' has not been declared
void sha1(const uint8_t * data, uint32_t size, uint8_t hash[20]);
^
.pioenvs/esp8266/Hash/Hash.h:32:48: error: 'uint8_t' has not been declared
void sha1(const uint8_t * data, uint32_t size, uint8_t hash[20]);
^
.pioenvs/esp8266/Hash/Hash.h:33:30: error: 'uint32_t' has not been declared
void sha1(const char * data, uint32_t size, uint8_t hash[20]);
^
.pioenvs/esp8266/Hash/Hash.h:33:45: error: 'uint8_t' has not been declared
void sha1(const char * data, uint32_t size, uint8_t hash[20]);
^
.pioenvs/esp8266/Hash/Hash.h:34:11: error: variable or field 'sha1' declared void
void sha1(String data, uint8_t hash[20]);
^
.pioenvs/esp8266/Hash/Hash.h:34:11: error: 'String' was not declared in this scope
.pioenvs/esp8266/Hash/Hash.h:34:24: error: 'uint8_t' was not declared in this scope
void sha1(String data, uint8_t hash[20]);
^
.pioenvs/esp8266/Hash/Hash.h:36:1: error: 'String' does not name a type
String sha1(uint8_t* data, uint32_t size);
^
.pioenvs/esp8266/Hash/Hash.h:37:1: error: 'String' does not name a type
String sha1(char* data, uint32_t size);
^
.pioenvs/esp8266/Hash/Hash.h:38:1: error: 'String' does not name a type
String sha1(const uint8_t* data, uint32_t size);
^
.pioenvs/esp8266/Hash/Hash.h:39:1: error: 'String' does not name a type
String sha1(const char* data, uint32_t size)

[...]
```

vitaminwater, 2016-04-03:
Got it to work by adding both lines:

```
#include <Arduino.h>
#include <Hash.h>
```

Is that normal ?

igrr, 2016-04-04:
Unlike Arduino IDE, Platformio doesn't add `#include <Arduino.h>` line automatically for you.

closed 2016-04-04 by igrr