esp8266/Arduino #1856: exception(2) with WiFi.begin(...)

wkraus, 2016-04-04:
Arduino  1.6.8
Core Version:      	2.1.0

Settings in IDE

Module:  			WeMos D1
Flash Size: 		?MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			??
Flash Frequency:	??
Upload Using:		SERIAL
Reset Method:		??

Hello,

I am using WeMos D1 board with Arduino 1.6.8 and Software 2.1.0.
The board works nice with blink also WiFi.ScanNetworks works and gives a good result.

But whenever I use - WiFi,begin(...,...); - I get a runtime error:

Exception (2):
epc1=0x3ffe8bfc epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffe8bfc depc=0x00000000

ctx: cont
sp: 3fff0920 end: 3fff0e00 offset: 01a0
followed by a whole stack dump.

I tried different examples with WiFi.begin to connect, always the same problem.
Changing to board software 2.0.0 same problem.

Any help?

raduprv, 2016-04-04:
Post your code

wkraus, 2016-04-04:
#include <ESP8266WiFi.h>

const char* ssid     = "xxxxx";
const char* password = "xxxxx";

void setup(void)
{
  delay(3000);
  Serial.begin(115200);
  Serial.println("WeMos D1 Board");

  Serial.println("try to connect");
  WiFi.begin(ssid,password);
  Serial.print("Connecting to ");

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to: ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.println("");
}

void loop(void)
{
}

Printout after Boot:

WeMos D1 Board
try to connect

Exception (2):
epc1=0x3ffe8444 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffe8444 depc=0x00000000

ctx: cont 
sp: 3ffef7d0 end: 3ffefca0 offset: 01a0

>>>stack>>>
3ffef970:  feefeffe feefeffe feefeffe feefeffe  
3ffef980:  feefeffe feefeffe feefeffe feefeffe  
3ffef990:  feefeffe feefeffe feefeffe feefeffe  
3ffef9a0:  feefeffe feefeffe feefeffe feefeffe  
3ffef9b0:  feefeffe feefeffe feefeffe feefeffe  
3ffef9c0:  feefeffe 00000004 3ffefa20 3ffefa31  
3ffef9d0:  40107020 00000000 3fff0da8 00000484  
3ffef9e0:  3ffefbc7 feefeffe 000000a5 feefeffe  
3ffef9f0:  feefeffe 00000000 00000000 3ffef980  
3ffefa00:  feefeffe feefeffe feefeffe feefeffe  
3ffefa10:  3ffefa30 3ffefa20 00000004 feefeffe  
3ffefa20:  feefeffe 3ffe8550 3fff0da8 00000031  
3ffefa30:  000000a5 3fff0924 feefeffe feefeffe  
3ffefa40:  feefeffe feefeffe feefeffe 4010031d  
3ffefa50:  feefeffe 00000c28 3ffeec24 40100426  
3ffefa60:  feefeffe feefeffe feefeffe 40100537  
3ffefa70:  feefeffe 00001000 00000484 401006f5  
3ffefa80:  00000484 00001000 000003fd 4010724c  
3ffefa90:  40004b31 3ffefac0 0000001c 4022427d  
3ffefaa0:  40105ea2 40224365 3fff0924 000003ff  
3ffefab0:  000003fd 3ffefbc7 3fff0924 000003fd  
3ffefac0:  ffffff00 55aa55aa 00000014 0000001c  
3ffefad0:  0000001c 0000000b 0000000b 000003ff  
3ffefae0:  40224754 3fff0924 3fff0924 000000ff  
3ffefaf0:  00000001 3ffefbe7 402248af 00000008  
3ffefb00:  3fff0924 000000ff 3ffefbc7 00000000  
3ffefb10:  3fff09e4 3ffefc28 00000001 4022493c  
[... 36 lines cut ...]
0x40100426: check_poison_block at C:\Users\Werner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266\umm_malloc/umm_malloc.c line 851
0x40100537: check_poison_all_blocks at C:\Users\Werner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266\umm_malloc/umm_malloc.c line 880
0x401006f5: calloc at C:\Users\Werner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266\umm_malloc/umm_malloc.c line 1667
0x4010724c: pvPortZalloc at C:\Users\Werner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/heap.c line 33
0x4022427d: wifi_param_save_protect_with_check at ?? line ?
0x40105ea2: spi_flash_read at ?? line ?
0x40224365: system_param_save_with_protect at ?? line ?
0x40224754: wifi_station_ap_number_set at ?? line ?
0x402248af: wifi_station_ap_number_set at ?? line ?
0x4022493c: wifi_station_ap_number_set at ?? line ?
0x40224978: wifi_station_set_config at ?? line ?
0x402027ba: ESP8266WiFiSTAClass::begin(char const*, char const*, int, unsigned char const*, bool) at C:\Users\Werner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp line 635
0x402026fa: ESP8266WiFiSTAClass::begin(char const*, char const*, int, unsigned char const*, bool) at C:\Users\Werner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp line 635
0x401006d8: malloc at C:\Users\Werner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266\umm_malloc/umm_malloc.c line 1658
0x402021bd: uart_init at C:\Users\Werner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/uart.c line 336
0x40202d64: esp_yield at C:\Users\Werner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/core_esp8266_main.cpp line 43
0x40224196: wifi_set_broadcast_if at ?? line ?
0x402035e0: Print::write(unsigned char const*, unsigned int) at C:\Users\Werner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/Print.cpp line 42
0x40202cc1: Print::write(char const*) at C:\Users\Werner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/Print.h line 60
0x40202cc1: Print::write(char const*) at C:\Users\Werner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/Print.h line 60
0x402036b8: Print::println() at C:\Users\Werner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/Print.cpp line 144
0x402010ae: delay at C:\Users\Werner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/core_esp8266_wiring.c line 70
0x40202410: setup at C:\Users\Werner\AppData\Local\Temp\arduino_modified_sketch_602007/D1_DHT_Server_kurz.ino line 15
0x40202da4: loop_wrapper at C:\Users\Werner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/core_esp8266_main.cpp line 43
0x40100958: cont_norm at C:\Users\Werner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/cont.S line 109

raduprv, 2016-04-04:
Hmm, looks ok, but try a WiFi.disconnect(); first, just in case.
Are you sure the ssid and password are correct? Did you try to connect to a different network? Like, maybe create one with your phone.

wkraus, 2016-04-04:
I am sure ssi and password are correct. Just in case they are not, would that cause such a reaction?
Didn't try a different network yet but that's an interesting idea. Will think about that.

With Disconnect same problem.

Code (only setup):
void setup(void)
{
  delay(3000);
  Serial.begin(115200);
  Serial.println("WeMos D1 Board");
  Serial.println("disconnect");
  WiFi.disconnect();

  Serial.println("try to connect");
  WiFi.begin(ssid,password);
  Serial.print("Connecting to ");

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to: ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.println("");
}

reaction:
WeMos D1 Board
disconnect

Exception (2):
epc1=0x3ffe8450 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffe8450 depc=0x00000000

ctx: cont 
sp: 3ffef860 end: 3ffefcb0 offset: 01a0

>>>stack>>>
3ffefa00:  feefeffe feefeffe feefeffe feefeffe  
3ffefa10:  feefeffe feefeffe feefeffe feefeffe  
.....
can send the rest if important.

raduprv, 2016-04-04:
If it happens even with WiFi.disconnect();, then I guess it is either a hardware problem, or maybe that board is not properly supported?
is it properly powered and everything?

wkraus, 2016-04-04:
Who knows. I got this board from China last week, it is powered via the USB line and all the code I tried worked. Even ScanNetwork works and I can receive the transmit signal of its WIFi on my phone. I don't know how I could give different power to the board. I tried a different USB entry to my PC - same problem.

wkraus, 2016-04-04:
What do You mean with 'not properly supported'?

raduprv, 2016-04-04:
I don't know, I mean maybe that board does something a bit different than the other boards, who knows? I think maybe contact the manufacturer? Or see if there are forums dedicated to it?

wkraus, 2016-04-04:
Thank you for your help!
Seems the board is not original WeMos. I am waiting for a D1 mini I ordered directly from WeMos and hope it will work better.

raduprv, 2016-04-07:
Well, maybe you should try to add a cap between +5v and ground, there might be some spikes or something. I have a supercap with a voltage booster, and when I power my MCU from it, it dies when the wifi turns on. I guess the booster can't handle the spikes.

If you have an oscilloscope, it would be a good idea to see what happens with the voltage when the radio starts.

wkraus, 2016-04-07:
@raduprv   I added 2 caps of 100 uF to 5v and 3.3v - no change. Don't have an oscilloscope, sorry. 
But I found this post:
http://forum.wemos.cc/topic/99/my-solution-to-very-unstable-wifi
Adding 
 -     WiFi.persistent(false)  
made my board connecting.
Thanks for your help.

tablatronix, 2016-04-08:
There seems to be a lot of times where the flash gets corrupted by persistent saves, maybe your flash size is set wrong in ide? 

Which might explain why this fixes it.

There are several threads of similar problems, usually manually erasing flash via esptool before uploading fixes it. Cause is unknown.

Could also be insufficient power supply.

wkraus, 2016-04-08:
@tablatronix  My board is rather new to me, just tested it with some blink sketches like CW blink. I don't think there was excessive use of RAM. I thought about using esptool to erase RAM but it needs a lot of additional software to run it.

Using WeMos D1 board powered by USB. 

I am content to have a running board now, but  like You, I would like to know what was the reason of the whole problem.

tablatronix, 2016-04-08:
There is no software, you just run esptool from the hardware/esp8266/tools/ folder

I dont have the erase commands on hand bit they are posted here you can search them.

Flash ram not memory ram, the ssid config gets stored in flash and can get corrupted.
I mean make sure you have the right flash chip size selected.

smj135, 2016-04-19:
@wkraus 

I had the almost exact same problem, and for me the solution was to add `WiFi.persistent(false);` in your setup()

hope this helps you too

source: http://forum.wemos.cc/topic/99/my-solution-to-very-unstable-wifi

tablatronix, 2016-04-19:
That prevents saving the config, which is what causes the corruption, possibly an undiagnosed bug.
So a bit of a workaround.

igrr, 2016-05-10:
Tracked same issue in #1997, closing this one.

closed 2016-05-10 by igrr

PanwarParamveer, 2020-05-09:
ESP.eraseConfig(); <--- use this before begin
  WiFi.begin("", "");

hvilela0, 2021-02-15:
> @raduprv I added 2 caps of 100 uF to 5v and 3.3v - no change. Don't have an oscilloscope, sorry.
> But I found this post:
> http://forum.wemos.cc/topic/99/my-solution-to-very-unstable-wifi
> Adding
> 
> * ```
>   WiFi.persistent(false)  
>   ```
>   
>   
>   made my board connecting.
>   Thanks for your help.

This just solved the problem for me! Thanks!
just to help: I put this 'persistent' line at the line right above the  WiFi.mode(WIFI_AP);

```  
  WiFi.persistent(false);
  WiFi.mode(WIFI_AP);
  WiFi.config(ip, gateway, subnet);
  WiFi.begin(ssid, passwd);
```