esp8266/Arduino #1864: error after upgrading to version 2.1.0 of ESP8266 package

wddean, 2016-04-05:
I'm using the "Sparkfun ESP8266 Thing Dev" board and sporadically my sketch crashes after upgrading to version 2.1.0. Downgrading to version 2.0 fixes the issue.

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			Sparkfun ESP8266 Thing Dev
Core Version:      	

### Description

Problem description

### Settings in IDE

Module:  			Generic ESP8266 Module or Sparkfun Thing Dev
Flash Size: 		512K (64K SPIFFS)
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Upload Using:		 SERIAL
Reset Method:		ck

### Sketch

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <Servo.h>

#define noMotion 0
#define Motion 1

// Extra features ---------------------------------------
bool Motion_enabled = true;
bool LED_enabled = true;

// Indicators -------------------------------------------------
const int PIR_pin = 5; // Motion sensor pin
const int LED_pin = 4; // Flashing LED pin
const int motionIndicatorLED = 13; // Indicator LED

int lastState = noMotion;
bool firstTime = true;

Servo myservo;
int pos = 0; // variable to store the servo position
const int ServoPin = 2; // Set to GPIO2

const char* ssid = ""; // Your Wifi Name
const char* password = ""; // Wifi Password

const char* host = "api.instagram.com";
const int httpsPort = 443;

const char* hashtag = ""; // Your Hashtag goes here

//Your Instagram Access Token goes here
const char* access_token = ""; // Your instagram access token

// Initilize the counters to 0
int NewMCount = 0;
int LastMCount = 0;

//---------------------------------------
void RotateServo() {
  if (NewMCount > LastMCount) {
    if (LED_enabled) {
      Serial.println("Flashing LED...");
      digitalWrite(LED_pin, HIGH);
    }

    Serial.println("Rotating Servo from 0 degrees to 180 degrees");
    for (pos = 0; pos < 180; pos += 1) // goes from 0 degrees to 180 degrees
    { // in steps of 1 degree
      myservo.write(pos);              // tell servo to go to position in variable 'pos'
      delay(15);                       // waits 15ms for the servo to reach the position
    }
    Serial.println("Rotating Servo from 180 degrees to 0 degrees");
    for (pos = 180; pos >= 1; pos -= 1) // goes from 180 degrees to 0 degrees
    {
      myservo.write(pos);              // tell servo to go to position in variable 'pos'
      delay(15);
    }

    if (LED_enabled) {
      digitalWrite(LED_pin, LOW);
      Serial.println("Turned off the LED...");
    }

  }
}

//---------------------------------------
void setup() {
  Serial.setDebugOutput(true);
  pinMode(LED_pin, OUTPUT);
  pinMode(motionIndicatorLED, OUTPUT);
  
  Serial.begin(115200);
  Serial.println("Starting...");
  Serial.println("Connecting to Wifi");
  //Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println();
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  myservo.attach(ServoPin);

  Serial.print("Motion feature: ");
  if (Motion_enabled) {
    Serial.println("Enabled");
  }
  else {
    Serial.println("Disabled");
  }

  Serial.print("Flashing LED feature: ");
  if (LED_enabled) {
    Serial.println("Enabled");
  }
  else {
    Serial.println("Disabled");
  }

  digitalWrite(LED_pin, LOW);
  digitalWrite(motionIndicatorLED, LOW);
  Serial.println("Setup complete.\n");

}

//---------------------------------------
void loop() {

  if (Motion_enabled) {
    if (!digitalRead(PIR_pin)) {
      if (lastState == Motion) {
        Serial.println("No motion detected...");
        Serial.println("Disabling the code...");
      }
      lastState = noMotion;
      digitalWrite(motionIndicatorLED, LOW);
      return;
    }
    else {
      if (lastState == noMotion) {
        Serial.println("Motion detected...");
        Serial.println("Activating the code...");
      }
      digitalWrite(motionIndicatorLED, HIGH);
      lastState = Motion;
    }
  }

  WiFiClientSecure client;

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(host);
  if (!client.connect(host, httpsPort)) {
    Serial.println("Connection failed");
    return;
  }

  String url = "/v1/tags/";
  url += hashtag;
  url += "?access_token=";
  url += access_token;

  Serial.println("Requesting Data..");

  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" +
               "User-Agent: ESP8266\r\n" +
               "Connection: close\r\n\r\n");

  // Parse http headers here
  while (client.connected()) {
    String line = client.readStringUntil('\n');
    if (line == "\r") {
      break;
    }
  }
  // Read data here
  String line = "";
  while (client.available()) {
    char c = client.read();
    line += c;
  }

  const String header = "{\"meta\":{\"code\":200},\"data\":{\"media_count\":";
  line = strstr(line.c_str() , header.c_str());

  // If everything is OK
  if (line.startsWith(header)) {
    int i = header.length();
    while (i <= line.length()) {
      if (line.charAt(i) == ',') break;
      i++;
    }
    LastMCount = NewMCount;
    NewMCount = line.substring(header.length(), i).toInt();
    Serial.print("\nNewMCount = ");
    Serial.println(NewMCount);
    Serial.print("LastMCount = ");
    Serial.println(LastMCount);

    if (firstTime) {
      firstTime = false;
      return;
    }
    RotateServo(); // Rotates the servo motor
  }
  else // Error
    Serial.println("\nUnexpected response ...");

  delay(5000);
}

### Debug Messages

Here is the error -

Exception (0):
epc1=0x40203d0c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffc40 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffde0: ffffffd3 00000000 40240000 ffffff1a 
3ffffdf0: 3fffc200 40107288 00000020 401072a4 
3ffffe00: 4020e576 3fff16b4 3fffc248 4000050c 
3ffffe10: 40004384 00000030 00000019 ffffffff 
3ffffe20: 60000200 00000004 ffffffff 80000000 
3ffffe30: 20000000 3fff4220 80000000 2007c1c0 
3ffffe40: 00000000 3fffc6fc fc70ffff 3fff4224 
3ffffe50: 00000134 0007c1c0 60000600 00000030 
3ffffe60: 3ffffe70 3ffeaa44 3ffeaa44 00000001 
3ffffe70: 40236afe 00000030 00000020 ffffffff 
3ffffe80: 40236ae4 80000002 00000002 fffff000 
3ffffe90: 00000000 00000000 0000001f 40105b55 
3ffffea0: 4000050c 00000005 3fffc248 4000050c 
3ffffeb0: 40000f68 00000030 0000001a ffffffff 
3ffffec0: 40000f58 00000000 00000020 00000000 
3ffffed0: 0038ac7f 00000001 00000000 00000000 
3ffffee0: ffffffff 3fffc6fc fc70ffff 3fffdab0 
3ffffef0: 00000000 3fffdcb0 3ffefa40 00000030 
3fffff00: 00000000 400042db 3ffebd91 60000600 
3fffff10: 40004b31 3fff4054 000002f4 0007c000 
3fffff20: 40105ea2 3ffefa30 3ffee440 401075f8 
3fffff30: 4020abfd 3ffee440 3ffefa30 00b60466 
3fffff40: 3fff4054 00001000 4020b092 00000008 
3fffff50: 4021d9c4 3ffe9fd4 4020b13f 3ffee4f4 
3fffff60: 3ffefa30 40224af9 3ffefa08 3ffefa30 
3fffff70: 40224af9 60000600 3ffea218 3ffea218 
3fffff80: 40224b3e 3fffdab0 00000000 3fffdcb0 
3fffff90: 3ffefa48 00000000 40000f65 3fffdab0 
3fffffa0: 40000f49 000182e1 3fffdab0 40000f49 
<<<stack<<<

ets Jan 8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

igrr, 2016-04-05:
Could you please decode the stack dump and paste the result here? Thanks.
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

wddean, 2016-04-05:
Decoding 29 results
0x40107288: timer0_isr_handler at C:\Users\Del\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/core_esp8266_timer.c line 71
0x401072a4: timer0_isr_handler at C:\Users\Del\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/core_esp8266_timer.c line 71
0x4020e576: ieee80211_ht_updateparams at ?? line ?
0x40102dca: wDev_MacTim1Arm at ?? line ?
0x40102e2e: wDev_ProcessFiq at ?? line ?
0x40105b55: ets_timer_disarm at ?? line ?
0x40105ea2: spi_flash_read at ?? line ?
0x401075f8: pvPortZalloc at C:\Users\Del\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/heap.c line 33
0x4020abfd: pm_set_sleep_time at ?? line ?
0x4020b092: pm_get_sleep_type at ?? line ?
0x4021d9c4: sys_check_timeouts at ?? line ?
0x4020b13f: pm_get_sleep_type at ?? line ?
0x40224af9: ets_timer_handler_isr at ?? line ?
0x40224af9: ets_timer_handler_isr at ?? line ?
0x40224b3e: ets_timer_handler_isr at ?? line ?

hakeemta, 2016-04-07:
I face similar issues. Though I have been thinking I have issues with my program. Since simple sample program like blink LED works. I have been facing the sketch crashing since I upgraded to 2.1.0.

Please assist. Thank you.

igrr, 2016-04-16:
Closing as duplicate of #1742.

closed 2016-04-16 by igrr