esp8266/Arduino #1867: std::min not declared here?

WereCatf, 2016-04-06:
I was playing around, took the graphicstest.ino from the Adafruit_ILI9341 - library that's modified for use with the ESP8266 and added `#include "ESP8266WiFi.h` -- as soon as that include is added the sketch won't compile anymore, complaining about std::min not being declared here

I ain't used to C++ and its quirks and I can't say I understand what's happening. Why does simply including that header-file break code, like e.g. `n     = min(tft.width(), tft.height());`, that works fine as soon as I remove the include? 

I am using IDE 1.6.8 and latest github-code for the ESP.

```
/home/nita/Arduino/ew/graphicstest/graphicstest.ino: In function 'long unsigned int testRects(uint16_t)':
graphicstest:229: error: 'min' was not declared in this scope
   n     = min(tft.width(), tft.height());
                                        ^
/home/nita/Arduino/ew/graphicstest/graphicstest.ino:229:40: note: suggested alternative:
In file included from /home/nita/Työpöytä/projects/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/memory:62:0,
                 from /home/nita/Työpöytä/arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.h:24,
                 from /home/nita/Työpöytä/arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h:39,
                 from /home/nita/Arduino/ew/graphicstest/graphicstest.ino:17:
/home/nita/Työpöytä/projects/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algobase.h:239:5: note:   'std::min'
     min(const _Tp& __a, const _Tp& __b, _Compare __comp)
     ^
```

marvinroger, 2016-04-07:
See #398, you will understand.

WereCatf, 2016-04-07:
@marvinroger I see, so using _min and _max is a workaround for now.

Well, I'll try and keep that in mind. Spanks!

closed 2016-04-07 by WereCatf