esp8266/Arduino #1868: Can't flash wroom anymore after using TX and RX as GPIO

Banzai00, 2016-04-06:
Hi guys, i have a Wroom 02, i wanted to use TX RX as gpio so uploaded a sketch where i set gpio 1 and gpio 3 as output. the code didn't work but that's not my problem.
The problem is that now i can't upload any other program and i get the message espcomm failed.
I tried different codes with an another wroom 02  and it's working fine so there's nothing wrong with my configuration and equipments.

So can i repair the first wroom or there's no solution ??

duncan-a, 2016-04-07:
I've just been trying to use TX/RX for I2C SCL/SDA on an ESP-01, and I couldn't get it to work either.

However, reverting to GPIO-0/2 threw up no more problems with uploading than I usually experience - i.e. nothing was 'bricked' on the ESP-01...

tablatronix, 2016-04-08:
erase_flash via esptool

Banzai00, 2016-04-11:
tablatronix i tried multiple flash tools but i get the same result : failed to connect.

tablatronix, 2016-04-11:
Are you able to get into flash boot mode manually? Do you get any serial output on reset?

Banzai00, 2016-04-11:
No, no response i cant' acess flash boot mode

philippe-gregoire, 2016-04-12:
Any possibility that you'd have inadvertently applied +5V to those pins?

Banzai00, 2016-04-14:
No i connected them to leds.
i think there's no hope and i'm gonna buy another one.

duncan-a, 2016-04-14:
What value current- limiting resistors did you use?

We're you sourcing current to the LEDs or sinking it?

Are the LEDs still connected when you are trying to upload new code to the
device?

philippe-gregoire, 2016-04-14:
Good point. Even output could toast GPIO (aren't they protected against this?)

Would there be a difference sinking (e.g. connecting LED to VCC, and having LED ON on Low GPIO ) or sourcing (e.g. connecting LED to GND and thus having LED ON on High GPIO)?

Thanks.

tablatronix, 2016-04-14:
They are rated at 20ma, a straight led could fry them i suppose. You didn't accidently conect them to a 5v supply did you ? That could also do it.

Some say you might try external pullups? But I would see about taking some measurements on the pins first.

duncan-a, 2016-04-14:
External pullups would eliminate the possibility of fried internal
pullups/pulldowns and provide known levels to the pins that need them, and
thus ensure correct bootup.

Obviously, that doesn't apply to TX and Rx - these should be direct to the
'programming lines', without LEDs, or anything else, pulling them in one
direction or the other.

The "20mA" is, I think, the quoted maximum current - much safer to use
less. Other than blue LEDs, which often have a Vf in excess of 3.3V, 330R
or 220R resistors work well in most circumstances.

Banzai00, 2016-04-15:
I use 1 kohm current- limiting resistors and it's working fine with other pins.
I'm sourcing current to the LED and the leds are no more connected when i am trying to upload the code.
tablatronix, philippe-gregoire : i didn't  applied +5V to those pins. i have a regulated 3.3v voltage supply .
thank you guys for your help and sorry for my bad english.

arlaor, 2016-04-22:
Do you have disconnected the LEDs RX and TX pins when you try to upload the code ?.
I had the same problem and I did was turn off the LEDs of these ports and then allowed me to program the board

closed 2016-07-22 by Banzai00