esp8266/Arduino #1869: Help - ESP8266WebServer or WiFiServer librarie ?

odilonafonso, 2016-04-06:
### Basic Infos

#### Hardware
Hardware:		ESP-8266 General
Core Version:      	I dont know how to see the core version

### Description

I dont know what librarie use: ESP8266WebServer or WiFiServer.
What is the difference ?
Why are there two libraries with such similar functions?
I have seen examples that use both libraries and apparently work the same way.

### Doc

There is an official documentation on ESP8266 libraries? I have used this link:
http://links2004.github.io/Arduino/index.html

The heritage of WiFiServer class is the Print class (??)
http://links2004.github.io/Arduino/da/de6/class_wi_fi_server.html
Print --> Server --> WiFiServer ???

Already ESP8266WebServer class is a primary class ...

In the documentation I find no reference to ESP8266WiFi.h library, but I have to include it in the sketch.

Finally, should I upgrade both my IDE as the ESP8266 libraries involved?
I am using the Arduino IDE 1.6.7 and version of ESP8266 I do not know where to check.

Grateful for any help!

igrr, 2016-04-09:
Please check examples bundled with the ESP8266WiFi library and ESP8266WebServer library.
WiFiServer is a TCP server, and ESP8266WebServer is an HTTP server built on top of WiFiServer.
If you use ESP8266WebServer library you still need to include `ESP8266WiFi.h` because you have to set up WiFi connection before setting up HTTP server anyway.

closed 2016-04-09 by igrr

odilonafonso, 2016-05-31:
Hi @igrr , please, what is the major differences between TCP Server (WiFiServer) and HTTP Server (ESP8266WebServer) ?
Can I get a Webserver working with a TCP Server and not a HTTP Server ?

dogmaphobic, 2016-05-31:
That's a bit like asking what is the difference between a car and a road. TCP is the road and HTTP is the car. Or the car is TCP and the people inside are HTTP (where the guy wrapped in duct tape so you can see him is HTTPS). Analogies just flying out everywhere:

https://en.wikipedia.org/wiki/Transmission_Control_Protocol

https://en.wikipedia.org/wiki/Hypertext_Transfer_Protocol

odilonafonso, 2016-06-01:
OK @dogmaphobic - you right ! Thanks :)

Actually my question is the practical aspect - how and when to use one or the other.

I like to have a Webserver and that it is the most efficient and easy to use.

I did some testing with an HTTP Webserver that worked to the satisfaction of clients connected by Firefox; however I had some problems with clients accessing the Chrome but especially when access is made for Android app.

It seems to me that Android is much more strict as the requirements of the HTTP protocol.

So I would like to understand better, asking experts, whether it is worth using a  TCP Webserver and the advantages and disadvantages of either.

odilonafonso, 2016-06-01:
Examples sometimes confuse beginners; look at this sample:

https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266mDNS/examples/mDNS_Web_Server/mDNS_Web_Server.ino

The comments at the beginning say:

```
/*
  ESP8266 mDNS responder sample
  This is an example of an **HTTP server** that is accessible
  via http://esp8266.local URL thanks to mDNS responder.

```
and soon after, in use:

```
// **TCP server** at port 80 will respond to HTTP requests
WiFiServer server(80);

```
I will continue testing to try to understand what the differences are **IN USE**. Not in theory.
Any help is welcome.

charles-haynes, 2017-09-03:
ESP8266WebServer implements a web (HTTP) server. It is built using WiFiServer. If you find ESP8266WebServer meets your needs, just use it. If it doesn't, you may implement your own HTTP server on top of WiFiServer. You can start by copying the ESP8266WebServer.cpp implementation if that's helpful.

plugboy, 2018-11-15:
As an example I just switched from ESP8266WebServer to WiFiserver for a project that I need to parse a huge json file posted from a client. 
If I used Esp8266WebServer I can only work with the string parsed by the library in server.arg("plain") but with WiFiserver I can define how I want to parse the data. 

The trade off is i need to write several things extra like how to do routing. Because in WiFiserver there is nothing like server.on() to set up the routes and handler

crypsky, 2019-01-13:
Yes, I am finding i cannot use ESP8266WebServer because it does not support server.available().  
WiFiserver does not support server.arg(), so I am dumpster diving the internet looking for the simplest way to emulate server.arg().

wrinkledmind, 2022-04-17:
> Yes, I am finding i cannot use ESP8266WebServer because it does not support server.available(). WiFiserver does not support server.arg(), so I am dumpster diving the internet looking for the simplest way to emulate server.arg().

in ESP8266WebServer server.handleclient() do the same thing as server.available();