esp8266/Arduino #1870: Does ESP8266 with Arduino IDE support multi-threading?

geoffreyhhh, 2016-04-06:
Dears,
was wondering if ESP8266 with Arduino IDE supports multi-threading to run multiple tasks/functions simultaneously? If it does, please help to provide the working example code.

any advise is highly appreciated.

Geoffrey

WereCatf, 2016-04-06:
ESP8266 is single-core, so you can't run stuff simultaneously anyways.

geoffreyhhh, 2016-04-07:
Thanks for the advice.

wondering if ESP8266 with Arduino IDE supports allocating processing resources for a function for specific time before it runs next function. 

// During the 5000ms allocation, processing resource should be available
void loop(){
  loop1(); // e.g. allocate 5000ms before it runs loop2
  loop2();
  loop3();
}

any advise is highly appreciated.
Geoffrey

WereCatf, 2016-04-07:
I'm not entirely sure what you're asking. You could do timestamps by tracking time spent with millis() and then running one or another function when a certain amount of time has passed, or you could use timers to set a flag when a certain amount of time has passed and just check whether the flag is true or flag in loop and run your functions accordingly -- the latter solution makes for cleaner code, even if there is no practical difference.

I might be able to offer more if you explained what you're trying to achieve.

krzychb, 2016-04-07:
@geoffreyhhh,

I guess you may be looking for a [cooperative scheduler](http://www.embedded.com/electronics-news/4434501/Writing-a-simple-cooperative-scheduler) like [this one](http://www.arduino.cc/en/Reference/Scheduler) prepared for Arduino sam and samd architectures.

Out of curiosity I made a quick check and sure enough [MultipleBlinks](http://www.arduino.cc/en/Tutorial/MultipleBlinks) sketch does not compile for ESP. It is  complaining a lot  about incompatibility of boards and by providing several other reasons it will not make it :smile: I like s/w that is saying in plan text what the problem is :+1:

Krzysztof

wkraus, 2016-04-07:
@geoffreyhhh Hi, I recently found this article: https://www.hackster.io/feilipu/using-freertos-multi-tasking-in-arduino-ebc3cc?utm_source=Hackster.io+newsletter&utm_campaign=9665148c8a-2015_4_17_Top_projects4_16_2015&utm_medium=email&utm_term=0_6ff81e3e5b-9665148c8a-140388981 - but have not tried it with ESP yet.

geoffreyhhh, 2016-04-08:
Thanks for the inputs and sorry for the late reply.

Below is the portion of the code relating to the question. it's a code for demo purpose, loop1() is to demo turning on/off esp8266 onboard led through web browser.

ESP8266WebServer server(80);

void loop(){
  loop1(); // e.g. allocate 5000ms before it runs loop2
  loop2();
  loop3();
}

void loop1(void){
  server.handleClient();
}

when there's no loop2() loop3() in loop(), and loop1() code is in loop(), I used millis() to measure the time required to complete the code, it gave between 3second ~ 5second, each time is slightly different.

when loop1() loop2() loop3() are in loop(), once web browser sends commands to esp8266, it's very hard to get complete response from esp8266 to display in web browser the response message from esp8266. t

igrr, 2016-04-09:
> when loop1() loop2() loop3() are in loop(), once web browser sends commands to esp8266, it's very hard to get complete response message from esp8266 to display in web browser the response message from esp8266.

This looks like the main issue here. Could you please post the minimal code sample which demonstrates the issue? Thanks.

geoffreyhhh, 2016-04-10:
Hello~
Thanks for asking and the support.

The minimal code relating to the question is as follows.

ESP8266WebServer server(80);

void handleRoot() {
  server.send(200, "text/plain", 
    "\n/on: to turn onboard LED ON \n/off: to turn onboard LED OFF \n");

}

void handleNotFound(){
  digitalWrite(led, 1);
  digitalWrite(ext_led, 1);
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET)?"GET":"POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i=0; i<server.args(); i++){
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  digitalWrite(led, LOW);
  digitalWrite(ext_led, LOW);
}

void setup() {
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  setup1();
}

void setup1(void){
  pinMode(led, OUTPUT);
  pinMode(ext_led, OUTPUT);
  pinMode(buttonPin, INPUT); //connect to ESP8266 GPIO12
  digitalWrite(led, 0);
  digitalWrite(ext_led, 0);
  
  server.on("/", handleRoot);

  server.on("/on", [](){
    digitalWrite(led, LOW); //BUILTIN_LED
    digitalWrite(ext_led, HIGH); //external GPIO5 LED
    msg1 += "LED ON; button status: ";
    msg1 += buttonState;
    msg1 += " sensorValue: ";
    msg1 += sensorValue;
    msg1 += "\n";
    server.send(200, "text/plain", msg1);
  });
  
  server.on("/off", [](){
    digitalWrite(led, HIGH); //BUILTIN_LED
    digitalWrite(ext_led, LOW); //external GPIO5 LED
    msg2 += "LED OFF; button status:  ";
    msg2 += buttonState;
    msg2 += " sensorValue: ";
    msg2 += sensorValue;    
    msg2 += "\n";
    server.send(200, "text/plain", msg2);
  });

  server.onNotFound(handleNotFound);
  
  server.begin();
  Serial.println("HTTP server started");

  //delay a moment, 
  //for terminal to receive inf, such as IP address
  delay(1000);
//  Serial.end();
  pinMode(led, OUTPUT); //BUILTIN_LED
  pinMode(ext_led, OUTPUT);
   digitalWrite(led, 0); 
}

void loop(){
  loop1(); // e.g. allocate 5000ms before it runs loop2
  loop2();
  loop3();
}

void loop1(void){
  server.handleClient();
}

WereCatf, 2016-04-10:
I have to say, I still don't understand your problem. Are your loop2() and loop3() terribly slow, then?

baruch, 2016-04-19:
As for cooperative scheduling, the ESP8266 Arduino code already has the coroutine functions, I wonder if they can be exposed somehow to the users. I would love to make use of them for some of my code instead of the state machines I need to code now.

Makuna, 2016-04-23:
FYI:  Github/Makuna/Task is a non-preemptive multitasking library support that cross Arduino compatible.

baruch, 2016-04-23:
While makuna/task does help a bit it isn't much different then having a set of loop functions and checking for triggers. 

A coroutine approach is much more convenient and while it is not portable I don't really care that much about portability anyway in my project.

malachib, 2016-05-11:
+1 on this topic; I have a lot of use cases for cooperative multitasking on the ESP

mkeyno, 2016-05-11:
its already done, unlike the AVR core, every modern core like ARM and extensa can prioritize its interrupts which is resulted  to multi-threading  task, but in old Arduino board which is base on ATMEGA core , many developer have to calculate precise timing sequence for multi tasking and in fact impossible because each event (interrupt)  have nor priority over another event (interrupt)

anmaped, 2016-11-02:
Take a look on https://github.com/anmaped/esp8266-scheduler it is a preemptive scheduler for esp8266 Arduino.

I have to submit the changes to Arduino esp8266 to weak the definition of some functions. It is not stable, but for now it is working very well.

@igrr Could I submit this pull resquest to weak the function 'esp_schedule', 'optimistic_yield', 'loop_task', and 'delay' ?

robSoofa, 2017-01-26:
I added Arduino-Scheduler, but it seems like there are some errors compiling. Do I need to reconfigure anything to use this library on ESP?

```
C:\Users\rober\AppData\Local\Temp\build5236869049569846302.tmp/arduino.ar(core_esp8266_wiring.c.o): In function `delay':
C:\Users\rober\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring.c:42: multiple definition of `delay'
Simple Scheduler\Scheduler.cpp.o:C:\Users\rober\Documents\Arduino\libraries\Arduino-Scheduler/Scheduler.cpp:96: first defined here
C:\Users\rober\AppData\Local\Temp\build5236869049569846302.tmp/arduino.ar(core_esp8266_main.cpp.o): In function `esp_yield':
C:\Users\rober\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp:82: multiple definition of `esp_yield'
Simple Scheduler\Scheduler.cpp.o:C:\Users\rober\Documents\Arduino\libraries\Arduino-Scheduler/Scheduler.cpp:89: first defined here
C:\Users\rober\AppData\Local\Temp\build5236869049569846302.tmp/arduino.ar(core_esp8266_main.cpp.o): In function `optimistic_yield':
C:\Users\rober\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp:104: multiple definition of `optimistic_yield'
Simple Scheduler\Scheduler.cpp.o:C:\Users\rober\Documents\Arduino\libraries\Arduino-Scheduler/Scheduler.cpp:93: first defined here
Simple Scheduler\Scheduler.cpp.o:(.text._ZN14SchedulerClass5yieldEv+0x4): undefined reference to `setjmp'
Simple Scheduler\Scheduler.cpp.o:(.text._ZN14SchedulerClass5yieldEv+0x8): undefined reference to `longjmp'
Simple Scheduler\Scheduler.cpp.o: In function `SchedulerClass::yield()':
Scheduler.cpp:(.text._ZN14SchedulerClass5yieldEv+0x18): undefined reference to `setjmp'
Scheduler.cpp:(.text._ZN14SchedulerClass5yieldEv+0x2e): undefined reference to `longjmp'
collect2.exe: error: ld returned 1 exit status
```

Didix57, 2017-02-12:
Let me get back to this thread.
The typical Webserver example has two handles: one to put an LED on and a second to put it off.

Example:
```
void handle_ledOff() {
  digitalWrite(ledPin, LOW);
  doHtmlPage();
}
```
Instead of just putting the LED on, I would like to do something that lasts forever - unless I stop it intentionally.
Example:
```
void handle_ledOn() {
  doHtmlPage();
  while (not interrupted) {
    digitalWrite(ledPin, HIGH);
    delay(100);
    digitalWrite(ledPin, LOW);
    delay(100);
  }
}
```
Well, do no just focus on blinking, it might be something more complex.

However, the second routine blocks the webserver and clicking on the button at the html-page to trigger ledOff therefore does not show any result than waiting.

How could I overcome that?
How can I get 'interrupted' become TRUE?

Would Makuna/Task solve that problem?

vicnevicne, 2017-02-12:
In such a case, I would just do something like:
```
void handle_ledOff() {
  digitalWrite(ledPin, LOW);
  doHtmlPage();
  flashLed = 0;
}

void handle_ledOn() {
  doHtmlPage();
  flashLed = 1;
}

void loop() {
  if (flashLed) {
    digitalWrite(ledPin, HIGH);
    delay(100);
    digitalWrite(ledPin, LOW);
    delay(100);
  }
}
```
The idea is that the loop lets the code "breathe", and the handlers just set indicator variables that can be tested in the main loop.

Vicne

Makuna, 2017-02-13:
And yes the makuna/task would handle this problem.  It will take the example Vicne above to the next level and allow you to get rid of delay completely.  For most Arduinos delay is blocking, so nothing else happens while you wait.  If you are using the Async Html Server object for Esp8266, then this isn't a problem though.

Some reference reading [Adafruit Multitasking](https://learn.adafruit.com/multi-tasking-the-arduino-part-1?view=all) that will help you understand the concept.

Didix57, 2017-02-13:
@ vicnevicne:
Where would you place 'server.handleClient()'?

@ Makuna:
Thanx, epecially for the link, too. This takes me further!

@ all:
I also stumbled over ESPAsyncWebServer
https://github.com/me-no-dev/ESPAsyncWebServer
Anybody experience with that approach?

Could anybody recommend me which path to follow?

vicnevicne, 2017-02-13:
Yeah, I omitted many things related to the server to make the case simpler. 
server.handleClient() should be in the loop of course. Something along the lines:

```
void loop() {
  server.handleClient();
  if (flashLed) {
    digitalWrite(ledPin, HIGH);
    delay(100);
    digitalWrite(ledPin, LOW);
    delay(100);
  }
}
```

Naguissa, 2017-02-25:
Is this supported on esp?

https://www.arduino.cc/en/Reference/Scheduler

HugoSH, 2017-10-05:
Any more discussions? 
I am having the same issue ....
loop(){
   server.handleclient();
   function-do-sth-else();
}
The handleClient() is so ... slow ... to send html to client. :(

devyte, 2017-10-05:
@HugoSH try #3362 , it updates lwip to v2, while we're currently on 1.4. There have been several reports about improving the handleClient time.

devyte, 2017-10-05:
@geoffreyhhh et al
this firmware doesn't support concurrent multithreading, and afaik, the Arduino standard Scheduler also hasn't been ported. Doing that requires a rather huge effort in an old piece of code that has very high impact, because it affects yield and delay functions, and it's therefore very risky.
There are two esp-specific scheduler-like libs out there in gits, and there is also the FreeRTOS firmware, but they all have certain drawbacks.
You shouldn't need to do multithreading, there are round-robin methods with timing checks that can be just as good, and use less resources.
But this is not the right place to discuss these thing. Closing as off-topic, see #3655 .

closed 2017-10-05 by devyte

HugoSH, 2017-10-05:
Thanks! Let me go there to take a look.

ozanhan, 2017-10-23:
@anmaped i got compile errors for nodemcu 1.0

MacLeod-D, 2019-12-04:
Here you can find a full blown up tutorial for multitasking with ESP8266:

Without WiFi:
https://github.com/MacLeod-D/CoopOS-Lessons-English

And a Demo with Wifi
https://github.com/MacLeod-D/ESP8266-Multitasking-CoopOS

Have fun ;)

Sagar73594, 2021-05-22:
> Here you can find a full blown up tutorial for multitasking with ESP8266:
> 
> Without WiFi:
> https://github.com/MacLeod-D/CoopOS-Lessons-English
> 
> And a Demo with Wifi
> https://github.com/MacLeod-D/ESP8266-Multitasking-CoopOS
> 
> Have fun ;)
> 

Great Scott.