esp8266/Arduino #1871: ESPhttpUpdate.update() uses huge amount of memory

PhilColbert, 2016-04-06:
Adding the line 

ESPhttpUpdate.update() uses a ridiculous amount of memory.

system_get_free_heap_size = 33224 without

system_get_free_heap_size = 25440 with that line added

Surely theres something not quite right there ?

Thanks

Version 2.0.0

sticilface, 2016-04-06:
Had a very similar issue with the http client. After some tracking I found that it was due to simultaneous use of http and https. 

Wonder if it is  something similar... Can check now though.  Igrr fixed it with some compiler, linker, virtual function kungfoo!

PhilColbert, 2016-04-06:
Thanks for the reply, this is just a simple sketch

extern "C" {
#include <user_interface.h>
}
#include "ESP8266httpUpdate.h"
#include "ESP8266HTTPClient.h"

void setup()
{
  Serial.begin(115200);
  delay(100);
}

void loop()
{
  Serial.println("GO!");
  uint32_t freeM = system_get_free_heap_size();
  Serial.println(freeM);
  ESPhttpUpdate.update("37.128.187.9", 80 , "/downloads/AirWare.ino.bin" ); // this line !! :(
  delay(1000);
}

sticilface, 2016-04-06:
So it uses 

#include <ESP8266HTTPClient.h>

I suspect it is the same issue. 
If u use GitHub try this pull it should go away. 
https://github.com/esp8266/Arduino/pull/1857

PhilColbert, 2016-04-06:
Thanks, will look into it when I get a free minute , 5kb DRAM is a lot to loose ! :)

igrr, 2016-04-06:
This is not a fair comparison, because without this line you aren't using ESP8266WiFi library, which is pretty big. Adding this line pulls in ESP8266WiFi, ESP8266HTTPClient, and ESP8266httpUpdate libraries.

PhilColbert, 2016-04-06:
yes thats true, sorry I am still learning here ! :)

igrr, 2016-04-09:
Closing as duplicate of #1855.

closed 2016-04-09 by igrr