esp8266/Arduino #1872: http.POST() limit of payload size?

TheAustrian, 2016-04-06:
Hi! 

I'm trying to send data as an HTTPClient to a server with http.POST() and it works great but the payload gets cut off around the 3000 character mark. Does this have something to do with "application/x-www-form-urlencoded"? It shouldn't, iirc, cause POST shouldn't have a size limit?

> HTTPClient http;
> http.begin("http://example.com/weather/espdata.php");
> http.addHeader("Content-Type", "application/x-www-form-urlencoded");
> http.POST(textToSend);
> http.end();

I just wanted to know if there is a limit on payload size so I can rule out some other issue with my server. The String I'm sending gets put out normally in the serial monitor, so that shouldn't be the problem.

Thanks!

Humancell, 2016-05-05:
Bump!

I'm seeing this same issue ... when my payload crossed the 3k mark I'm now getting a http.POST() return of -3 ... HTTPC_ERROR_SEND_PAYLOAD_FAILED ... but can't find any info on why this is, and what exactly failed?

Is there a buffer that needs to be adjusted, or?

chaeplin, 2016-05-05:
I think payload size > one packet size --> err.
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp#L376-L380

Humancell, 2016-05-05:
Ok ... this is what I wondered.  I actually created a workaround that I'm using successfully, just in case anyone else requires a similar fix.  To do it I had to use WifiClient:

    #include <ESP8266HTTPClient.h>

    String jsonPayload = "{ my big giant 3.5k byte json payload ....}";
    int contentLength = jsonPayload.length();

      // create the request and headers
      String request = "POST " + String(endpointURL) + " HTTP/1.1\r\n" +
        "Host: " + String(endpointHost) + "\r\n" + 
        "Accept: application/json" + "\r\n" + 
        "Content-Type: application/json\r\n" +
        "Content-Length: " + String(contentLength) + "\r\n\r\n";

      // Use WiFiClient class to create TCP connections
      WiFiClient client;

      connectResult = client.connect(endpointHost, atoi(endpointPort));

      // This will send the request and headers to the server
      client.print(request);

      // now we need to chunk the payload into 1000 byte chunks
      int cIndex;
      for (cIndex = 0; cIndex < contentLength; cIndex = cIndex + 1000) {
        client.print(jsonPayload.substring(cIndex, cIndex+1000));
        //Serial.print(jsonPayload.substring(cIndex, cIndex+1000));
      }
      client.print(jsonPayload.substring(cIndex));
      //Serial.println(jsonPayload.substring(cIndex));

I have a little more response checking and other stuff ... but you get the idea.  Using this I am now sending pretty big payloads, and it seems to work fine.  The 1000 bytes was just an arbitrary guess ... if I have time I might try to adjust this to fine tune the operation.

Obviously there are limitation sin this approach as I can't do SSL easily (or can I?) ... but maybe something like this ought to be rolled into the handling of the payload on the call to http.POST in the HTTPClient?

chaeplin, 2016-05-05:
@Humancell use WiFiClientSecure.h for ssl.
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/HTTPSRequest/HTTPSRequest.ino

https://github.com/chaeplin/esp8266_and_arduino/blob/master/_56-gopro-control/a-91-config-on-rtc/a-91-config-on-rtc.ino#L1331-L1372

Chunk can be large to 1460.

Humancell, 2016-05-05:
Thanks!  Nice bonus!  I'll make my changes.  :-)

espmut, 2017-05-15:
I just ran into this issue as well.   I believe the limit is just under 2000 Bytes.  I hope there is a work around or solution for this.  I'd rather use HTTPClient.

melvinpmathew, 2017-05-31:
Is there a work around for this issue? I need to send 8000bytes on POST>

devyte, 2017-10-17:
@TheAustrian @Humancell is this issue still valid with latest git?

mcanyucel, 2018-01-29:
I am trying to upload a csv file with size of around 200 kB and using the method `int HTTPClient::sendRequest(const char * type, Stream * stream, size_t size)` and get the error `HTTPC_ERROR_SEND_PAYLOAD_FAILED`. Smaller size files work perfectly so I am assuming there is an unwritten file size limit or kind of a bug. This is mentioned in few stackoverflow questions as well but they have no answer ([here](https://stackoverflow.com/questions/41388417/send-payload-error-in-http-post-arduino-esp8266). 
Any new information, suggestions, progress, workaround?

espmut, 2018-01-29:
For me, I broke up the post into multiple posts.

Silberlachs, 2018-01-30:
I see an issue when receiving rss feeds from a news portal over http GET (get string to be clear)

One RSS page with kind of high priority news gets transmitted normally and the esp can send it to a lcd display.

But whenever I try to get the "normal" news feeds, the esp just doesn't do anything at all.
Maybe it restarted? 

The normal rss feed has 14527 Characters, the high priority one has 2676 Characters...

I've gone with the high priority one for now and I can see why nearly 15k characters is a bit overkill for this little thing, but still it's an issue....

mcanyucel, 2018-01-30:
@espmut How did you manage to divide the file? I used another approach that reads the file into a WiFiClient using MTU_Size chunks but it floods everywhere with stack errors :(

d-a-v, 2018-01-30:
I could reproduce the error.
Can you retry sending big data without chunking into pieces, with DebugLevel HTTP_CLIENT (in Tools menu, with Serial for DebugPort) ?
I get:
```
[HTTP-Client][sendRequest] short write, asked for xxxx but got yyyy retry...
```

2MB data are working flawlessly with all debug options activated (they slow down processing a lot).
Bug is located, fix will come hopefully.

Full sketch:
```
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

#define ssid "open"
#define password ""

void setup()
{
  Serial.begin(115200);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");

  // Print the IP address
  Serial.println(WiFi.localIP());

}

class FakeStream: public Stream
{
  protected:
    size_t fakesize;

  public:
    FakeStream (size_t size): fakesize(size) { }

    virtual int available() {
      return fakesize;
    }
    virtual int read() {
      if (fakesize) {
        --fakesize;
        return 0x5a;
      }
      return -1;
    }
    virtual int peek() {
      return 0x5a;
    }
    virtual void flush() {
      fakesize = 0;
    };
    virtual size_t write (uint8_t data) {
      (void)data;
      return 1;
    }
    size_t size () {
      return fakesize;
    }
    const char* name () {
      return "5a";
    }
};

void loop()
{
  // put your main code here, to run repeatedly:
  HTTPClient http;
  http.begin("http://10.43.1.254:80/test/upload.php");

  int contentLength = 2000000;

  http.addHeader("Content-Type", "application/octet-stream");
  http.addHeader("Content-Length", String(contentLength));

  Serial.println("a----------------------------------------------");
  FakeStream s(contentLength);
  int httpCode = http.sendRequest("POST", &s, contentLength);
  Serial.println(httpCode);
  Serial.println("b----------------------------------------------");

  while (true)delay(1000);
}
```

The ```upload.php``` (got in some other issue):
```
<?php
file_put_contents('/var/www/html/test/test.txt', file_get_contents('php://input'));
?>
```

d-a-v, 2018-01-31:
@TheAustrian @mcanyucel @espmut can you unbreak-to-one-piece your POST and try this pull-request #4265 ?

@Silberlachs the fix is about outgoing data (not incoming data). If it still does not work for you, then please open a new issue with a minimal sketch to highlight the bug. It will be a lot easier for us to reproduce then fix.

martinius96, 2018-02-01:
Why people use HTTPclient, when they can use normal HTTP request under ESP8266Wifi.h library?

mcanyucel, 2018-02-01:
@martinius96 Are you referring the `WiFiClient`? I could not see any http request under the [ESP8266Wifi.h](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFi.h) library.

martinius96, 2018-02-01:
@mcanyucel Yes WIficlient. What is difference between wificlient and [HTTPclient (BasicHttpClient.ino file)](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/examples/BasicHttpClient/BasicHttpClient.ino)? It do the same thing..

mcanyucel, 2018-02-01:
In a manner of speaking. As far as I have seen `WiFiClient` has a very broad `virtual size_t write(const uint8_t *buf, size_t size);` method, therefore you have to generate all the `POST` headers manually. If you are making a post request of form-url-encoded with a short string, it may not be very hard to do that. But if you are attaching a binary file, the post request file identifiers are not very user friendly. moreover you have to read the file in buffer sized chunks and I am not very proficient with cpp so I keep getting stack errors all over the place. I think I could not find a single example that works on the internet that uploads a file on the SD card to a server using `WiFiClient`,
`HttpClient` on the other hand exposes more streamlined methods that accept the post parameters and generates the necessary strings automatically. It also handles buffer operations better than me; it does not flood everywhere with stack overflows. You can use a single line `int sendRequest(const char * type, Stream * stream, size_t size = 0);` which in my opinion better and more appealing than to have [this](https://github.com/esp8266/Arduino/issues/1853#issuecomment-249217498).
In the end it is "not reinventing the wheel"; the SD card operations or OTA updates can be done manually using basic core libraries, but it is easier to use the already developed "advanced" libraries.

d-a-v, 2018-02-01:
@martinius96 ```HTTPClient``` uses ```WiFiClient``` to make it easier to deal especially with http requests and answers.

@TheAustrian @mcanyucel @espmut the fix is now in master. Can you test it ?

mcanyucel, 2018-02-01:
@d-a-v Thank you very much for your extremely quick response. I am a little bit overwhelmed with some other stuff nowadays but hopefully I will be able to test it within a few days or maybe early next week.

devyte, 2018-02-20:
@mcanyucel @TheAustrian @Humancell could you please confirm the fix? I think this is the only pending for release 2.4.1.

closed 2018-03-08 by igrr

Humancell, 2018-09-15:
I'm sorry that I somehow missed your request, and the Github notice back in February!

It was funny, but I ran into this issue again last night, with the payload being too large, and found this thread to check back on the status.  I *did* update to v2.4.2 ... and this *does* fix the issue that I was seeing!  I'm now POSTing bigger payloads just fine!

One other issue that I did see, is that I has to add a line to set:

> httpClient->setTimeout(250);

Before reading the response stream.  Without this, my code that was running just fine before, (v2.3.0) now seemed to have a default timeout of 4000ms!  It would just hang.  I added the line above, and now it's fine.

ligantx, 2020-03-15:
> Ok ... this is what I wondered. I actually created a workaround that I'm using successfully, just in case anyone else requires a similar fix. To do it I had to use WifiClient:
>
> ```
> #include [...]

After spending MANY hours trying to upload ESP32-CAM photos, i found out the problem was only with big photos (e.g. bigger than VGA) and big files (txt) in general, so i finally found the problem was with the POST body in general..
The quote solution works fine with Strings but esp32-cam produce a uint8_t * buf that has to be used differently..
So, the code below that i wrote worked fine with every FRAMESIZE (even UXGA - 1600 x 1200):
`
    camera_fb_t * fb = esp_camera_fb_get();
    int imgLenInt = fb->len;
    String imgLenStr = (String)imgLenInt;
    //headers already POSTed
    int howMuch = 0;
    int sumHowMuch = 0;
    int i = 0;
    int myChunkSize = 1300;  //try 1000-2000 values if you want
    for (i = 0; i < imgLenInt - myChunkSize; i = i + myChunkSize) {
      howMuch = client.write(fb->buf + i, myChunkSize);
      sumHowMuch += howMuch;
    }
    int left = imgLenInt % myChunkSize;
    howMuch = client.write(fb->buf + (i - myChunkSize), left);
    sumHowMuch += howMuch;
    Serial.print("you had to send:");
    Serial.print(imgLenStr);
    Serial.print(", and you send:");
    Serial.println(sumHowMuch);
`

AndrewIvashkin, 2021-05-28:
@ligantx Hey i am trying to send base64 image in POST request 2
Right now i has the correct base64 string but when i am trying to send it i end up with error code -1 

I tryed simmilar approach like you:  I divide my string by 10 parts with code like this 

`void SendChunks()
{ 
  HTTPClient http;
  http.setTimeout(6000);
  http.begin(endpoint,fingerprint);
  http.addHeader("Content-Type", "application/json"); 
  
  int ChunkCount = 10;
  int PicLenght = B64Pic.length();
  Serial.println(PicLenght);
  int ChunkLenght =  PicLenght / ChunkCount;
  Serial.println(ChunkLenght);
  
  for(int i = 0; i < ChunkCount; i++)
  {
    //Create json
    DynamicJsonDocument doc(40000); //size of the doc can 
    doc["index"] = i;
    
    doc["Base64Chunk"] = B64Pic.substring(i*ChunkLenght, (i+1)*ChunkLenght);
    //Serialize json
    String requestBody;
    serializeJson(doc, requestBody);
    Serial.println(requestBody);
    delay(500);

    //Do POST request
    int httpResponseCode = http.POST(requestBody);
    if(httpResponseCode>0)
    {
      Serial.println(httpResponseCode);   //Print return code
    }
    else
    { 
      Serial.print("Error on sending request: ");
      Serial.println(httpResponseCode);
    }
    
  }

  //send tail
  if(PicLenght -(ChunkLenght*ChunkCount)> 0)
  {
    DynamicJsonDocument doc(40000); //size of the doc can 
    doc["index"] = ChunkCount + 1;
    doc["Base64Chunk"] = B64Pic.substring(ChunkCount * ChunkLenght);
    //Serialize json
    String requestBody;
    serializeJson(doc, requestBody);
    Serial.println(requestBody);
    delay(500);
    //Do POST request
    int httpResponseCode = http.POST(requestBody);
    if(httpResponseCode>0)
    {
      Serial.println(httpResponseCode);   //Print return code
    }
    else
    { 
      Serial.print("Error on sending request: ");
      Serial.println(httpResponseCode);
    }
  }
  
  http.end();
}`

And finished with output like this(i made base64 chunks short on purpose) :
`{"index":0,"Base64Chunk":"/9j/U"}
400
{"index":1,"Base64Chunk":"UCF"}
Error on sending request: -11
{"index":2,"Base64Chunk":"ASi"}
Error on sending request: -1
`
So only first request end up with 400 code, second with -11 all the other return -1.

PR #4265 (2018-02-01): ClientContext:_write_some: release buffer once data really accepted for sending
may fix #1872
commit 199fe0f1 (2018-02-01): ClientContext:_write_some: release buffer once data really accepted for sending (#4265)

may fix #1872