esp8266/Arduino #1873: int ESP8266WebServer::hasArg(String  name) {

odilonafonso, 2016-04-07:
### Basic Infos

#### Hardware
Hardware:		ESP8266 General
Core Version:      	2.0.0+

### Description

Another (faster) way to get a ESP8266WebServer arg by name

### ESP8266WebServer.cpp

```cpp

int ESP8266WebServer::hasArg(String  name) {
  for (int i = 0; i < _currentArgCount; ++i) {
    if (_currentArgs[i].key == name)
      return i;
  }
  return -1;
}
```
### ESP8266WebServer.h

```h
....
  bool hasArg(const char* name);  // check if argument exists
  int hasArg(const char* name);   // check if argument exists and return index or -1 if arg not exists
  void collectHeaders(const char* headerKeys[], const size_t headerKeysCount); // set the request headers to collect
....
```

### sketch (sample)
```
.....
ESP8266WebServer esp8266WebServer(port);
....

igrr, 2016-04-09:
C++ doesn't allow you to have two function overloads which only differ by their return type:
```c++
  bool hasArg(const char* name);  // check if argument exists
  int hasArg(const char* name);   // check if argument exists and return index or -1 if arg not exists
```
I think better naming for the function which you are proposing would be `findArg`.

odilonafonso, 2016-04-10:
Thanks Ivan ! I will test this new method now.

closed 2016-05-31 by odilonafonso