esp8266/Arduino #1876: Using VS2015 as ID, Makefile

luni64, 2016-04-07:
I need / want to compile my ESP Arduino sketches using the VS2015 IDE. I did this for other arduino platforms (UNO, Teensy) without any major problems. But this time I'm stuck. 

As usual I try to generate the makefile by analyzing the debug output of the Arduino IDE. In principle my makefile works. The core files and blink.ino compile, the *.bin uploads but: _it doesn't run_. I try to find the difference in both compilations for  a couple of days now but without success. I checked all the flags and options to the compiler, everything looks OK. This drives me crazy. 
Maybe somebody can help me with that. 

Here is what I found out so far: 
If I compare the *.elf file which was generated by Arduino with the one my makefile generates I see that my *.elf file includes an additional memory section .iram.text. I have no idea where this comes from. But I belive that this is a significant difference. 

[image]

To nail the error down, I exchanged the arduino.ar file generated by my buildsystem with that generated by the arduino build system. If I rebuild with my makefile and the other arduino.ar everything works. This is good, because linking and esptool obviously work at least. 
To find out whats wrong with my arduino.ar archive I compared some of the *.o files with the ones arduino generates using objectdump.  They show no difference except the filename which is coded in the .rodata section (BTW: why does GCC code the filename in the object file????) 

[image]

Differences marked, the rest of both files is absolutely identical. Do you have any idea what I can check further? Does the Arduino Build system play some tricks with Environment Variables which can't be seen in the debug output? 

Thanks for any help
Lutz

luni64, 2016-04-09:
Ta ta... I finally got it going. The problem was the object file naming scheme. I usually just name them *.o. The xtensa linker seems to only work correctly if the object files are named *.c.o *.cpp.o and so on. Very strange behavior, maybe a bug. Anyway, I'm happy that it works now and I finally can use Visual Studio 2015 for my project.

closed 2016-04-09 by luni64

Terstegge, 2017-02-15:
I just ran into the very same problem when setting up a makefile-based environment for arduino.
Actually the xtensa linker is working fine with .o _and_ .c.o files, but the linker script relies on this strange object file naming:

`  .text : ALIGN(4)
  {
    _stext = .;
    _text_start = ABSOLUTE(.);
     *(.UserEnter.text)
    . = ALIGN(16);
    *(.DebugExceptionVector.text)
    . = ALIGN(16);
    *(.NMIExceptionVector.text)
    . = ALIGN(16);
    *(.KernelExceptionVector.text)
    LONG(0)
    LONG(0)
    LONG(0)
    LONG(0)
    . = ALIGN(16);
    *(.UserExceptionVector.text)
    LONG(0)
    LONG(0)
    LONG(0)
    LONG(0)
    . = ALIGN(16);
    *(.DoubleExceptionVector.text)
    LONG(0)
    LONG(0)
    LONG(0)
    LONG(0)
    . = ALIGN (16);
    *(.entry.text)
    *(.init.literal)
    *(.init)
    *(.literal .text .literal.* .text.* .stub .gnu.warning .gnu.linkonce.literal.* .gnu.linkonce.t.*.literal .gnu.linkonce.t.*)
    *.cpp.o(.iram.text)
    *.c.o(.iram.text)
    *(.fini.literal)
    *(.fini)
    *(.gnu.version)
    _text_end = ABSOLUTE(.);
    _etext = .;
  } >iram1_0_seg :iram1_0_phdr`

So when the linker script is changed accordingly (so that *.o files go into the correct section), then things are also working with *.o files!