esp8266/Arduino #1882: Long filename support for SD

aalku, 2016-04-08:
### Basic Infos

#### Hardware
Hardware:			ESP-12 or 12E, I don't know
Core Version:      	2.1.0

### Description

I would like to get long filename support on SD.

This SdFat fork got it.
https://github.com/tannewt/SdFat/

There are small use differences with SD. Most serious is you have to pass a buffer to get a file name.

I'm using this around it:

    String getFilename(File f) {
      char buff[256] = "";
      f.getName(buff, 256);
      return String(buff) + "";
    }

`+""` is because I don't know if String wraps the passed buffer or copies it.

I had to remove before write to overwrite too:

    if (SD.exists(filename.c_str())) {
      SD.remove(filename.c_str());
    }
    fsUploadFile = SD.open(filename.c_str(), FILE_WRITE);

WereCatf, 2016-04-08:
I would like this too.

aalku, 2016-04-08:
Related https://github.com/greiman/SdFat/pull/31#event-620342440

aalku, 2016-04-08:
Confirmed. This beta version works on ESP8266 without third-party patches. https://github.com/greiman/SdFat-beta

WereCatf, 2016-04-08:
Enabling SPI-transactions support breaks it.

WereCatf, 2016-04-08:
Changing both instances of `SPISettings()` to `SPISettings(ESP8266_CLOCK, MSBFIRST, SPI_MODE0)` in SdSpi.h fixes SPI-transactions and allows for the library to work. I think I'll submit a pull-request.

igrr, 2016-04-09:
Could you please clarify, is this issue about adding ESP8266 support to SdFat library, or about something else?

aalku, 2016-04-09:
It was about "get(tting) long filename support on SD" (sd card) on ESP8266. No matter how. :) I don't know the criteria to include third party library on this repository either. SdFat-beta looks very good.

igrr, 2016-04-09:
Ok, clear enough.
Personally i don't think we should be including third-party libraries into the repository, because upgrading bundled libraries is not very straightforward. Library authors may release updates more often than we can update the core. Arduino IDE offers a library manager which can automatically install libraries and download updates, so I would encourage library authors to use that mechanism.

Adding a link to this library to the "known working" libraries list certainly makes sense though.

aalku, 2016-04-09:
I think you are right. Thank you.

WereCatf, 2016-04-11:
@aalku The SdFat-beta - library now works properly with the ESP8266, both when running at 80MHz and at 160MHz, using hardware- or software-based SPI and both when using SPI-transactions and not.

I made a pull-request to add it to the libraries.

igrr, 2016-04-11:
Does it conflict with `#include <FS.h>` like SD library does?

WereCatf, 2016-04-11:
@igrr Yes, unfortunately.

aalku, 2016-04-11:
I think there is a core and an Arduino API around it. I think the File class is in the later. Maybe you can use the core API directly without conflicts. The Arduino API looks like a wrapper to offer SD library compatibility.

closed 2016-04-19 by igrr