esp8266/Arduino #1883: Wifi.begin() does not connect if no delay() in timout loop

hallard, 2016-04-08:
I've just spend some time trying to figure out why I was unable to connect to my AP anymore, I've got 10s timeout loop when trying to connect to AP, in this loop I'm ligthin some WS2812 LED but the problem was not LED.

What I was doing and not working was to pat the dog (to avoid WDT reset) and call my routine has follow
```arduino
unsigned long my_start = millis();
while ( ( ( ret =  WiFi.status() ) != WL_CONNECTED ) &&  (millis() - my_start < 10000) ) {
  // Was doing some stuff here but it's not the pb
  // LedRingAnimation();
  // Feed the dog
  ESP.wdtFeed();
}
```
With this code Wifi never get connected!! So I investigated with basic sample, and the added some delay in the loop like `delay(1)`
and that did the trick
```arduino
unsigned long my_start = millis();
while ( ( ( ret =  WiFi.status() ) != WL_CONNECTED ) &&  (millis() - my_start < 10000) ) {
  // Was doing some stuff here but it's not the pb
  LedRingAnimation();
  // Feed the dog/Do WiFi stuff
  delay(1);
}
```
So I guess `delay()` is doing some interesting things needed my WiFi to connect, if not called it just does not work.
Good to know.

WereCatf, 2016-04-08:
Did you try replacing ESP.wdtFeed() with yield() instead, does that change anything? And what if you place a single delay(1) before the while-loop instead of inside it?

hallard, 2016-04-08:
@WereCatf 
Yeah, `yield()` did it!
But only `delay(100)` before while not, but that makes sense ;-)

WereCatf, 2016-04-08:
Yeah, yield() feeds the watchdog and does all sorts of background-stuff if there is anything pending, including some WiFi-stuffs. If there is nothing to do, though, it takes a lot less time to execute than a delay(1) or similar.

hallard, 2016-04-08:
Yeah, thanks, I wanted to try this one, you confirmed worth trying 
now it's working ;-)

igrr, 2016-04-09:
Cases when you *have to* interact with WDT in the ESP are extremely rare and certainly outside of the normal scope of things you do in a sketch. As mentioned above, you need to give WiFi stack a chance to connect, and to do that, you must `yield` (or do a `delay`, which `yield`s internally). Since the system is cooperatively scheduled, rather than preemptively, WiFi stack will not get a chance to do its job unless you allow it to.

closed 2016-04-09 by igrr