esp8266/Arduino #1885: OTA Sketch why callbacks in setup?

danbicks, 2016-04-08:
Hi guys,

Going to ask a silly one here, In the basicOta sketch, why are the callbacks located within the setup subroutine?

Can they be placed outside of setup or is there a specific reason. I can understand that ArduinoOTA.begin(); needs to be there, but don't quite get why a separate callback such as

 ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\n", (progress / (total / 100)));
  });

is located within the setup scope.

Cheers

Dans

igrr, 2016-04-09:
You don't have to put them into setup — it's just a handy way to write short code fragments without having to declare another function.

closed 2016-04-09 by igrr

danbicks, 2016-04-10:
Thanks Ivan,

That is a neat way and make sense now.

Superb.

Cheers

Dans