esp8266/Arduino #1887: Test timing loop runs slower the longer it runs on ESP but not on AVR

digimeas, 2016-04-09:
Hi!  Just getting started with ESP and I LOVE it, but ...

A simple deterministic timing test slows the longer it runs on my nodemcu board (v1.0, USB + battery) but the same prog (except LED pin) on my AVR board (Pololu A-Star 32U4 Micro, just USB) does not.  

'Cause I'm an ESP-noob I was testing a new "blink" routine to see how tragic it was. The loop() func contains: 
`
  bdT -= micros();
  dqblinker();             //  routine I'm testing
  bdT += micros();
  
  tdT -= micros();
  if(bTmr <= millis())  //  similar code for comparison.  
    cnt2++;                 //  bTmr is extern uint32_t from dqblinker(), others are local.
  tdT += micros();

  and some Serial.print(results) stuff at the end of the cycle
`   (Huh!?   "Preview" doesn't look like "code")
where dqblinker() is the func I'm timing and the other block was to compare. All vars are (u)int32_t.

The dT's are printed at the end of the timing cycle which runs 3.25 seconds (3,250,xxx micros()), as expected. The longer the prog runs on the ESP, the longer both dT's get and fewer loop()/sec. When I run the same prog on the Pololu A-star, the dT's don't increase.  When I reset the ESP, the dT's start SHORTER than fresh upload, then lengthen.

First few few loops from ESP. The non-me stuff (the "tDt" code above) is:
  the second col (1573) which is cdt2 , above, which gets smaller, fewer hits/cycle;
  the fourth col "68482 l/s" => 68482 loop()'s/sec, which gets smaller, fewer loops/sec;
  the "3.8 u/l / 257389 u/s for tst" cols, => 3.8 usec/"call" and 257389 usec/sec (average) accumulated, which gets longer.
` 
BlinkerTester!
1: (1573)   3.250738 s/c.   68482 l/s.   4.3 u/l / 292000 u/s in blinker.   3.8 u/l / 257389 u/s for tst.   171.7 u/l for prt.   
2: (1576)   3.250357 s/c.   68516 l/s.   4.3 u/l / 292142 u/s in blinker.   3.8 u/l / 257503 u/s for tst.   121.8 u/l for prt.   
3: (1579)   3.250537 s/c.   68522 l/s.   4.3 u/l / 292152 u/s in blinker.   3.8 u/l / 257558 u/s for tst.   121.8 u/l for prt.   
`
Later + reset
`
44: (1504)   3.250536 s/c.   65479 l/s.   4.6 u/l / 301249 u/s in blinker.   4.1 u/l / 268187 u/s for tst.   122.4 u/l for prt.   
45: (1508)   3.250523 s/c.   65482 l/s.   4.6 u/l / 301276 u/s in blinker.   4.1 u/l / 268232 u/s for tst.   122.7 u/l for prt.   
46: (1507)   3.250531 s/c.   65484 l/s.   4.6 u/l / 301288 u/s in blinker.   4.1 u/l / 268243 u/s for tst.   122.4 u/l for prt.  
 <reset garbage>
BlinkerTester!
1: (1676)   3.250702 s/c.   72900 l/s.   3.8 u/l / 278616 u/s in blinker.   3.3 u/l / 241835 u/s for tst.   171.7 u/l for prt.   
2: (1633)   3.250366 s/c.   70948 l/s.   4.0 u/l / 284956 u/s in blinker.   3.5 u/l / 248984 u/s for tst.   121.8 u/l for prt.   
3: (1615)   3.250523 s/c.   70157 l/s.   4.1 u/l / 287232 u/s in blinker.   3.6 u/l / 251772 u/s for tst.   122.1 u/l for prt.   
`
Note:
  cycle 45's 268232 usec/sec is 4% longer than cycle 2's 257503 usec/sec;
  cycle 1's "257389 usec/sec" after upload is much shorter than cycle 46's "268243 usec/sec";
  cycle1's "241835 usec/sec" after reset is SHORTER than  cycle 1's "257389 usec/sec" after upload.

AVR results:
`
1: (791)   3.251180 s/c.   36166 l/s.   8.8 u/l / 318874 u/s in blinker.   6.2 u/l / 225068 u/s for tst.   896.9 u/l for prt.   
2: (790)   3.250912 s/c.   36161 l/s.   8.8 u/l / 318441 u/s in blinker.   6.2 u/l / 225138 u/s for tst.   895.7 u/l for prt.   
3: (827)   3.250900 s/c.   36160 l/s.   8.8 u/l / 318639 u/s in blinker.   6.2 u/l / 224988 u/s for tst.   892.1 u/l for prt.   
  ...
113: (790)   3.249784 s/c.   36159 l/s.   8.8 u/l / 318536 u/s in blinker.   6.2 u/l / 225119 u/s for tst.   921.9 u/l for prt.   
114: (792)   3.250812 s/c.   36161 l/s.   8.8 u/l / 318580 u/s in blinker.   6.2 u/l / 225086 u/s for tst.   920.4 u/l for prt.   
115: (826)   3.250812 s/c.   36160 l/s.   8.8 u/l / 318547 u/s in blinker.   6.2 u/l / 225097 u/s for tst.   920.4 u/l for prt.  
`
115 cycles of jitter, but no slowing.

Can't tell from "Hardware/" or "libraries/" folders what version of ESP stuff, but I'm using Arduino 1.6.8 and the board manager string 'http://arduino.esp8266.com/stable/package_esp8266com_index.json".

Enjoy!
Doug

P.s. I hope this doesn't look like the "preview" :-)  Let's see.

igrr, 2016-04-09:
Could you please post complete sketch so that I can reproduce this and help you? Thanks.

digimeas, 2016-04-09:
Hi Ivan!Thanks for your interest. Is this the easy way to post it?Don't laugh :-)Doug-----Original Message-----
From: Ivan Grokhotkov 
Sent: Apr 9, 2016 12:48 PM
To: esp8266/Arduino 
Cc: digimeas 
Subject: Re: [esp8266/Arduino] Test timing loop runs slower the longer it runs on ESP but not on AVR (#1887)

Could you please post complete sketch so that I can reproduce this and help you? Thanks.

igrr, 2016-04-11:
I'm sorry, i was trying to understand how your code works but i just can't get it.
Isn't the following sufficient to time your `dqblinker` function?

```c++
void loop() {
  uint32_t start = micros();
  dqblinker();
  uint32_t stop = micros();
  Serial.println(stop - start);
}
```

digimeas, 2016-04-12:
Hi @igrr 

I, too, hate looking at other's code :) This post wasn't about dqblinker(), but that the time spent running loop() increases , and loop()s/sec decreases with time running on anESP, but not when the identical program is run on an AVR.

I should have done this the first time. It's not like you don't have other things to do and we thank you immensely!!!!

Here's a stripped down version that shows the same slowing behavior. (Both "Quote" and "Code" look \<ahem\> not pretty, so here it is unformatted:

#if 1
  #define  bug    Serial.print
  #define  bugln  Serial.println
#else  
  #define  bug    (void)
  #define  bugln  (void)
#endif

const  uint32_t  TO=1*1000*1000;   //  << this doesn't work on an AVR, use TO=1000000
uint32_t  lcnt=0,  hcnt=0,  ccnt=0,  tmr, twas;
int32_t   tdT=0;

// 45678 0 2345678 0 2345678 0 2345678 0 2345678 0 2345678 0 2345678 0 2345678 0 2345678 0 2345678 0
void setup() {
  Serial.begin(115200);
  bug("\n TimerTester!   Cycle=");  bug(TO, DEC);  bug(" usec.");  bugln();
  tmr = micros() + TO;
  twas = micros();
}

// 45678 0 2345678 0 2345678 0 2345678 0 2345678 0 2345678 0 2345678 0 2345678 0 2345678 0 2345678 0
void loop() {
  lcnt++;
  
//  arbitrary, non-optimizable, test for timing
  tdT -= micros();                     //  accumulate time in test
  if(0 == (micros() % 10000))          //  the famous test :-)
    hcnt++;                            //  "hit" count
  tdT += micros();                     //  tdT += Tnow - Tbefore
  
  if(tmr <= micros()) {
    ccnt++;              //  "cycle" count
    float dT = float(micros() - twas)/1000000.;
    //
    bug(ccnt);  //  cycle count
    bug(":  Time=");  bug(micros());  bug(" usec.");
    //
    bug("  Loop:");  ///  bug(lcnt, DEC);  bug(" loop()/cycle => ");    //  loop()s / sec
    bug(lcnt/dT, 0);  bug(" loop()/sec.");                       //  loop()s / sec
    //
    bug("  Test:");  bug(tdT);  bug(" usec => ");                //  time spent in test (usec)
    bug(float(tdT)/lcnt, 1);  bug(" usec/test");                 //  usec / loop()
    /// bug(" => ");bug(tdT/dT, 0);  bug(" usec/sec.");              //  time spent / sec
    //
    bug("  (Hits=");  bug(hcnt);                       //  number of "hits" on if()
    /// bug(" => ");bug(hcnt/dT, 0);  bug(" hits/sec");                         //    / sec
    //
    bugln(")");
    //
    /// if(tgl)  digitalWrite(LED, HIGH);  else  digitalWrite(LED, LOW);
    lcnt = 0;
    hcnt = 0;
    tdT = 0;
    tmr += TO;
    twas = micros();
    }
}

Here's a few seconds of output

>  TimerTester!   Cycle=1000000 usec.
1:  Time=19800086 usec.  Loop:108233 loop()/sec.  Test:291282 usec => 2.7 usec/test  (Hits=10)
[... 51 lines cut ...]
18:  Time=18252687 usec.  Loop:108359 loop()/sec.  Test:294013 usec => 2.7 usec/test  (Hits=12

The same program on the AVR (actually the 8 bit AVR didn't like the construct TO=1\*1000\*1000 so it's now TO=1000000; otherwise unchanged) produced:
>1:  Time=1001468 usec.  Loop:18045 loop()/sec.  Test:802260 usec => 44.5 usec/test  (Hits=6)
2:  Time=2001508 usec.  Loop:17963 loop()/sec.  Test:801776 usec => 44.7 usec/test  (Hits=9)
3:  Time=3001472 usec.  Loop:17944 loop()/sec.  Test:801796 usec => 44.8 usec/test  (Hits=6)
4:  Time=4001488 usec.  Loop:17922 loop()/sec.  Test:802332 usec => 44.8 usec/test  (Hits=6)
5:  Time=5001488 usec.  Loop:17880 loop()/sec.  Test:802200 usec => 44.9 usec/test  (Hits=5)

and five minutes later:
>327:  Time=327001576 usec.  Loop:17546 loop()/sec.  Test:805648 usec => 46.0 usec/test  (Hits=7)
328:  Time=328001580 usec.  Loop:17656 loop()/sec.  Test:805032 usec => 45.7 usec/test  (Hits=8)
329:  Time=329001548 usec.  Loop:17926 loop()/sec.  Test:802260 usec => 44.8 usec/test  (Hits=7)
330:  Time=330001560 usec.  Loop:17895 loop()/sec.  Test:801972 usec => 44.9 usec/test  (Hits=6)
331:  Time=331001556 usec.  Loop:17888 loop()/sec.  Test:802356 usec => 44.9 usec/test  (Hits=9)

Jitter but no monotonic slowing.

Here's the prog and output 
[TimerTester.zip](https://github.com/esp8266/Arduino/files/215657/TimerTester.zip)

Again, NodeMCU v1.0, Pololu A-Star 32U4 Micro, Arduino 1.6.8.

Thanks again for your interest and time
Doug

igrr, 2016-04-13:
Okay, so I went to test three hypothesis:
- there is a growing delay between calls to `loop` function
- microsecond counter is slowing down (with respect to the CPU frequency)
- calls to `micros` function start to take longer

Here is a pretty simple sketch which tests all three:
https://gist.github.com/igrr/3c57d465a300b74b42a3d937c8381be1

`measure_time` is a helper function which measures time taken to execute a function passed as an argument.

First we measure how long does it take to spin in a busy loop for 1000000 iterations. This is `empty_loop_duration`, in microseconds.

Next we measure how long does it take to spin in a loop, calling `micros` on each iteration, for 1000000 iterations. This is `micros_loop_duration`, in microseconds.

Finally we save the timestamp before exiting `loop`. Upon next `loop` call, we calculate the time passed from exiting `loop` to entering `loop` again.

Serial output follows. Looking at the results, i don't see any monotonic increase in any of three timing values, which probably indicates that (in these conditions) none of the three hypothesis is valid.
I suppose that your sketch measures some other effect, although I have no idea what it is.

```
Loop    0 t(empty loop)=187579 t(micros loop)=862523 t(loop end -> loop start)=5649111
Loop    1 t(empty loop)=187519 t(micros loop)=862511 t(loop end -> loop start)=14
Loop    2 t(empty loop)=187504 t(micros loop)=862501 t(loop end -> loop start)=6
Loop    3 t(empty loop)=187504 t(micros loop)=862520 t(loop end -> loop start)=6
Loop    4 t(empty loop)=187504 t(micros loop)=862511 t(loop end -> loop start)=6
Loop    5 t(empty loop)=187504 t(micros loop)=862501 t(loop end -> loop start)=5
Loop    6 t(empty loop)=187573 t(micros loop)=862521 t(loop end -> loop start)=5
Loop    7 t(empty loop)=187504 t(micros loop)=862511 t(loop end -> loop start)=5
Loop    8 t(empty loop)=187504 t(micros loop)=862511 t(loop end -> loop start)=6
Loop    9 t(empty loop)=187504 t(micros loop)=862501 t(loop end -> loop start)=6
rl l��| �l�|�l�b|����r�b�b��nn�lnn���bp��lrlrlp�n��l��bn�|l����b��nn� l��l`�nnl`nr���n��lr��n��l��n�
start
Loop    0 t(empty loop)=187567 t(micros loop)=862535 t(loop end -> loop start)=1245706
Loop    1 t(empty loop)=187524 t(micros loop)=862511 t(loop end -> loop start)=14
Loop    2 t(empty loop)=187504 t(micros loop)=862501 t(loop end -> loop start)=6
Loop    3 t(empty loop)=187504 t(micros loop)=862520 t(loop end -> loop start)=6
Loop    4 t(empty loop)=187504 t(micros loop)=862511 t(loop end -> loop start)=6
Loop    5 t(empty loop)=187504 t(micros loop)=862501 t(loop end -> loop start)=6
Loop    6 t(empty loop)=187573 t(micros loop)=862522 t(loop end -> loop start)=6
Loop    7 t(empty loop)=187504 t(micros loop)=862511 t(loop end -> loop start)=6
Loop    8 t(empty loop)=187504 t(micros loop)=862511 t(loop end -> loop start)=6
Loop    9 t(empty loop)=187504 t(micros loop)=862501 t(loop end -> loop start)=5
Loop   10 t(empty loop)=187504 t(micros loop)=862519 t(loop end -> loop start)=5
Loop   11 t(empty loop)=187504 t(micros loop)=862511 t(loop end -> loop start)=6
Loop   12 t(empty loop)=187575 t(micros loop)=862501 t(loop end -> loop start)=5
Loop   13 t(empty loop)=187514 t(micros loop)=862510 t(loop end -> loop start)=5
Loop   14 t(empty loop)=187505 t(micros loop)=862511 t(loop end -> loop start)=5
Loop   15 t(empty loop)=187504 t(micros loop)=862511 t(loop end -> loop start)=6
Loop   16 t(empty loop)=187504 t(micros loop)=862501 t(loop end -> loop start)=6
Loop   17 t(empty loop)=187504 t(micros loop)=862520 t(loop end -> loop start)=6
Loop   18 t(empty loop)=187563 t(micros loop)=862523 t(loop end -> loop start)=6
Loop   19 t(empty loop)=187504 t(micros loop)=862501 t(loop end -> loop start)=6
Loop   20 t(empty loop)=187504 t(micros loop)=862501 t(loop end -> loop start)=6
Loop   21 t(empty loop)=187504 t(micros loop)=862556 t(loop end -> loop start)=6
Loop   22 t(empty loop)=187504 t(micros loop)=862511 t(loop end -> loop start)=10
Loop   23 t(empty loop)=187504 t(micros loop)=862501 t(loop end -> loop start)=6
Loop   24 t(empty loop)=187564 t(micros loop)=862531 t(loop end -> loop start)=5
Loop   25 t(empty loop)=187505 t(micros loop)=862510 t(loop end -> loop start)=5
---snip---
Loop  175 t(empty loop)=187504 t(micros loop)=862511 t(loop end -> loop start)=6
Loop  176 t(empty loop)=187504 t(micros loop)=862501 t(loop end -> loop start)=6
Loop  177 t(empty loop)=187505 t(micros loop)=862520 t(loop end -> loop start)=5
Loop  178 t(empty loop)=187504 t(micros loop)=862511 t(loop end -> loop start)=6
Loop  179 t(empty loop)=187504 t(micros loop)=862501 t(loop end -> loop start)=6
```

digimeas, 2016-04-14:
TL;DR ESP slows and I've got plots.

Hi @Igrr!   This is fun. Like old-fashion postcard chess.

Another hypothesis: it's the number of loop()s. Your program ran a few hundred, mine several million.
I re-ran on the AVR for a longer time and guess what?  It slowed too (I thought), but over longer times one could see that the slowing was not monotonic.

I plotted the data using octave and saw weird bunching in the loops/sec and "test times" for both processors.

If you're interested, TimerTester.ino, the serial data, octave .m files and plots are in 
[TimerTester.zip](https://github.com/esp8266/Arduino/files/218171/TimerTester.zip)

Figure 1: loops per second vs. time.
Figure 2: diff(loops per second) vs. time  Very interesting patterns.  Pascal-y.
Figure 3: "test time" vs. time.
Figure 4: diff(test time) vs. time.
Figure 5: reported micros() at each print vs. time.  Detrended.
Figure 6: diff(micros()) vs. time.

Figures 1 show a weird bunching "stick-slip" pattern for both processors. 
  The AVR jitter is about 200 loops/sec on about 17700 lps, ~1%.  The AVR slows (with jitter) for ~300 seconds then jumps 400 lps then slows, etc.  The overall trend is pretty flat, though.
  The ESP's jitter is ~2000 lps on 108000-101000 lps, ~2%.  The ESP's slowing trend is very obvious, though it speeds up for about 200 seconds then falls, etc.

Figures 2 show an interesting pattern with a 1000 second period (1024?) for both processors.  It reminds me of something I feel I should remember, but nothing's coming.  Pretty though.

Figures 3 are similar to Figures 1.

Figures 4 are dull; some scatter but no patterns like Figures 2.

Figures 5 & 6 are also dull; just the clock scatter.
 
 
So, I can see something's going on but haven't a clue what.  Does look interesting.

Let me know if you find something.  Thanks,
Doug

Makuna, 2016-05-12:
I would like to add, the way you are doing your math is invalid.  I will take a single line
```
tdT -= micros();                     //  accumulate time in test
```
when the program runs long enough, micros() will return a number larger that the int32_t tdT can hold and you can get a overflow (underflow).

what is (0 - 65000) equal in a signed value?

digimeas, 2016-05-12:
Hi @Makuna.  It's perfectly valid (if you don't run it too long :-)  The original implementation was intended to run a few seconds to collect some timing data.  And, as you point out, it x-flows at 71+ minutes.  But that's not really the point of the post.

Something weird is happening long before tDt glitches.  See the plot in Figure 1, above.
Doug

devyte, 2017-09-26:
@digimeas I implemented my own test, measuring the time outside of loop() and inside of loop(). Essentially, I call millis() at the start of loop() and at the end of loop() within my application (complicated webserver with dht sensor). After 15mins running, I agree with @igrr: there is no increment.
Your calculations seem overly complicated for measuring something so trivial as timing. For starters, you shouldn't be calling micros() all over the place, you should instead call it once, maybe twice, assign to a variable, and use the result for all your calculations.
Given that in addition you are observing the same behavior for both AVR and ESP, which is unlikely, I suspect that your calculations have some bug.
I am therefore closing.

closed 2017-09-26 by devyte