esp8266/Arduino #1889: different ESP board diferent web server behiavior

mkeyno, 2016-04-09:
Hi dear @igrr 
I run the modified  webserver example of last stable code in 3 different ESP board, there is couple of pages  on flash but strange is whereas speed and performance is good on NodeMCU(ESP-12 core) module but for the  ESP-01/1MB flash , ESP-07/56KB flash modules most time pages not loaded or partially loaded , it seems same code run on the  ESP-01/1MB flash , ESP-07/56KB flash so slowly regards to ESP-12E core , I  choose different  Settings in IDE for ESP-01 & ESP-07 (such as Generic ESP, ESPdunio , Sparkfun ESP thing ,...) . I can program and upload data for couple of setting and not for others , is there any special setting for such module ? how could I have same performance like ESP-12E?is this probably related to @luc-github [issue ](https://github.com/luc-github/ESP8266/issues/85)

BR

igrr, 2016-04-09:
ESP-xx modules do not have any differences except for their physical layout and antenna options. If something works on one module but doesn't work on another, you might want to check if
- the misbehaving module is adequately powered and decoupled
- the code is compiled for the correct flash size. AFAIK there are no 56kb flash modules, so perhaps you need to double-check that specification
- erasing flash completely (using `esptool.py` or `esptool-ck`) may help remove corrupted SDK config

mkeyno, 2016-04-09:
thanks dear @igrr  but why module has different behavior when we select different Settings in IDE (name of board , speed of flash , reset method, flash size,...) why each vendor define it's own Settings in IDE? what's the benefit of that ?
so sorry for this much questions

WereCatf, 2016-04-09:
@mkeyno Because some manufacturer uses 512KB flash, another uses 4MB flash, some have cut off the two pins that go to the flash so it can only be driven in DIO-mode, some have all the pins connected, allowing the flash to be driven in the much faster QIO-mode, and then there are the pin-definitions, like e.g. on the Nodemcu the pin that's marked as D0 is actually GPIO16, whereas on another manufacturer's board a similarly-named pin D0 could be some entirely different GPIO-pin underneath, or they may use some other naming-scheme altogether -- these board-specific pin-definitions are there for ease-of-use reasons.

Basically, a lot of settings that vary between manufacturer to manufacturer.

mkeyno, 2016-04-10:
thanks @WereCatf  for your answer , so you were saying the only main different is the flash size & flash connection type(hardware connection) , other should not has much impact on module  programming , however I was wondering if you have any suggestion for ESP-07 IDE setting

WereCatf, 2016-04-10:
There is no such a thing as an ESP8266 with only 56KB Flash, you probably mean 512KB. Setting IDE to generic ESP and flash-mode to DIO should work every ESP-module out there, you just have to make sure the flash-size is correct. In your case, it might be a good idea to format your SPIFFS, too, to make sure it's not corrupted.

mkeyno, 2016-04-10:
thanks @WereCatf , I ran some test to find correct size of flash and SPIFF, and it run now but somethings new come  up , I put about 200kb pages on the SPIFF and simple webserver to serve those pages , but when I connected to AP, it shows nothings , no request form browser send to AP node whereas my laptop correctly connected to that AP,  I've tried with other browser but nothings happened, it seems AP can not received any header request from browser  whereas with example of library , node received all request form browser , I might add my sketch totally work all browser 

module ESP-07
flash size 1mb , SPIFF size 512 or 256
sketch size  270kb 
IDE Arudino 1.6.7
`

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <DNSServer.h>
#include <FS.h>
#include <WebSocketsServer.h>
#include <Hash.h>
#include <EEPROM.h>

WebSocketsServer webSocket = WebSocketsServer(81);
ESP8266WebServer server(80);
const byte DNS_PORT = 53;
DNSServer dnsServer;
/// parameters
IPAddress apIP(192, 168, 4, 1),S_IP;

IPAddress Gateway(192, 168, 4, 1);
IPAddress netMsk(255, 255, 255, 0);

File fsUploadFile;

void EEPROM_Write(String s,byte from, byte to)
{
  for (byte i = from; i < to; i++) EEPROM.write(i, 0); //earase
  for (byte i = from; i < from+s.length() && i < to; i++) EEPROM.write(i, s[i-from]);
  EEPROM.commit();
}

void webSocketEvent(uint8_t num, WStype_t type, uint8_t * payload, size_t lenght) {

 switch(type) {
        case WStype_DISCONNECTED:
                Serial.printf("[%u] Disconnected!\n", num); 
    break;
        case WStype_CONNECTED:   { 
                             IPAddress ip = webSocket.remoteIP(num);  
                  Serial.print("Clinet "); Serial.print(num); Serial.print(" CONNECTED with IP: ");Serial.println(ip);
                 } 
     break;
        case WStype_TEXT:    // this for  webSocket request          
              String text = String((char *) &payload[0]);
              Serial.print("webSocket recieved: ");Serial.println(text);              

        break;
        
        
      }

}

String formatBytes(size_t bytes){
       if (bytes < 1024)                 return String(bytes)+"B"; 
  else if(bytes < (1024 * 1024))         return String(bytes/1024.0)+"KB";
  else if(bytes < (1024 * 1024 * 1024))  return String(bytes/1024.0/1024.0)+"MB";
  else                                   return String(bytes/1024.0/1024.0/1024.0)+"GB";
}

String getContentType(String filename){
       if(server.hasArg("download")) return "application/octet-stream";
  else if(filename.endsWith(".htm")) return "text/html";
[... 150 lines cut ...]
  server.collectHeaders(headerkeys, hz );
  
  webSocket.begin();
  webSocket.onEvent(webSocketEvent);
}

void loop() 
{
 
  dnsServer.processNextRequest();
  server.handleClient();  
  webSocket.loop();
  
`

out put 
`
Starting.....

 Entering the Access Mode
AP IP address: 192.168.4.1
HTTP server started

`
if any page by browser requested should print the `Serial.print(" Request page : " + path);` but nothings happened

WereCatf, 2016-04-10:
So, the same sketch, with no modifications to it, works correctly on the Nodemcu but not your ESP-07?

I see you are using Arduino IDE 1.6.7, are you also using an older version of the ESP8266-stuff for Arduino? Perhaps you could try installing IDE 1.6.8 and grabbing the latest stuff from here and testing if it works any better?

If the answer to the first question is yes, and it doesn't work any better with IDE 1.6.8 and latest stuff from here, then it sounds like a bug in the SPIFFS-library or the webserver-lib.

mkeyno, 2016-04-10:
dear @WereCatf  I ran another test , I upload the same file on SPIFF and sketch with two same module (nodeMCU) but supersized when one module work file and other not ,  one  get any request from browser and respond fast and other seems dumb to any browser request , how is happened ? when I connected to AP node I set the both network on home type in windows 7 64x

mkeyno, 2016-04-10:
following is the setup and loop code , the only different is the module have different EEPOM and may restore different parameter in setup but the other code is same
`
void setup() 
{
   
  Serial.begin(115200);
  Serial.println(F("\n\n\n Keyno Guidance & Control\nFirmware ver(1.01)\nwww.Keyno-GC.com\nStarting....."));
  pinMode(0,OUTPUT); digitalWrite(0,HIGH);
  pinMode(2,OUTPUT); digitalWrite(2,HIGH);
  SPIFFS.begin();
  EEPROM.begin(128); 
  
    MyMode=EEPROM_Read(0,1);
  myHostname =EEPROM_Read(1,15); //add 1-14
         SSID=EEPROM_Read(15,30); //add 15-29
     ssidPass=EEPROM_Read(30,45);//add 30-44
      AP_SSID=EEPROM_Read(45,60);//add 45-59
  AP_ssidPass=EEPROM_Read(60,75);//add 60-74
       IDsNum=EEPROM_Read(75,76).toInt();//add 75
 reset_number=EEPROM_Read(76,77).toInt();//add 76
 
 
  Serial.print("Mode=");          Serial.println(MyMode);
  Serial.print("Hostname=");      Serial.println(myHostname);
 Serial.print("SSID=");           Serial.println(SSID);
 Serial.print("Password =");      Serial.println(ssidPass);
 Serial.print("AP SSID=");        Serial.println(AP_SSID);
 Serial.print("AP Password =");   Serial.println(AP_ssidPass);
 Serial.print("reset_number =");  Serial.println(reset_number);
 Serial.print("Server IP Address =");  Serial.println(S_IP);
 if(MyMode[0]=='A' || reset_number>4 || SSID.length()<3 )
                            {
                               Serial.println("\t\t\t Entering the Access Mode");
                               WiFi.mode(WIFI_AP);
                               WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
                               String temp_name="Smart Node_" +String(ESP.getChipId(), HEX);
                               if(AP_SSID.length()<3) WiFi.softAP(temp_name.c_str(),"12345678");
                               else                   WiFi.softAP(AP_SSID.c_str(),AP_ssidPass.c_str());                                
                               delay(500); // Without delay I've seen the IP address blank
                               EEPROM_Write(String(0),76,77);   
                               PerfixID=WiFi.softAPIP().toString().substring(8);
                               Serial.print("AP IP address: ");Serial.println(WiFi.softAPIP());
                               dnsServer.setErrorReplyCode(DNSReplyCode::ServerFailure);
                               dnsServer.start(DNS_PORT, "*", apIP);  
                            }
else if(reset_number<5)
            {
              Serial.println("\t\t\t Entering the Station Mode");
              WiFi.mode(WIFI_STA);
              WiFi.begin(SSID.c_str(), ssidPass.c_str());             
                unsigned long startTime = millis();
                while (WiFi.status() != WL_CONNECTED && millis() - startTime < 10000)  {Serial.write('%');delay(500); } 
                if(WiFi.status() != WL_CONNECTED) {
                                              reset_number++;
                                EEPROM_Write(String(reset_number),76,77);
                                ESP.restart();
                                }            
               PerfixID=WiFi.localIP().toString().substring(8);
               Serial.print("AP IP address: ");Serial.println(WiFi.localIP());
              if (!MDNS.begin(myHostname.c_str())) {
                              Serial.println("Error setting up MDNS responder!");
                              while(1) { 
                                delay(1000);
                              }
                            }
              Serial.println("mDNS responder started");
            }
  server.on("/process", process);    
  server.on("/update", HTTP_POST, [](){ Serial.println("Upload post");  /* server.send(200, "text/plain", ""); */ Redirect("setting");  }, handleFileUpload);
  server.on("/login", handleLogin); 
  server.on("/check", check); //check user name pass word link
  //server.on("/", [](){   if(is_authentified()) Redirect("/"); else redirect2login();  }); //check user name pass word link
  server.onNotFound(handleOther);
  server.begin();
  Serial.println("\t\t\tHTTP server started");  
   //here the list of headers to be recorded
  const char * headerkeys[] = {"User-Agent","Cookie"} ;
  size_t hz = sizeof(headerkeys)/sizeof(char*);
  //ask server to track these headers
  server.collectHeaders(headerkeys, hz );  
  webSocket.begin();
  webSocket.onEvent(webSocketEvent);
}

void loop() 
{
  
 if(MyMode[0]=='A') dnsServer.processNextRequest();
  server.handleClient();  
  webSocket.loop();
 
}

`

WereCatf, 2016-04-10:
So, you didn't try with newer IDE and ESP8266-libs?

mkeyno, 2016-04-10:
not for IDE but my ESP8266-libs is the last one

mkeyno, 2016-04-10:
I did , I test it with Arduino 1.6.8 hourly nights , the same issue , it is possible somehow my node not registered for windows 7 and no request pass it to my AP node

mkeyno, 2016-04-10:
dear @igrr do you have any idea why same sketch and file have different behavior for same module

mkeyno, 2016-04-10:
dear @igrr  & @WereCatf  I set debug level to "All" in Arduino deice setting (Generic ESP8266)  but what is out come is the endless following info, I don't know why my code work fine with one module and respond browser request but other not or partially catch the browser header request 
wifi evt: 7
chg_A2:-40
wifi evt: 7
add 1
aid 1
station: b8:76:3f:45:64:43 join, AID = 1
wifi evt: 5
:urn 43
:urch 43, 50
:urch 93, 35
:urch 128, 48
:urch 176, 51
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
wifi evt: 7
wifi evt: 7
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
wifi evt: 7
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
wifi evt: 7
wifi evt: 7
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
wifi evt: 7
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
wifi evt: 7
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
-----------------------endless

WereCatf, 2016-04-10:
You are absolutely certain that your module is getting enough power? Unstable power-supply or too little power could also cause something like this, I guess. I do not have an ESP-07 to test with.

mkeyno, 2016-04-10:
I've check with NodeMCU, one module work and respond find and other not, can you kindly advise me how can enable  debug  on nodeMCU just like the generic esp8266 , it seems this option not available for nodeMCU

WereCatf, 2016-04-10:
You can use the same generic settings for the Nodemcu-kit as you use for the ESP-07, except set Flash to 4MB - size. Ie. flash size to 4MB, flash-mode to DIO.

mkeyno, 2016-04-10:
no change buddy
wifi evt: 7
add 1
aid 1
station: b8:76:3f:45:64:43 join, AID = 1
wifi evt: 5
:urn 43
:urch 43, 45
:urch 88, 35
:urch 123, 47
:urch 170, 52
LmacRxBlk:1
LmacRxBlk:1
wifi evt: 7
wifi evt: 7
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
wifi evt: 7
wifi evt: 7
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
chg_A2:-40
LmacRxBlk:1
wifi evt: 7
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
wifi evt: 7
wifi evt: 7
LmacRxBlk:1
LmacRxBlk:1
wifi evt: 7
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
wifi evt: 7
wifi evt: 7
wifi evt: 7
LmacRxBlk:1
LmacRxBlk:1

mkeyno, 2016-04-10:
no change , I'm so confuse and exhausted, how come  pair of same module behave differently ?

WereCatf, 2016-04-10:
Did you check the power-supply you're using for the ESP-07?

mkeyno, 2016-04-10:
right now I work with 2 pair of nodeMCU rev 1conected to my laptop USB

WereCatf, 2016-04-10:
I didn't ask about your Nodemcu, since that's working. I asked about the ESP-07.

mkeyno, 2016-04-10:
when I have continue issue with my ESP-07 I've change it with nodeMCU but code work with one and not work for other

mkeyno, 2016-04-10:
@WereCatf  I got some info , in my code there is function which is restore some parameter from EEPROM , when I  disable it the module work fine, the restore parameter has not to do with wifi parameter and setting , how this affect the wifi performance

mkeyno, 2016-04-10:
`
void setup()
{

Serial.begin(115200);
SPIFFS.begin();
EEPROM.begin(128);

                                                           MyMode=EEPROM_Read(0,1);
                                                            myHostname =EEPROM_Read(1,15); //add 1-14
                                                          SSID=EEPROM_Read(15,30); //add 15-29
                                                          ssidPass=EEPROM_Read(30,45);//add 30-44
                                                          AP_SSID=EEPROM_Read(45,60);//add 45-59
                                                          AP_ssidPass=EEPROM_Read(60,75);//add 60-74
 
if(MyMode[0]=='A' || reset_number>4 || SSID.length()<3 )
{
           Serial.println("\t\t\t Entering the Access Mode");
           WiFi.mode(WIFI_AP);
           WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
           String temp_name="Smart Node_" +String(ESP.getChipId(), HEX);
           if(AP_SSID.length()<3) WiFi.softAP(temp_name.c_str(),"12345678");
           else WiFi.softAP(AP_SSID.c_str(),AP_ssidPass.c_str());
           delay(500); // Without delay I've seen the IP address blank
           EEPROM_Write(String(0),76,77);
           PerfixID=WiFi.softAPIP().toString().substring(8);
           Serial.print("AP IP address: ");Serial.println(WiFi.softAPIP());
           dnsServer.setErrorReplyCode(DNSReplyCode::ServerFailure);
           dnsServer.start(DNS_PORT, "", apIP);

}
else if(reset_number<5)
{
           Serial.println("\t\t\t Entering the Station Mode");
           WiFi.mode(WIFI_STA);
           WiFi.begin(SSID.c_str(), ssidPass.c_str());
           unsigned long startTime = millis();
           while (WiFi.status() != WL_CONNECTED && millis() - startTime < 10000)                                  {Serial.write('%');delay(500); }
          if(WiFi.status() != WL_CONNECTED) {
          reset_number++;
          EEPROM_Write(String(reset_number),76,77);
          ESP.restart();
}

          PerfixID=WiFi.localIP().toString().substring(8);
          Serial.print("AP IP address: ");Serial.println(WiFi.localIP());
          if (!MDNS.begin(myHostname.c_str())) {
          Serial.println("Error setting up MDNS responder!");
          while(1) {
                    delay(1000);
                              }
}
          Serial.println("mDNS responder started");
}

server.on("/process", process);
server.on("/login", handleLogin);
server.on("/check", check); //check user name pass word link
server.onNotFound(handleOther);
server.begin();
Serial.println("\t\t\tHTTP server started");
//here the list of headers to be recorded
const char * headerkeys[] = {"User-Agent","Cookie"} ;
size_t hz = sizeof(headerkeys)/sizeof(char);
//ask server to track these headers
server.collectHeaders(headerkeys, hz );
webSocket.begin();
webSocket.onEvent(webSocketEvent);
}

void loop()
{

if(MyMode[0]=='A') dnsServer.processNextRequest();
server.handleClient();
webSocket.loop();

}

`

mkeyno, 2016-04-10:
the module that already work has preset EEPROM, but when I program my code on new module there is no preset EEPROM data , how this brutally affect the module performance , dear @WereCatf  can you kindly check my setup function

WereCatf, 2016-04-10:
There are no such functions in the library as EEPROM_Read and EEPROM_Write, are those something you've written yourself?

mkeyno, 2016-04-10:
yes buddy
void EEPROM_Write(String s,byte from, byte to)
{
  for (byte i = from; i < to; i++) EEPROM.write(i, 0); //earase
  for (byte i = from; i < from+s.length() && i < to; i++) EEPROM.write(i, s[i-from]);
  EEPROM.commit();
}
`
String EEPROM_Read(byte from, byte to)
{
  String s="";
  for (byte i = from; i < to; i++)
                  {
                  byte c=EEPROM.read(i);
                  if(c>31 && c<128)
                  s += char(c);
                  }
  return s;
}
`

closed 2016-04-11 by mkeyno