esp8266/Arduino #1890: esptool problem in esp8266 2.1.0

smallworld-network-wupeng, 2016-04-10:
### Basic Infos

#### Hardware
Hardware:	        ESP-12E
Core Version:      	2.1.0

### Description

I update to ESP8266 2.1.0 and use FTDI232 . I can not upload sketch to esp8266 , this is the error log:

> esptool v0.4.8 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to nodemcu
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/cu.usbserial-A400f0Tx
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/cu.usbserial-A400f0Tx at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed

========================================================
I replace the esptool to 0.4.6 , it is can upload sketch to esp8266 when I select NodeMCU1.0(ESP-12E Module). But if I select Generic ESP8266 Module upload failed.

> esptool v0.4.6 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/cu.usbserial-A400f0Tx
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	stat /var/folders/m5/h87zjffj7hb5xzyg_11xflt40000gn/T/build7bdcd80118947ef29b6eb1f6f650d5b7.tmp/irobot_ir_esp8266.ino.bin success
opening port /dev/cu.usbserial-A400f0Tx at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
warning: espcomm_sync failed
error: espcomm_open failed

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/3MB
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:	       SERIAL?
Reset Method:		ck

igrr, 2016-04-10:
First log says:
    setting board to nodemcu

Second log says:
    setting board to ck

Apparently only one of the two will work. How is your module wired to
FT232's DTR and RTS pins?

smallworld-network-wupeng, 2016-04-10:
No, I just wired GND TX RX. It is work well in esp8266 2.0 with esptool 0.4.6 .  I make GPIO15 to low, GPIO0 and GPIO2 to high , and wired GND TX RX , it is work very well before.

smallworld-network-wupeng, 2016-04-10:
oh, it is worked ,when I setting board reset mode to nodemcu !

closed 2016-04-10 by igrr

sarwadenj, 2017-04-08:
Following thing coming again and again while uploading

esptool v0.4.6 - (c) 2014 Ch. Klippel ck@atelier-klippel.de
setting board to ck
setting baudrate from 115200 to 115200
setting port from COM1 to COM7
setting address from 0x00000000 to 0x00000000
espcomm_upload_file

My connection are 
http://www.instructables.com/id/Getting-Started-with-the-ESP8266-ESP-12/?ALLSTEPS

GPIO 0--GND (WHILE UPLOADING)
GPIO 15-->GND
GPIO 2-->VCC