esp8266/Arduino #1891: C++ function never returns back to loop

PhilG, 2016-04-10:
Hi,
I am not even really sure if this is the right place to ask, but here goes: I'm programming a little framework to easily add different kinds of animations for a WS2812B LED Strip. Now the code below is an excerpt and when I flash all of my code onto the ESP it runs until I see `"Fill::update() is over"` in the serial output. It then hangs and eventually the WDT resets the ESP since either `Fill::update()` does not return or `Animator::update()` does not return to loop and hence there is no delays the ESP can't do his WiFi magic. 

Now is this anything that could be a bug or am I getting something seriously wrong because I have absolutely no clue why those functions wouldn't return.

### Sketch

```cpp
void loop() {
  Serial.println("before anim.update()");
  anim.update();
  Serial.println("after anim.update()");
}
```

Animator.cpp
```cpp
void Animator::update() {
  if(s_state == state_t::PLAYING && s_activeAnim) {
    Serial.println("animating next frame");

    s_activeAnim->update(s_strip);
  }
  return;
}
```

Fill.cpp
```cpp
void Fill::update(Adafruit_NeoPixel& strip) {
  Serial.println("Fill::update()");
  
  unsigned long currentMillis = millis();
  unsigned long delta = currentMillis - m_lastMillis;
  if(delta >= m_delay) {
    m_lastMillis = currentMillis;
    
    if(m_count >= strip.numPixels()) {
      m_count = 0;
    }
    
    for(int i = 0; i <= m_count; i++) {
      strip.setPixelColor(i, m_color);
    }
    strip.show();
    m_count++;
  }

  Serial.println("Fill::update() is over");
  //Serial.printf("%x\n", ESP.getFreeHeap());
  //Serial.println(ESP.getResetReason());
}
```

#### Hardware
Hardware:			ESP-01
Core Version:      	2.1.0

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1MB
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

PhilG, 2016-04-10:
So I just "confirmed" that it is the call of `s_activeAnim->update(s_strip);` that doesn't return by adding a `Serial.println` behind it which did not get called.

WereCatf, 2016-04-10:
I'm going to assume that your Fill::update() takes so long to execute that it triggers the watchdog-timer. Try adding a call to yield() in the for-loop and see if that fixes it.

PhilG, 2016-04-10:
It's not taking long I think and I see the Serial outputs at the end of it right before it returns, then nothing happens for a few seconds and then I get the `wdt reset`.

PhilG, 2016-04-10:
I think something starts to go off here as well since on boot the first LED is already green which it shouldn't be.

```cpp
Adafruit_NeoPixel strip = Adafruit_NeoPixel(60, STRIP_PIN, NEO_GRB + NEO_KHZ800);
Animator anim = Animator(&strip);
```

WereCatf, 2016-04-10:
I don't have these LEDs, so I can't comment on that. If adding the call to yield() doesn't fix things then I'd say the library is faulty and it may be overwriting stack, corrupting the function return-address.

PhilG, 2016-04-10:
It seems to be an issue of either the library or my code really... Apparently passing around a pointer to the strip object will set the first LED to green? I have no idea right now but I'll go back and dig

closed 2016-04-10 by PhilG

igrr, 2016-04-10:
Adafruit Neopixel library disables interrupts inside ::show function, this can cause WiFi stack to crash (which is what seems to be happening). Better try using a version of Neopixel library from @Makuna which uses UART or I2S and doesn't disable interrupts.

PhilG, 2016-04-10:
I could run the strandtest example just fine with WiFi connected but I will try those, thank you!

TheAustrian, 2016-04-10:
FastLED also allows interrupts for the ESP in the standard config. https://github.com/FastLED/FastLED/

PhilG, 2016-04-10:
I've been looking at FastLED before :) In my setup I'm pretty sure it's not the interrupts though because the ESP can keep a connection up and run the strandtest example, just tried it again.

PhilG, 2016-04-10:
I've changed the general usage of my code a bit now and it works fine!

Also I found out that the order of these instructions matters and that `strip.clear()` is important.
```cpp
strip.begin();
  
WiFi.begin(ssid, password);

strip.clear();
strip.show(); // Initialize all pixels to 'off'
```

Somehow `WiFi.begin(ssid, password)` manages to set the first LED on the strip to green and if you do all the LED setup before that you're stuck with a green LED until the main program starts.

Makuna, 2016-04-10:
@PhilG As igrr stated, avoid the Adafruit NeoPixel library for Esp8266 platform.  It was based on my old  BitBang (they gave me credit) but it is erratic if you have WiFi traffic and can cause crashes.  Even LadyAda states this several times in her videos.  The issue arose last summer with an SDK change; so it worked fine for a few months.
Using the strandtest example is not a good test for this issue as it doesn't exercise the WiFi stack with actual traffic.  Its the handling of the traffic in the WiFi core that is sensitive to the interrupts being disabled.

You also need to be aware that pins are used for all sorts of reasons, so some care is needed for picking the pin.  You may get unexpected results during flashing, boot-up, or using other libraries that may use pins.  Some hardware features are restricted to their pin use.

My library has a very similar API to the AdaFruit library, and my low level support for other platforms than esp8266 is their code.  But the architecture of it allows for easy switching between methods of how data is sent, for esp8266, you can switch between BitBang, Uart, and I2s DMA; with the default being DMA.  For Esp01 you will need to use the Uart as outlined in the [Wiki](https://github.com/Makuna/NeoPixelBus/wiki/NeoPixelBus-object).
I leave the bit bang around only for diagnostics as it is not functional with WiFi active.

My library already has a similar animation system, see [NeoPixelAnimator](https://github.com/Makuna/NeoPixelBus/wiki/NeoPixelAnimator-object) in my library.

PhilG, 2016-04-13:
It's a fair point, I'm very aware the interrupts on the ESP can get in the way of writing out the data via Bitbang and ready to change as required. For now I'd like to see how far I can go with the current setup I have. 

I figured the WiFi library would have something in it that somehow reproducably put the first LED to green, so I just restructured the `setup()` method so it's not noticeable anymore. Currently I have WiFi active on the ESP with a connection to an MQTT server and it's been doing fine all along, no resets and no exceptions. If it ever starts acting up I'll look more into your NeoPixelBus library (UART method would fit with the soldered board I have) or FastLED where interrupts are kept enabled if I understood the little I read about it!