esp8266/Arduino #1893: Flash bin file to esp

Lexander24, 2016-04-11:
I want to ask.
Can i flash the bin file to esp after i made the code using arduino?

Thx

Rickgg, 2016-04-11:
Sure, you can use [EspTool](https://github.com/themadinventor/esptool) to upload it directly. Also support for OTA updates is already in the Arduino IDE.

Lexander24, 2016-04-11:
Hey rick thx for reply.
Yesterday i try to upload using flash download tool by expressif but i confuse with the configuration. What configuration i must set? Thx

igrr, 2016-04-11:
You need to upload a single "bin" file, at address 0x00000. Also make sure you set flash size to the same value as the size you selected in Arduino IDE.

Lexander24, 2016-04-11:
Thx for reply, i will try again today and inform you what happen. Thx

closed 2016-06-23 by igrr

hason124, 2018-01-02:
I have flashed your firmware and I get the green flash on boot. But I am guessing its configured for your network so I cant access it on the hue app since its not connected to my network. I dont see the New Hue Light ssid.