esp8266/Arduino #1894: Nodemcu reset when trying to establish a secured connection with WiFiClientsecure

HW-Siew, 2016-04-11:
Hi, I tried the HTTPSRequest example. It worked fine to verify the domain "api.github.com". However, when i tried to establish a connection with my own server "t2.qrwow.com'. The nodemcu reset, and following are the stack dump. I couldn't understand the messages tho.  I confirmed that my server supports tls 1.1. Any advice are very much appreciated. Thank you.

```
Decoding 13 results
0x40228f32: ssl_read at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 260
0x40229121: do_client_connect at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 161
0x4021b5d6: tcp_connect at ?? line ?
0x402291ac: ssl_client_new at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 71
0x4020334e: SSLContext::connect(ClientContext*) at C:\Users\HW\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 428
:  (inlined by) WiFiClientSecure::_connectSSL() at C:\Users\HW\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 249
0x4020403c: esp_yield at C:\Users\HW\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\cores\esp8266/core_esp8266_main.cpp line 43
0x4020361a: WiFiClientSecure::connect(IPAddress, unsigned short) at C:\Users\HW\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 233
0x40202809: WiFiClient::connect(char const*, unsigned short) at C:\Users\HW\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\libraries\ESP8266WiFi\src/WiFiClient.cpp line 362
0x40204880: Print::println() at C:\Users\HW\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\cores\esp8266/Print.cpp line 131
0x40203630: WiFiClientSecure::connect(char const*, unsigned short) at C:\Users\HW\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 239
0x402021f4: setup at C:\Users\HW\AppData\Local\Temp\arduino_modified_sketch_509988/HTTPSRequest.ino line 52
0x40204080: loop_wrapper at C:\Users\HW\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\cores\esp8266/core_esp8266_main.cpp line 43
0x40100114: cont_norm at C:\Users\HW\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0-rc2\cores\esp8266/cont.S line 109
```

igrr, 2016-04-18:
Is this the same issue as #1927?

HW-Siew, 2016-04-18:
Not exactly the same. I worked around to the other way to avoid the issue, but i think the latest version library should solve this as well. Haven't tested tho. Close this for now.

Thanks ;)

HW-Siew, 2016-04-20:
Hi @igrr, I am referring to #1375. You mentioned that not all cert with 2048 key can work for ssl. Is that still valid after the issue is closed and fixed? For the reason, i am still experiencing WDT reset for some domains i tested which all of them has 2048 key.

Thanks for the advice.

HW-Siew, 2016-04-20:
I think i know why. This is probably current lib doesn't support SNI yet as reported in #1285.

closed 2016-04-20 by HW-Siew