esp8266/Arduino #1896: In WiFi STA if static IP is used there is no way to go back to dhcp

baruch, 2016-04-11:
### Basic Infos

When I set static IP I can't go back to dhcp mode unless I reset the esp completely since _useStaticIp is set and cannot be reset in any way.

In addition, if I just set WiFi.config() with all zeros this is considered as using a static ip instead of resetting to dhcp as I would have expected.

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0?

igrr, 2016-06-12:
`WiFi.config(0U, 0U, 0U);` should now work in git version.

baruch, 2016-06-19:
If the issue is committed to 2.3.0, shouldn't the issue be closed?

igrr, 2016-06-20:
Lacking more detailed issue states (i.e. "resolved" versus "closed"), i use "staged-for-release" label and keep issues open until the fix is released. 
Reason for that is that people tend to search only for open issues (this is the default behaviour of issue search box). As it happened here on multiple occasions, once I close an issue, another identical one gets opened because the old one wasn't visible in search anymore.

baruch, 2016-06-20:
Understood. Thanks.

closed 2016-06-23 by igrr