esp8266/Arduino #1898: 18 Line Sketch fails to compile with ESP8266

pieman64, 2016-04-11:
### Basic Infos

#### Hardware
Hardware:	        ESP-07
Core Version:      	2.1.0
Windows 8.1 IDE 1.6.7

### Description

Problem description ESP8266 stops compiler, if I remove the ESP library in the first line of the sketch all is fine. As lowInterrupt() calls highInterrupt() and vice versa there is no way for me to do the function dance brought in by Arduino after IDE 1.6.5. Spent all day trying to get a basic switch to work with an ESP and after finally cracking it the IDE will not support it once I include ESP8266. What is the fix for this?

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1M/64K
CPU Frequency:      80Mhz
Flash Mode:		DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck 

### Sketch

#include <ESP8266WiFi.h>
#define BuzzerPin 0

void highInterrupt(){
    detachInterrupt(BuzzerPin);
    attachInterrupt(BuzzerPin, lowInterrupt, FALLING);
}

void lowInterrupt(){
    detachInterrupt(BuzzerPin);
    attachInterrupt(BuzzerPin,  highInterrupt, RISING);
}

void setup() {
}

void loop() {
}
```

### Debug Messages

'lowInterrupt' was not declared in this scope
```
messages here
```

igrr, 2016-04-11:
Have you tried 1.6.8? I think this issue was fixed after 1.6.7 release.

pieman64, 2016-04-11:
@igrr are you running 1.6.8 and if so can you compile the sketch before I go to the trouble of moving to 1.6.8 to find it still fails?

igrr, 2016-04-11:
Sure, I'll try in a few hours once I'm at my desk.

pieman64, 2016-04-11:
Compiles with IDE 1.6.8.
Problem is my libraries are now messed up as my real sketches will no longer accept Strings.

pieman64, 2016-04-11:
Correction String is now fine but the original bug is still there with IDE 1.6.8 in an actual 1000 row sketch.

The only compilation error is highInterrupt() calling lowInterrupt and vice versa.

Any fix for this crazy function order mess?

igrr, 2016-04-11:
Could you please post the sketch which triggers the issue? Thanks.

pieman64, 2016-04-11:
This is the 1000+ line sketch reduced to less than 100 lines with the compile error still present.
I have included all the libraries that I use in the full sketch. The error is at line 65, comment this out and it is fine.

```
// *******************************************************************************/
#include <FS.h>                   //this needs to be first, or it all crashes and burns...
#include <DNSServer.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager
#include <ArduinoJson.h>      //https://github.com/bblanchon/ArduinoJson

#define BLYNK_PRINT Serial 
#include <BlynkSimpleEsp8266.h>
#include <SimpleTimer.h>
#include <TimeLib.h>
#include <WidgetRTC.h>

// this is for webserver update of firmware
#include <Arduino.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>
//#define USE_SERIAL Serial
ESP8266WiFiMulti WiFiMulti;

WiFiClient client;  // Use WiFiClient class to create TCP connections

#include <OneWire.h>
#include <DallasTemperature.h>
#define ONE_WIRE_BUS 14      // DS18B20 GPIO 14 Wemos D5
OneWire oneWire(ONE_WIRE_BUS);
DallasTemperature DS18B20(&oneWire);

// GPIO 0 has 10K pullup
#define BuzzerPin 0  // was 14 WeMos D5 now WeMos D3 on GPIO 0
#define TONE_PIN 13
bool soundbuzzer = false;
int volume = 2000; // TODO change to 2000 for maximum volume 

SimpleTimer timer;

void playTone(int _pin, int _frequency, int _length){
  analogWriteFreq(_frequency);
  analogWrite(_pin, 512);
  delay(_length);
  analogWrite(_pin, 0);
} 

void buzzerloop(){  
      playTone(TONE_PIN, volume, 200);  // GPIO 13 Frequency 2000 Hz and 200 milliseconds duration
      delay(200);
      playTone(TONE_PIN, volume, 200);
      delay(200);
} 

void toggle(){
  if(soundbuzzer == true){
    buzzerloop();
  }
}

void lowInterrupt(){
    detachInterrupt(BuzzerPin);
    attachInterrupt(BuzzerPin,  highInterrupt, RISING);
}

void highInterrupt(){
    detachInterrupt(BuzzerPin);
    attachInterrupt(BuzzerPin, lowInterrupt, FALLING);
    soundbuzzer = !soundbuzzer;
    if(soundbuzzer == true){
      Serial.println("Buzzer ON");
    }
    else{
      Serial.println("Buzzer OFF");
    }
}

void setup(){
    Serial.begin(115200);
    delay(10);
    pinMode(BuzzerPin, INPUT_PULLUP); // for GPIO 0 INPUT or INPUT_PULLUP seems ok
    attachInterrupt(BuzzerPin, highInterrupt, RISING); // use for Gnd connection
    //timer.setInterval(1000L, toggle);
}

void loop(){
    //timer.run();
    toggle();
    delay(10);
}
```

igrr, 2016-04-11:
Thanks, I have reproduced the issue. 
The minimal test case seems to be 
```c++
#include <vector>

void setup() {
  test();
}
void loop() {}
void test() {}
```
I will report this to arduino-builder team.

It seems that removing `#include <ESP8266WiFiMulti.h>` and `ESP8266WiFiMulti WiFiMulti;` lines works around this bug.

pieman64, 2016-04-11:
Thanks @igrr I was just going to say the problem is associated with `<ESP8266WiFiMulti.h>`

See 55 line sketch and with `<ESP8266WiFiMulti.h>` commented out it compiles ok but obviously I need `<ESP8266WiFiMulti.h>`

```
//#include <ESP8266WiFiMulti.h>

#define BuzzerPin 0  // was 14 WeMos D5 now WeMos D3 on GPIO 0
#define TONE_PIN 13
bool soundbuzzer = false;
int volume = 2000; // TODO change to 2000 for maximum volume 

void playTone(int _pin, int _frequency, int _length){
  analogWriteFreq(_frequency);
  analogWrite(_pin, 512);
  delay(_length);
  analogWrite(_pin, 0);
} 

void buzzerloop(){  
      playTone(TONE_PIN, volume, 200);  // GPIO 13 Frequency 2000 Hz and 200 milliseconds duration
      delay(200);
      playTone(TONE_PIN, volume, 200);
      delay(200);
} 

void toggle(){
  if(soundbuzzer == true){
    buzzerloop();
  }
}

void lowInterrupt(){
    detachInterrupt(BuzzerPin);
    attachInterrupt(BuzzerPin,  highInterrupt, RISING);
}

void highInterrupt(){
    detachInterrupt(BuzzerPin);
    attachInterrupt(BuzzerPin, lowInterrupt, FALLING);
    soundbuzzer = !soundbuzzer;
    if(soundbuzzer == true){
      Serial.println("Buzzer ON");
    }
    else{
      Serial.println("Buzzer OFF");
    }
}

void setup(){
    Serial.begin(115200);
    delay(10);
    pinMode(BuzzerPin, INPUT_PULLUP); // for GPIO 0 INPUT or INPUT_PULLUP seems ok
    attachInterrupt(BuzzerPin, highInterrupt, RISING); // use for Gnd connection
}

void loop(){
    toggle();
    delay(10);
}
```

igrr, 2016-04-11:
Do you really need it? Since you are using WiFiManager library, i think you can live without ESP8266WiFiMulti perfectly well.
In fact, I don't even see how these two libraries may be used together.

fivdi, 2016-04-11:
Is the underlying issue here a missing forward declaration?

This works:
```cpp
#include <vector>

void test();

void setup() {
  test();
}
void loop() {}
void test() {}
```

igrr, 2016-04-11:
This forward declaration shouldn't be necessary in Arduino environment, because it is supposed to generate forward declarations for you. It is also supposed to generate forward declarations for stuff defined in other files of a multi-file sketch.

The underlying issue is that when `#include <vector>` happens to be in the sketch, something wrong happens inside the tool which is generating these forward declarations.

Adding forward declarations manually is obviously a workaround as well.

pieman64, 2016-04-11:
@igrr  I was under the impression I needed those libraries to be able to retrieve a file from one of my servers i.e. the firmware updates to the device actually making the request.

Back to the 1000+ lines sketch when I remove `#include <ESP8266WiFiMulti.h> and ESP8266WiFiMulti WiFiMulti;`the compiler still fails with the highinterrupt calling the low interrupt and vice versa so there are other bugs at play here.

igrr, 2016-04-11:
ESP8266WiFiMulti is a library which manages connection to multiple access points. It doesn't have anything to do with downloading firmware updates. It just happened that the firmware update example also used ESP8266WiFiMult instead of "normal" ESP8266WiFi.

igrr, 2016-04-11:
As mentioned above, you may add `void lowInterrupt(); void highInterrupt();` declarations above these functions as a workaround.
If you post the failing 1000-line sketch, that would also be cool. You may post it at [gist.github.com](https://gist.github.com) to keep this thread readable.

pieman64, 2016-04-11:
@igrr can you explain how I add forward declarations as I still can't compile the real sketch?

igrr, 2016-04-11:
1. Add `void lowInterrupt(); void highInterrupt();` declarations above definitions of `lowInterrupt` and `highInterrupt`.

2. If it still fails to compile, copy the complete sketch, paste it into pastebin.com or gist.github.com and place a link here please.

pieman64, 2016-04-11:
Compiling now with your fix, thanks.

gmag11, 2016-04-19:
This happens since some recent update of Arduino IDE, I don't remember which one.
Anyway this forces us to declare functions before using them, for instance using .h files. I think it is a good practise.

marvinroger, 2016-04-19:
@gmag11 This is a C++ practice (even requirement), but not an Arduino one, as it is aimed at beginners.

devyte, 2017-10-17:
I can't reproduce this with IDE 1.8.3. Closing.

closed 2017-10-17 by devyte