esp8266/Arduino #1899: New staging release?

obsoleted, 2016-04-11:
Would it be possible to get a new version with the latest changes published to the staging url? (http://arduino.esp8266.com/staging/package_esp8266com_index.json)

There have been some great fixes that would be easier to get at if available via the board manager method. (Easier to document anyway :wink: )

marvinroger, 2016-04-12:
Definitely! What about a release schedule? The staging version is often older than the stable version.

igrr, 2016-04-12:
The release *was* scheduled on April 1. I'm working to get it ready as soon as possible, running tests against pending pull requests for the last few days :)

WoodsterDK, 2016-04-15:
Hi there,
I think the board manager link for the latest release of the staging version hasn't been updated...

marvinroger, 2016-04-15:
@WoodsterDK I see 2.2.0-rc1 in the boards manager, so it seems it is.

PhilColbert, 2016-04-15:
Think thats just the documentation link that hasnt been changed, yes it is confusing ! :)

WoodsterDK, 2016-04-15:
Looking in the http://arduino.esp8266.com/staging/package_esp8266com_index.json file just shows links to 2.1zips. I think the header is OK, but not the content....

igrr, 2016-04-15:
Have you scrolled down a bit? 2.2.0-rc1 is the last on in the list of packages, before the list of tools.

obsoleted, 2016-04-15:
I see 2.2.0-rc1 and was able to use it. Thanks for updating it!

The issue can be closed as far as I'm concerned. Unless others object?

 Perhaps a separate issue is needed for doc updates if any are needed though.

WoodsterDK, 2016-04-15:
But why does the json file link to https://github.com/esp8266/Arduino/releases/download/2.1.0-rc1/esp8266-2.1.0-rc1.zip",

igrr, 2016-04-15:
Json file lists all previous versions as well. You can select the version to install from the dropdown in boards manager.

WoodsterDK, 2016-04-15:
Ahh sorry my mistake.... Didn't look at the end of the file....

closed 2016-04-15 by obsoleted