esp8266/Arduino #1901: Switch to stop all serial output

PhilColbert, 2016-04-12:
Hi

Firstly can I thank the developers of this amazing project, moving from 2.0.0 to 2.1rc1 has made a massive difference to my project, crashes out are now every 12-15 hours from every hour ! Thank you !

I was wondering if there is, or there could be a switch to stop all serial output, including exception stack dumps for live applications, currently I remove all the ets_print statements to stop as much as possible.

Currently using Serial for transmitting directly to an RF module so for live versions if the program crashes it sends all the data out over the RF, this is bad ! :)

Thanks.

devyte, 2017-10-17:
@PhilColbert Is this request meant for HardwareSerial or SoftwareSerial? It is not possible to mute all output for HardwareSerial due to messages printed at boot time. Therefore, any devices connected to the HardwareSerial _must be tolerant_.

devyte, 2018-07-12:
Closing due to lack of feedback.

closed 2018-07-12 by devyte