esp8266/Arduino #1902: Cannot call more than once ESP8266HttpClient POST()

far5893, 2016-04-12:
If i define ESP8266HttpClient websebd;
websend.begin("http://192.168.1.1/htpxml.php");
websend.addHeader("Content-Type", "text/xml");
websend.addHeader("Cookie:","XDEBUG_SESSION_START=session0"); 

then i call it two times:

websend.POST("some xml text");
websend.end();

websend.POST("another  xml text");
websend.end();

POST /htpxml.php HTTP/1.1
Host: 192.168.1.1
User-Agent: ESP8266HTTPClient
Connection: close
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
Content-Type: text/xml
Cookie:: XDEBUG_SESSION_START=session0
Content-Length: 744
Content-Length: 744

(Wireshark log)

if(payload && size > 0) {
        addHeader("Content-Length", String(size));
    }

don't know if it's the right mode to use object, but i would like to avoid 
recreate the same object every time i have to send same file to same 
url with same header!!

 

 if(payload && size > 0) {
        addHeader("Content-Length", String(size));
 }

TheAustrian, 2016-04-12:
I don't know if it's cause you call the .end() but regarding setting up stuff only once I made myself a function that I then use in the loop().

int SendSensorData (String Sensordata) {
  // returns negative value if unsuccessful
  // returns 0 if successful
  String returnvalue = "-1";
  if (Sensordata != "" ) {
    HTTPClient http;
    http.begin(phppage);
    http.addHeader("Content-Type", contenttype);
    int httpCode = http.POST(Sensordata);
    if (httpCode != 200) {
      Serial.println("not successful");
      } else {
      returnvalue = http.getString();
      Serial.println("successful:" + returnvalue);   
      }
    http.end();
  } 
  int returnval = returnvalue.toInt();
  return returnval;
}

far5893, 2016-04-13:
Apache does not agree with more than one Content-Length  line in request, 
i'dont know why your reuse work!!

closed 2016-06-23 by igrr