esp8266/Arduino #1904: installing the board support using github downloads fails on windows using python3

hlima, 2016-04-12:
### Basic Infos

#### Hardware
Hardware:			any
Core Version:      	?2.1.0-rc2?

### Description
Instructions to install from github does not work if using python 3.
One of the steps to install is to run "python get.py". 
The file get.py  is written for python 2 and fails if python3 is used. 
I was able to get it to work doing the following additional steps (on windows platform)

1. run 2to3 on get.py and build.py files to get them converted to python3:
     2to3 -w get.py
     2to3 -w build.py

2. edit the resulting get.py  file and change line 58 as shown below:
    #rename_to = re.match(r'^([a-z][^\-]*\-*)+', dirname).group(0).**encode('ascii')**.strip('-')
    rename_to = re.match(r'^([a-z][^\-]*\-*)+', dirname).group(0).strip('-')

After those steps I was able o complete the installation.
Here are the files I modified:

[get.py.zip](https://github.com/esp8266/Arduino/files/215798/get.py.zip)

igrr, 2016-04-12:
Does the modified script still work with python2?
If it does, could you please open a pull request with these xhanges? Thanks.

devyte, 2017-10-17:
The rename_to line is already changed as specified in latest git, and there have been several commits for p3 compat in the interim, so closing. If the issue is still valid, please open a new issue with updated details.

closed 2017-10-17 by devyte